margelo/react-native-vision-camera #3336: 🐛 Frame processor is getting distorter, whereas inbuild code scanner is working fine

Swikriti0998, 2024-12-20:
### What's happening?

I have to process the barcode and OCR both in my application therefore I am relying on frame processor but the frame processor returns distorted UI and even without using any other plugin. When I use the builtIn barcode scanner the video is not distorted but then for the OCR I have to rely on taking snapshots and processing that instead.

There is a major difference between built in frame processor and code scanner in all android devices. 
This is from frame processor
[image]
This when using code scanner
[image]

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
		'worklet';

		console.log(frame.height);
	}, []);

	const codeScanner = useCodeScanner({
		codeTypes: ['qr', 'ean-13'],
		onCodeScanned: (codes) => {
			'worklet';

			console.log(`Scanned ${codes.length} codes!`);
		},
	});
```

### Relevant log output

```shell
12-20 15:06:34.738 31241 31879 I CameraView: invokeOnAverageFpsChanged(19.95798319327731)
12-20 15:06:34.772 31241 31430 I ReactNativeJS: 1080
12-20 15:06:34.776   449 31942 E awb_adpt: 2336, awb_sprd_ctrl_calculation_v3: check handle success
12-20 15:06:34.776   449 31936 E smart_ctrl: 1911, smart_ctl_calc_atm: jhin_atm atm_switch_state:0 1
12-20 15:06:34.776   449 31936 E smart_ctrl: 1914, smart_ctl_calc_atm: jhin_atm atm_version 2 
12-20 15:06:34.776   449 31936 D ATM     : 1099, atm_calc: smooth out frame_id 70 u4EVD 256 smooth u4EVD 256.000000  bv -266
12-20 15:06:34.779   449 31924 I Libae[HM]: 1073,hm:cam:ae,WEIGHT IS -267,0,6,15
```

### Camera Device

```json
{
  "hardwareLevel": "limited",
  "minExposure": -32,
  "neutralZoom": 1,
  "minZoom": 1,
  "supportsFocus": true,
  "formats": [],
  "supportsLowLightBoost": false,
  "hasTorch": true,
  "supportsRawCapture": false,
  "minFocusDistance": 0.09775171065493646,
  "sensorOrientation": "landscape-left",
  "maxZoom": 4,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "isMultiCam": false,
  "position": "back",
  "id": "0",
  "hasFlash": true,
  "maxExposure": 32,
  "name": "0 (BACK) androidx.camera.camera2"
}
```

### Device

narzo 50i (Android)

### VisionCamera Version

4.5.1/4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Swikriti0998, 2024-12-20:
Apparently the issue is with the camera format if you use the format key in the camera component is causes the frame to give distorted image even if you are using the current resolution. 

`			<Camera
					device={device}
					isActive={true}
					format={format} // this is causing distorted video with frame processor.
					onError={handleError}
					style={{ height: cameraHeight, width }}
				/>`

gyanbhartip, 2025-01-07:
> Apparently the issue is with the camera format if you use the format key in the camera component is causes the frame to give distorted image even if you are using the current resolution.
>
> `<Camera [...]

I can confirm the same is happening. I wanted to limit barcode scanning FPS but it requires passing the format prop, which even though is set at the current best, results in a very low quality preview.

Swikriti0998, 2025-02-07:
@mrousavy, can you please take a look at this and let us know if we're missing anything? With the format, the reading is faster, but the video appears distorted. It’s working fine on iOS, though.

ronickg, 2025-03-05:
Noticing the same issue on android when just enabling/disabling frame processor. On the emulator it seems to be working correctly, but on the real device the image gets very blurry/pixelated.

hellomalith, 2025-03-06:
I also get the same issue on Ulefone Note 14 (Android 12) physical device. Same code works perfectly on other physical devices.

[image]

only happens when passing both `format` and `frameProcessor` props.
Even with `max` resolution, it doesn't work.

Tried with both `react-native-vision-camera` **4.6.1** and **4.6.4**

MuratUlugg, 2025-06-26:
I have the same problem