margelo/react-native-vision-camera #3344: 🐛 Crash occurring during native module initialisation while opening app

FarzanAkhtar, 2024-12-30:
### What's happening?

The app is crashing on some of our devices as soon as it opens even when not in the flow which requires Camera component.
Occurring at `androidx.concurrent.futures.b$b`
`The completer object was garbage collected - this future would otherwise never complete. The tag was: Initialize extensions`
As far as I can tell this looks to be happening from when we create an `ExtensionsManger` object via `ExtensionsManager.getInstanceAsync(context, provider).await(mainExecutor)`, somehow the completer object provided to the future is getting garbage collected, so we will need to handle this edge case scenario.

Note: was not able to replicate this locally, getting crashes on production app, so putting in sample value for Camera Device field

### Reproduceable Code

```tsx
<Camera
  ref={cameraRef}
  format={selectedFormat?.current}
  isActive={true}
  photo={true}
  style={styles.container}
  device={device}
  video={false}
  onError={onCameraError}
  onOutputOrientationChanged={onOrientationChange}
  photoQualityBalance="speed"
/>
```

### Relevant log output

```shell
Fatal Exception: androidx.concurrent.futures.b$b: The completer object was garbage collected - this future would otherwise never complete. The tag was: Initialize extensions
CameraManagerGlobal:
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:350)
       at android.os.Looper.loopOnce(Looper.java:187)
       at android.os.Looper.loop(Looper.java:368)
       at android.os.HandlerThread.run(HandlerThread.java:67)
mrousavy/VisionCamera.video:
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:350)
       at android.os.Looper.loopOnce(Looper.java:187)
       at android.os.Looper.loop(Looper.java:368)
       at android.os.HandlerThread.run(HandlerThread.java:67)
CameraX-scheduler:
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:350)
       at android.os.Looper.loopOnce(Looper.java:187)
       at android.os.Looper.loop(Looper.java:368)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 6,
  "minZoom": 1,
  "maxExposure": 4,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -4,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Realme RMX3998 (Android 14),  Realme 12 5G (Android 14)

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

FarzanAkhtar, 2024-12-30:
Don't have adb logcat logs as crashes are occurring for production users and we were unable to replicate this locally.

lve0805, 2025-03-20:
we are facing the same crash =(