margelo/react-native-vision-camera #3345: 🐛 MinZoom Problem in some devices (0,5)

dorakadri, 2024-12-31:
### What's happening?

in some devices the min zoom is always 1 even tho its supposed to be 0,5 and that the physical device shows as ultra wide range 

### Reproduceable Code

```tsx
const devices = Camera.getAvailableCameraDevices()

  const device = getCameraDevice(devices, 'back', {
    physicalDevices: ['ultra-wide-angle-camera', 'wide-angle-camera', 'telephoto-camera']
  })
  const zoom = useSharedValue(device?.neutralZoom)
  const format = useCameraFormat(device, Templates.Snapchat)
  const zoomOffset = useSharedValue(0)
  const [zoomState, setZoomState] = useState(device?.neutralZoom || 1)

  console.log()
  useDerivedValue(() => {
    if (!zoom.value) return
    runOnJS(setZoomState)(zoom.value)
    return zoom.value
  }, [zoom.value])
  const gesture = Gesture.Pinch()
    .onBegin(() => {
      if (zoom.value) {
        zoomOffset.value = zoom.value
      }
    })
    .onUpdate(event => {
      const z = zoomOffset.value * event.scale
      if (device) {
        zoom.value = interpolate(z, [1, 10], [device.minZoom, device.maxZoom], Extrapolation.CLAMP)
      }
    })

<ReanimatedCamera
          ref={cameraRef}
          outputOrientation='preview'
          style={isInitialized ? StyleSheet.absoluteFill : styles.noSize}
          onTouchEnd={focus}
          onShutter={playShutterSound}
          device={device}
          isActive={isActive}
          format={format}
          animatedProps={animatedProps}
          photo
          onError={console.warn}
          onLayout={event => {
            const { layout } = event.nativeEvent
            setCameraLayout(layout)
          }}
          onInitialized={() => {
            setTimeout(() => {
              if (
                device.physicalDevices.includes('ultra-wide-angle-camera') ||
                  device.physicalDevices.includes('wide-angle-camera') ||
                  device.physicalDevices.includes('telephoto-camera')
              ) {
                setWideRangeAvailable(true)
              }
              setIsInitialized(true)
            }, 10)
          }}
```

### Relevant log output

```shell
no relevant log only the json ill show before
```

### Camera Device

```json
{
  "id": "2",
  "devices": [
    "ultra-wide-angle-camera"
  ],
  "formats": [{...},{...},{...},{...},{...},{...},{...},{...},{...},{...},{...},{...},{...},{...},{...},],
  "hardwareLevel": "limited",
  "hasFlash": false,
  "hasTorch": false,
  "isMultiCam": false,
  "minZoom": 1,
  "neutralZoom": 1,
  "maxZoom": 8,
  "name": " 2 (BACK) androidx.camera.camera2",
  "position": "back",
  "supportsFocus": true,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false
}
```

### Device

Samsung Galaxy a73 5g 

### VisionCamera Version

4.5.1 / 4.6.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

wadiasalem, 2024-12-31:
I have the same issue

prakashpro3, 2025-01-16:
I have same issue
[
  {
    "sensorOrientation": "landscape-left",
    "hardwareLevel": "full",
    "maxZoom": 10,
    "minZoom": 1,
    "maxExposure": 18,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": false,
    "minFocusDistance": 0,
    "minExposure": -18,
    "name": "0 (BACK) androidx.camera.camera2",
    "hasFlash": true,
    "hasTorch": true,
    "position": "back",
    "id": "0"
  },
  {
    "sensorOrientation": "landscape-right",
    "hardwareLevel": "full",
    "maxZoom": 4,
    "minZoom": 1,
    "maxExposure": 12,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": false,
    "minFocusDistance": 0,
    "minExposure": -12,
    "name": "1 (FRONT) androidx.camera.camera2",
    "hasFlash": false,
    "hasTorch": false,
    "position": "front",
    "id": "1"
  }
]

dorakadri, 2025-01-16:
@prakashpro3  well i've talked with the support of android and it seems like the problem is in some devices but they left a snippet of code that maybe can help but it need to be done inside the package maybe  @mrousavy  is interested ill leave the link here maybe it will give you some clarity https://groups.google.com/a/android.com/g/camerax-developers/c/N4YtXK-6-CU

mrousavy, 2025-01-20:
hey - hm this actually seems like a CameraX inconsistency. Thanks for creating the bug report in the CameraX issue tarcker @dorakadri ! :)

mrousavy, 2025-01-20:
I think the snippet Scott shared in the CameraX issue tracker does not apply here. VisionCamera already grabs _all_ Cameras, so you can select the one you want. This is not the same issue if I read this correctly

mrousavy, 2025-01-20:
>  From what I’ve read in the documentation, I expected to see naturalZoom = 2.

@ugar0ff min/neutral/max zoom are _factors_. Max zoom of 8 means you can zoom in 8x of the default view.

ugar0ff, 2025-01-20:
I continue testing,
:white_check_mark: on iOS everything worked as expected according to the documentation. Here is the camera I selected:
`"maxZoom": 123.75, "minZoom": 1, "neutralZoom": 2, "physicalDevices": ['ultra-wide-angle-camera', 'wide-angle-camera', 'telephoto-camera'], "position": "back"`

As a result, on the UI, I display the zoom range from 0.5 to 15, but I send twice the value to the library (from 1 to 30). The neutralZoom: 2 is displayed on the UI as 1.

However, 
:x: on Android, there are several issues. I tested on three different Android devices (1 Xiaomi and 2 Samsung), and I can say the issues are common:

1. The `physicalDevices` array always contains only one value.
2. The `neutralZoom` is always 1, even for the `ultra-wide-angle camera`, even though in this case, a zoom of 1 is ultra-wide and gives the same result as the native camera at 0.5.
3. The `maxZoom` is smaller than what the native camera can achieve.

Example of all "back" cameras from my Samsung:
```
"id": "0", "physicalDevices": ["wide-angle-camera"], "position": "back", "minZoom": 1, "neutralZoom": 1, "maxZoom": 8
"id": "2", "physicalDevices": ["ultra-wide-angle-camera"], "position": "back", "minZoom": 1, "neutralZoom": 1, "maxZoom": 8
```

qq382724935, 2025-03-08:
I have also encountered this problem. Is there a solvable solution！