margelo/react-native-vision-camera #3351: 🔧 Execution failed for task ':react-native-vision-camera:prepareBoost'.

bacancyshreyansh, 2025-01-04:
### How were you trying to build the app?

The build is failing on android due to this react-native-vision-camera/android/build/downloads/boost_1_76_0.tar.gz

### Full build logs

```tsx
> Task :react-native-vision-camera:compileDebugKotlin
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (7, 44): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (42, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (42, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (48, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (48, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt: (27, 44): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewManager.kt: (11, 25): Parameter 'reactContext' is never used
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt: (72, 11): 'onCatalystInstanceDestroy(): Unit' is deprecated. Overrides deprecated member in 'com.facebook.react.bridge.NativeModule'. Deprecated in Java

> Task :react-native-webview:compileDebugKotlin
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (40, 71): Parameter 'webView' is never used, could be renamed to _
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (82, 18): 'setter for allowFileAccessFromFileURLs: Boolean' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (83, 18): 'setter for allowUniversalAccessFromFileURLs: Boolean' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (94, 102): Parameter 'contentLength' is never used, could be renamed to _
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (111, 32): Variable 'urlObj' initializer is redundant
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (125, 21): 'allowScanningByMediaScanner(): Unit' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (162, 36): 'setter for systemUiVisibility: Int' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (344, 15): Condition 'args == null' is always 'false'
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (353, 34): Condition 'args != null' is always 'true'
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (372, 38): 'setter for allowUniversalAccessFromFileURLs: Boolean' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (384, 19): Parameter 'viewWrapper' is never used
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (433, 51): Unchecked cast: Any! to kotlin.collections.HashMap<String, String> /* = java.util.HashMap<String, String> */
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (489, 23): 'setter for savePassword: Boolean' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (490, 23): 'setter for saveFormData: Boolean' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (560, 23): 'setter for allowFileAccessFromFileURLs: Boolean' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (661, 65): Unchecked cast: ArrayList<Any!> to List<Map<String, String>>
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (682, 23): 'setter for saveFormData: Boolean' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/RNCWebViewManagerImpl.kt: (717, 36): Parameter 'viewWrapper' is never used
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/events/TopCustomMenuSelectionEvent.kt: (5, 44): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/events/TopCustomMenuSelectionEvent.kt: (11, 3): 'constructor Event<T : Event<(raw) Event<*>>!>(Int)' is deprecated. Deprecated in Java
w: /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-webview/android/src/main/java/com/reactnativecommunity/webview/events/TopCustomMenuSelectionEvent.kt: (22, 42): 'RCTEventEmitter' is deprecated. Deprecated in Java
[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2025-01-04 by maintenance-hans[bot]

bacancyshreyansh, 2025-01-04:
Hi @mrousavy ,

I am using 

"react-native-vision-camera": "2.15.2",
    "react-native": "0.70.15",
    
    The build was completely working fine till 31st December 2024, but from 2nd Jan 2025, While building the project i am getting errors due to 
    
    android/build/downloads/boost_1_76_0.tar.gz.
    
    Seems like an error with boost_1_76_0.tar.gz
    
    * What went wrong:
Execution failed for task ':react-native-vision-camera:prepareBoost'.
> Could not read /Users/apple/Desktop/git-xcira/flc-mobile-fastlane2/node_modules/react-native-vision-camera/android/build/downloads/boost_1_76_0.tar.gz.
   > Not in GZIP format
   
   
   Can you please help me on this

bacancyshreyansh, 2025-01-04:
Fix:

Insise
node_modules/react-native-vision-camera/android/build.gradle

replace 

  def srcUrl = "https://boostorg.jfrog.io/artifactory/main/release/${transformedVersion}/source/boost_${BOOST_VERSION}.tar.gz"
  
  
  with 

 def srcUrl = "https://archives.boost.io/release/${transformedVersion}/source/boost_${BOOST_VERSION}.tar.gz"
 
 
 
 
 Reference:  https://github.com/facebook/react-native/pull/42118/files

eralpkaraduman, 2025-01-07:
> Fix:
> 
> Inside node_modules/react-native-vision-camera/android/build.gradle
> 
> replace
> 
> def srcUrl = "[https://boostorg.jfrog.io/artifactory/main/release/${transformedVersion}/source/boost_${BOOST_VERSION}.tar.gz](https://boostorg.jfrog.io/artifactory/main/release/$%7BtransformedVersion%7D/source/boost_$%7BBOOST_VERSION%7D.tar.gz)"
> 
> with
> 
> def srcUrl = "[https://archives.boost.io/release/${transformedVersion}/source/boost_${BOOST_VERSION}.tar.gz](https://archives.boost.io/release/$%7BtransformedVersion%7D/source/boost_$%7BBOOST_VERSION%7D.tar.gz)"
> 
> Reference: https://github.com/facebook/react-native/pull/42118/files

Thanks @bacancyshreyansh 

For those who want to include this change in your [patch-package](https://www.npmjs.com/package/patch-package) patches,

I had success by excluding everything else than the file with related change

`npx patch-package react-native-vision-camera --exclude '^(?!android/build.gradle$)'`

otherwise patch-package tries to diff everything else and runs out of buffer eventually fails.