margelo/react-native-vision-camera #3360: 🔧 Every now and then vision camera stops building

agoransson, 2025-01-10:
### How were you trying to build the app?

`npx expo run:android`

It seems to me that the project is very fragile; even though it builds at one time. From one day to the other, if the expo server has been shut down in the afternoon - the morning after it will just not build any more. Often with this linking error.

Really stops working without having changed anything. Very odd behavior I think.

### Full build logs

```tsx
› Building app...
Configuration on demand is an incubating feature.

> Configure project :app
 ℹ️  Applying gradle plugin 'expo-dev-launcher-gradle-plugin' (expo-dev-launcher@5.0.21)

> Configure project :shopify_react-native-skia
react-native-skia: node_modules/ found at: /Users/agoransson/Projects/ReactNativeProject/my-app/node_modules
react-native-skia: RN Version: 76 / 0.76.5
react-native-skia: isSourceBuild: false
react-native-skia: PrebuiltDir: /Users/agoransson/Projects/ReactNativeProject/my-app/node_modules/@shopify/react-native-skia/android/build/react-native-0*/jni
react-native-skia: buildType: debug
react-native-skia: buildDir: /Users/agoransson/Projects/ReactNativeProject/my-app/node_modules/@shopify/react-native-skia/android/build
react-native-skia: node_modules: /Users/agoransson/Projects/ReactNativeProject/my-app/node_modules
react-native-skia: Enable Prefab: true
react-native-skia: aar state post 70, do nothing

> Configure project :expo

Using expo modules
  - expo-asset (11.0.1)
  - expo-constants (17.0.3)
  - expo-crypto (14.0.1)
  - expo-dev-client (5.0.8)
  - expo-dev-launcher (5.0.21)
  - expo-dev-menu (6.0.15)
  - expo-file-system (18.0.6)
  - expo-font (13.0.2)
  - expo-image-loader (5.0.0)
  - expo-image-manipulator (13.0.5)
  - expo-json-utils (0.14.0)
  - expo-keep-awake (14.0.1)
  - expo-linear-gradient (14.0.1)
  - expo-manifests (0.15.4)
  - expo-modules-core (2.1.2)

> Configure project :react-native-reanimated
Android gradle plugin: 8.6.0
Gradle: 8.10.2

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at /Users/agoransson/Projects/ReactNativeProject/my-app/node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core found, Frame Processors are enabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-vision-camera:generateCodegenSchemaFromJavaScript
No modules to process in combine-js-to-schema-cli. If this is unexpected, please check if you set up your NativeComponent correctly. See combine-js-to-schema.js for how codegen finds modules.
The Daemon will expire after the build after running out of JVM Metaspace.
The project memory settings are likely not configured or are configured to an insufficient value.
The daemon will restart for the next build, which may increase subsequent build times.
These settings can be adjusted by setting 'org.gradle.jvmargs' in 'gradle.properties'.
The currently configured max heap space is '2 GiB' and the configured max metaspace is '512 MiB'.
For more information on how to set these values, please refer to https://docs.gradle.org/8.10.2/userguide/build_environment.html#sec:configuring_jvm_memory in the Gradle documentation.
To disable this warning, set 'org.gradle.daemon.performance.disable-logging=true'.
Daemon will be stopped at the end of the build after running out of JVM Metaspace

[... 200 lines cut ...]
```

### VisionCamera Version

4.6.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

agoransson, 2025-01-10:
Yes Hans. My usual attempts include cleaning node_modules, running expo-doctor, checking installed versions (`expo install --check`). Sometimes it starts to build again, but I can't find the combination of steps that _always_ makes it work again.

It still seems random how I manage to get it building again. Tonight I managed by cleaning the gradle cache and removing node_modules. Earlier it worked from removing node_modules. Some days ago I think the final step was running `expo install --check`.

Kickbykick, 2025-03-10:
Hi there, did you find a fix for this

agoransson, 2025-03-10:
Not yet, I ended up not using this library for my project. It seems there's some incompatibility with some packages - sometimes it works, with other it just won't work for some reason.

ebrimasamba, 2025-03-12:
Hi @agoransson am facing the same issue, any fix for this???

billel-boudchicha, 2025-04-02:
hi guys , my workaround is to delete the android/.gradle , then build : npx expo run:android. it happen every time and install a new library or changing the native code , android only.

hrjaffery, 2025-06-29:
Facing the same. I follow the following steps

npm cache clean --force
rm -rf node_modules ios android
npx expo prebuild --clean
npx expo run:ios -d