margelo/react-native-vision-camera #3362: 🐛

maxximee, 2025-01-13:
### What's happening?

Using frame processor shows black image on emulator and some devices (can't reproduce the real devices, read it from android play store reviews).

With frameProcessor
[image]

Without frameProcessor
[image]

Note: The image is black when negative is false and white when negative is true, meaning the shader is applied correctly, but the image isn't displayed.  Also relevant, the resulting image taken is correct and shows the real picture. It's just the live preview with frame processor that isn't working.

### Reproduceable Code

```tsx
using following shader:

const invertColorsFilter = Skia.RuntimeEffect.Make(`
 uniform shader image;
 half4 main(vec2 pos) {
   vec4 color = image.eval(pos);
   return vec4((1.0 - color).rgb, 1.0);
 }
`)
const shaderBuilder = Skia.RuntimeShaderBuilder(invertColorsFilter)
const imageFilter = Skia.ImageFilter.MakeRuntimeShader(shaderBuilder, null, null)
const paint = Skia.Paint()
const defaultPaint = Skia.Paint()
paint.setImageFilter(imageFilter)
```

set here:
```
    const frameProcessor = useSkiaFrameProcessor(frame => {
        'worklet'
        if (negativeColors) {
            frame.render(paint)
        } else {
            frame.render(defaultPaint)
        }
    }, [paint, defaultPaint, negativeColors])
```

to my component:

```
 <ReanimatedCamera
                                        style={StyleSheet.absoluteFill}
                                        device={device}
                                        isActive={isActive}
                                        ref={camera}
                                        onInitialized={onInitialized}
                                        onError={onError}
                                        onStarted={() => console.log('Camera started!')}
                                        onStopped={() => console.log('Camera stopped!')}
                                        onPreviewStarted={() => console.log('Preview started!')}
                                        onPreviewStopped={() => console.log('Preview stopped!')}
                                        onOutputOrientationChanged={(o) => console.log(`Output orientation changed to ${o}!`)}
                                        onPreviewOrientationChanged={(o) => console.log(`Preview orientation changed to ${o}!`)}
                                        onUIRotationChanged={(degrees) => console.log(`UI Rotation changed: ${degrees}°`)}
                                        format={format}
                                        fps={30}
                                        photoHdr={photoHdr}
                                        photoQualityBalance="speed"
                                        lowLightBoost={device.supportsLowLightBoost}
                                        enableZoomGesture={false}
                                        animatedProps={cameraAnimatedProps}
                                        exposure={0}
                                        enableFpsGraph={false}
                                        outputOrientation="device"
                                        photo={true}
                                        video={false}
                                        audio={false}
                                        enableLocation={false}
                                        frameProcessor={frameProcessor}
                                    />
```
```

### Relevant log output

```shell
no errors in the logs:
 LOG  Preview started!
 LOG  Camera: 10 (BACK) androidx.camera.camera2 | Format: (1280x720 photo) | Lens: ultra-wide-angle-camera (1/1)
 LOG  Camera: 10 (BACK) androidx.camera.camera2 | Format: (1280x720 photo)
 LOG  Camera initialized!
 LOG  Camera started!
 LOG  Preview orientation changed to portrait!
 LOG  UI Rotation changed: 0°
 LOG  Output orientation changed to portrait!
 LOG  Camera stopped!
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 1,
  "minZoom": 1,
  "maxExposure": 6,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -6,
  "name": "10 (BACK) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```

### Device

android emulator with API 35 and "VirturalScene"

### VisionCamera Version

4.6.3 with skia 1.7.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

maxximee, 2025-01-13:
@mrousavy updated the issue with relevant adb logs pointing to the exception

maxximee, 2025-01-15:
@mrousavy According wcandillon fromSkia it's an issue related to devices who are using Vulkan and Angle for OpenGL support, including a lot of samsung devices. It was fixed in skia 1.7.7, maybe the same fix could be applied here?

mrousavy, 2025-01-15:
1. Please stop pinging me, I am not free support.
2. Skia is a `*` peerDependency - it is not pinned to a version. If you install 1.7.7., VisionCamera will use Skia 1.7.7. - not 1.4.2.

cleber-dev, 2025-01-28:
Has anyone managed to solve this problem?

luicfrr, 2025-01-30:
In some tests I made black screen is related to canvas translation:
https://github.com/mrousavy/react-native-vision-camera/blob/b194ef17902d8702f584b8f0a5dee0f5c96fceb1/package/src/skia/useSkiaFrameProcessor.ts#L98

If I use `canvas.translate(0, frame.width)` black screen is solved but preview is inverted (left is right and right is left).
Also, drawing on frame is buggy too because seems the frame is -90deg rotated while on IOS it works fine

maxximee, 2025-02-10:
Nothing works on my side, also I logged and am getting 'landscape-right' by default, not landscape-left. 
Black screen on some samsung devices like Pixel 9 with api 35 and crash on samsung s8.

seppemp, 2025-06-05:
I also experience crashes using the Pixel 7 Pro and 6 with API 35 and frame processors, but I haven't been able to resolve the issue. Has something worked for anyone?

maxximee, 2025-06-06:
> I also experience crashes using the Pixel 7 Pro and 6 with API 35 and frame processors, but I haven't been able to resolve the issue. Has something worked for anyone?

The issue is still there, lots of issues with skia frame processor (useSkiaFrameProcessor). The bug is open in the rn skia side, but hasn't been looked at yet.