margelo/react-native-vision-camera #3365: 🐛 Camera Screen goes black when using Skia Frame Processors (doesn't happen with normal frame processors)

DragonDare, 2025-01-14:
### What's happening?

I'm trying to implement pose detection by drawing stick-figure of the pose on my app. But with Skia Frame processor, the camera screen goes black with only the stick figure being visible. However, if i use a normal frame processor, the camera is visible and not black.

Packages:
1. "react-native-fast-tflite": "^1.5.1",
2. "@shopify/react-native-skia": "1.5.0",
3. "react-native-worklets-core": "^1.5.0",
4. "vision-camera-resize-plugin": "^3.2.0"

[image]

### Reproduceable Code

```tsx
import TopAppBar from '@/components/TopAppBar';
import { SlamdunkTheme } from '@/constants/Theme';
import { scaleHeight, scaleWidth } from '@/utils/scaling';
import React, { useEffect, useMemo } from 'react';
import { StyleSheet, View, Text, Platform } from 'react-native';
import { SafeAreaView } from 'react-native-safe-area-context';
import { Camera, useCameraDevice, useCameraPermission, useSkiaFrameProcessor } from 'react-native-vision-camera';
import { useResizePlugin } from 'vision-camera-resize-plugin';
import { TensorflowModel, useTensorflowModel } from 'react-native-fast-tflite';
import { PaintStyle, Skia } from '@shopify/react-native-skia';
import getBestFormat from './formatFilter';

function tensorToString(tensor: TensorflowModel['inputs'][number]): string {
  return `${tensor.dataType} [${tensor.shape}]`;
}

const LINE_WIDTH = 1;
const MIN_CONFIDENCE = 0.3;

const VIEW_WIDTH = scaleWidth(375);

const CameraScreen = () => {
  const device = useCameraDevice('back');

  const { hasPermission, requestPermission } = useCameraPermission();

  useEffect(() => {
    if (!hasPermission) {
      requestPermission().catch((err) => {
        console.error('Camera permission request failed:', err);
      });
    }
  }, [hasPermission, requestPermission]);

  if (!device) {
    return <Text>Loading camera...</Text>;
  }
  
  const { resize } = useResizePlugin();

  const delegate = Platform.OS === 'ios' ? 'core-ml' : undefined;
  const plugin = useTensorflowModel(
    require('../assets/models/singlepose-lightning-tflite-float16.tflite'),
    delegate,
  );
  const format = useMemo(
    () => (device != null ? getBestFormat(device, 720, 1000) : undefined),
    [device],
  );
  console.log(format?.videoWidth, format?.videoHeight);

  const pixelFormat = Platform.OS === 'ios' ? 'rgb' : 'yuv';

  useEffect(() => {
    const model = plugin.model;
[... 575 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Motorola Edge 40

### VisionCamera Version

^4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

maxximee, 2025-01-15:
Having same issue, I think a lot of people are. I think react-native-vision-camera should update its skia dependency from current 1.4.2 to at least 1.7.4 where this issue was addressed, ideally > 1.7.7

DragonDare, 2025-01-15:
> Having same issue, I think a lot of people are. I think react-native-vision-camera should update its skia dependency from current 1.4.2 to at least 1.7.4 where this issue was addressed, ideally > 1.7.7

@mrousavy  Important.

DragonDare, 2025-01-15:
@maxximee have u tried on 1.7.7 yet as @wcandillon advised?

maxximee, 2025-01-15:
> @maxximee have u tried on 1.7.7 yet as @wcandillon advised?

I've forked the project and updated skia dependency, but having hard time compiling it

```
  ld.lld: error: undefined symbol: typeinfo for RNWorklet::JsiHostObject
  >>> referenced by JVisionCameraProxy.cpp
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(typeinfo for RNWorklet::JsiWorklet)
  clang++: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.
```

DragonDare, 2025-01-15:
> I've forked the project and updated skia dependency, but having hard time compiling it
>
> ```
>   ld.lld: error: undefined symbol: typeinfo for RNWorklet::JsiHostObject
>   clang++: error: linker command [...]

i just tried with 1.7.7, doesnt change anything. Same problem.

wcandillon, 2025-01-15:
@mrousavy should we have a look? On our side things are pretty stable and we have new capabilities related to texture sharing with third-party, maybe we can do a refresh

mrousavy, 2025-01-15:
Hmm, maybe it's an orientation thing? Maybe it tries to render in a wrong orientation? In that case, it'd be the fault of the CameraX implementation on this specific android device reporting a wrong orientation. You can play around with this switch in the VisionCamera useSkiaFrameProcessor code: https://github.com/mrousavy/react-native-vision-camera/blob/54df86ace1716039f690e865c97f29f87dfe43b7/package/src/skia/useSkiaFrameProcessor.ts#L91-L113

If none of those options render properly, it migth actually be a texture issue.

DragonDare, 2025-01-15:
> Hmm, maybe it's an orientation thing? Maybe it tries to render in a wrong orientation? In that case, it'd be the fault of the CameraX implementation on this specific android device reporting a wrong [...]

But will the orientation issue result in the camera screen turning to black?

maxximee, 2025-01-15:
> Hmm, maybe it's an orientation thing? Maybe it tries to render in a wrong orientation? In that case, it'd be the fault of the CameraX implementation on this specific android device reporting a wrong [...]

I've found this patch in another post, tried it but doesn't resolve the black screen.
@mrousavy  I think it's what has been fixed in here: https://github.com/Shopify/react-native-skia/pull/2801 . vision-camera still uses skia 1.4.2

mrousavy, 2025-01-15:
> vision-camera still uses skia 1.4.2

No. VisionCamera has a peerDependency on Skia which is set to `*` - that means whatever version of RN Skia you install, will be used. If you install 1.7.7. in your app, it will use 1.7.7.

mrousavy, 2025-01-15:
> But will the orientation issue result in the camera screen turning to black?

It could be, because if the wrong orientation is rendered, the canvas is translated and rotated by the wrong amount, resulting the Frame being drawn outside the visible bounds. Thats why I'm asking you to test this.

maxximee, 2025-01-15:
> > But will the orientation issue result in the camera screen turning to black?
> 
> It could be, because if the wrong orientation is rendered, the canvas is translated and rotated by the wrong amount, resulting the Frame being drawn outside the visible bounds. Thats why I'm asking you to test this.

Didn't help with the black screen. Note on samsung galaxy s22 / s24 it causes thousands of null pointer crashes:
[image]

wcandillon, 2025-01-15:
the crashes on samsung galaxy s22 / s24  has been fixed right?

maxximee, 2025-01-15:
> the crashes on samsung galaxy s22 / s24 has been fixed right?

in skia > 1.7.4 yes. But vision-camera uses skia 1.4.2 so it's still there

DragonDare, 2025-01-15:
> > the crashes on samsung galaxy s22 / s24 has been fixed right?
> 
> In skia > 1.7.4 yes. But vision-camera uses skia 1.4.2 so it's still there.
> 
> I can use skia in my app, but "useSkiaFrameProcessor" in the vision-camera crashes

> > vision-camera still uses skia 1.4.2
> 
> No. VisionCamera has a peerDependency on Skia which is set to `*` - that means whatever version of RN Skia you install, will be used. If you install 1.7.7. in your app, it will use 1.7.7.

mrousavy, 2025-01-15:
> But vision-camera uses skia 1.4.2 so it's still there

I just told you - you can use whatever Skia version you want with VisionCamera. It doesn't use 1.4.2.

DragonDare, 2025-01-15:
> Hmm, maybe it's an orientation thing? Maybe it tries to render in a wrong orientation? In that case, it'd be the fault of the CameraX implementation on this specific android device reporting a wrong [...]

So my device was reporting orientation as 'landscape-right' and I played around with that switch statement as you said. As in put other orientations' cases' in landscape-right's switch but all it did was either rotate my stick figure or not display it at all. The Camera screen is still black tho.

Another interesting observation, [in your Skia Vision Camera Demo github repo](https://mrousavy.com/blog/VisionCamera-Pose-Detection-TFLite), is showing my device orientation to be landscape-left, instead of landscape-right, like in my own project. Does this have to do with the older version of vision camera + skia in your repo?

DragonDare, 2025-01-16:
@mrousavy @wcandillon any updates on this issue?

whalemare, 2025-01-16:
We discovered same issue after upgrading react-native, vision-camera, skia, worklet-core to the latest versions.
We don't use `useSkiaFrameProcessor` at all, but using `codeScanner` prop

This setup is showing white screen on preview:
```
    "react-native": "0.76.6",
    "@shopify/react-native-skia": "1.5.0",
    "react-native-worklets-core": "1.5.0",
    "react-native-vision-camera": "4.6.3",
```

This one (from your example) works perfectly:
```
    "react-native": "^0.75.4",
    "react-native-worklets-core": "^1.3.3",
    "@shopify/react-native-skia": "^1.4.2",
    "react-native-vision-camera": "4.6.3",
```

I can do any action if you need, like logs, device info, screenshots, reproducible demo etc.

usamaabutt, 2025-01-16:
I am getting a black screen error on the Android `RedMe 10C` device while using the Skia frame processor. I am getting this error in the onError callback from the vision Camera tag.

> session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.

[image]

DragonDare, 2025-01-17:
> In my case SkiaFrameProcessor is reporting 'landscape-left' when my phone is in 'portrait' orientation. I noticed that the black screen dissapeared when I rotates the phone, and modifying the canvas translation parameters for 'landscape-left' switch case did the trick:
> 
> ```
> case 'landscape-left':
>         // rotate two flips on (0,0) origin and move X + Y into view again
>         canvas.translate(0, frame.width)
>         canvas.rotate(270, 0, 0)
>         break
> ```

on what basis did you do the changes? What were the calculations involved?

maxximee, 2025-01-18:
> on what basis did you do the changes? What were the calculations involved?

I've seen this patch on a few other github issues, tried it but didn't fix black screen for me.

DragonDare, 2025-01-18:
> I've seen this patch on a few other github issues, tried it but didn't fix black screen for me.

same, reversing the translation and rotation didnt remove the black screen for me

DragonDare, 2025-01-18:
@mrousavy @wcandillon could we please get any updates on this issue? Is that something which is worked on for the next package version or it won't be? I'm running this on a production app, it's affecting us quite a lot.

mrousavy, 2025-01-18:
> [@mrousavy](https://github.com/mrousavy) [@wcandillon](https://github.com/wcandillon) could we please get any updates on this issue? Is that something which is worked on for the next package version or it won't be? I'm running this on a production app, it's affecting us quite a lot.

@DragonDare if you're running this in a production app and your business depends on this, maybe you should consider supporting me or William through GitHub sponsors or finding an enterprise support plan that works for you.

I am not working on a fix for this in the next few days, I suggested what I thought could be the issue (rotation switch/case, try playing with that!), but I don't have time to work on this as my priority is focused towards Nitro right now.
If anyone finds a fix (or wants to pay me to fix this) , we can find a solution that works for everyone :)

mrousavy, 2025-01-18:
Also - we don't know yet if it is a RN Skia or a VisionCamera bug. We'd need to figure out if the texture (SkImage) is being created successfully - if not, it's a Skia bug, if yes it's a VisionCamera bug (because apparently we don't render it correctly later on)

DragonDare, 2025-01-18:
> everyone

I see. I'll surely look into if my company can hop on one of the enterprise plans. I understand why it won't be your priority right now.

Also, I did play with the orientation switch cases, it didn't help except just rotate my stick figure to the correct position, which is portrait. Earlier it was showing in landscape-right even if I was holding phone in portrait. Still getting the black screen nonetheless.

maxximee, 2025-01-19:
My productive app is free and not bringing me anything, so I don't have any money to put in the plugin sadly. I did add a very ugly workaround for android, basically disabling the skia frame processor by default, that means users won't see negative films in the correct colors but it's better than a black screen.
I hope to find time to see if I can debug the plugin to find the source of the error, but from what I see it's something to do with " Failed to convert NativeBuffer to SkImage!"

[image]

mrousavy, 2025-01-20:
> but from what I see it's something to do with " Failed to convert NativeBuffer to SkImage!". 

Oh I didn't see this - well that's definitely coming from react-native-skia, see [`MakeImageFromNativeBuffer(...)`](https://github.com/mrousavy/react-native-skia/blob/02d3e1343f9b3d9f2baf6218e0a8f6b10b6fefc8/packages/skia/cpp/api/JsiSkImageFactory.h#L34-L37)

DragonDare, 2025-01-23:
Guys @mrousavy @wcandillon @maxximee 

I just realised I didn't add the "frame.render()"  in my skia frame processor. This fixed the issue. The Camera is visible now.

```
  const frameProcessor = useSkiaFrameProcessor((frame) => {
    'worklet';

    **frame.render();**
```

I don't know why I'm just coming across this. https://react-native-vision-camera.com/docs/guides/skia-frame-processors.
Completely missed this when doing my initial research.

mrousavy, 2025-01-23:
lol ok no worries happens to the best of us 😂

closed 2025-01-23 by mrousavy