margelo/react-native-vision-camera #3367: 🐛 Barcode scanner returns different barcodes

EmanuilM, 2025-01-14:
### What's happening?

When using the barcode scanner, it occasionally returns different barcodes for the same item when scanned multiple times. For example, scanning a specific barcode from an image may sometimes result in a different value being returned.

The issue is inconsistent and difficult to reproduce reliably. It may occur on the second scan, or sometimes after 5-6 scans or more.

This behavior has been observed on both iOS and Android devices, although it occurs less frequently on iOS.
Also i have noticed that the different barcode is of type `upc-a`

[image]

### Reproduceable Code

```tsx
Here's a sample of how my component looks like

const Camera = () => {
    const { hasPermission, requestPermission } = useCameraPermission();

    const [cameraState] = useState<"back" | "front">("back");

    const codeScanner: CodeScanner = {
        codeTypes: [
            'ean-13',
        ],
        onCodeScanned: (codes) => {
            // My logic here
        }
    }

    const device: any = useCameraDevice(cameraState);
    const [cameraDevice, setCameraDevice] = useState<CameraDevice>(device);

    useEffect(() => {
        setCameraDevice(device);
    }, [])

    if (!device) {
        return (
            <SafeAreaView style={{ flex: 1, backgroundColor: "black" }}>
                <Container style={{ backgroundColor: "black" }}>
                    <FontAwesomeIcon icon={faTriangleExclamation} color={"white"} size={25} />
                    <Text>Failed to configure the camera device.</Text>
                    <Button>
                        <ButtonText>Go back</ButtonText>
                    </Button>
                </Container>
            </SafeAreaView>
        )
    }

    return (
        <GestureHandlerRootView style={{ flex: 1, backgroundColor: "black" }}>
            <PinchGestureHandler>
                <View style={{ flex: 1 }}>
                    <VisionCamera
                        ref={cameraRef}
                        photoQualityBalance={"quality"}
                        style={StyleSheet.absoluteFill}
                        device={cameraDevice}
                        isActive={true}
                        photo={true}
                        codeScanner={codeScanner}
                    />
                </View>
            </PinchGestureHandler>
        </GestureHandlerRootView >
    )
}

export default Camera;
```

### Relevant log output

```shell
LOG  codes [{"corners": [[Object], [Object], [Object], [Object]], "frame": {"height": 55, "width": 78, "x": 225, "y": 311}, "type": "ean-13", "value": "7614500010013"}]

 LOG  codes [{"corners": [[Object], [Object], [Object], [Object]], "frame": {"height": 54, "width": 78, "x": 224, "y": 305}, "type": "ean-13", "value": "7614500010013"}]

 LOG  codes [{"corners": [[Object], [Object], [Object], [Object]], "frame": {"height": 61, "width": 81, "x": 218, "y": 299}, "type": "upc-a", "value": "644924010013"}]
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

any android and ios device

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

EmanuilM, 2025-01-15:
@mrousavy in this log you can see that at `2025-01-15 14:05:04.852` it returns different barcode with different type

```text
2025-01-15 14:04:58.355  1237-1237  CameraService           cameraserver                         E     Conflicts with: Device 0, client package com.app.productscan (PID 3618, score 0, state 2)
2025-01-15 14:04:58.478  3618-4026  ReactNativeJS           com.app.productscan                 I  'code', [ { frame: { width: 134, y: 227, height: 92, x: 226 },
                                                                                                        corners: 
                                                                                                         [ { y: 232, x: 226 },
                                                                                                           { y: 227, x: 358 },
                                                                                                           { y: 314, x: 360 },
                                                                                                           { y: 319, x: 227 } ],
                                                                                                        value: '7614500010013',
                                                                                                        type: 'ean-13' } ]
2025-01-15 14:04:58.483  3618-4026  ReactNativeJS           com.app.productscan                 I  Scanning
2025-01-15 14:04:58.488  3618-4210  CameraView              com.app.productscan                 I  invokeOnAverageFpsChanged(0.0)
2025-01-15 14:04:58.495  3618-4026  ReactNativeJS           com.app.productscan                 I  'CODE:', '7614500010013'
2025-01-15 14:04:58.502  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:04:58.643  3618-3635  cvx.productscan         com.app.productscan                 I  NativeAlloc concurrent copying GC freed 79421(4767KB) AllocSpace objects, 14(1124KB) LOS objects, 49% free, 13MB/27MB, paused 97us,57us total 112.155ms
2025-01-15 14:04:58.720  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:04:58.760  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:04:58.762  1543-5124  VibratorManagerService  system_server                        D  vibrate - uid: 11053, opPkg: com.app.productscan, effect: Mono{mEffect=Composed{segments=[Step{amplitude=0.0, frequencyHz=0.0, duration=0}, Step{amplitude=0.19607843, frequencyHz=0.0, duration=40}, Step{amplitude=0.0, frequencyHz=0.0, duration=100}, Step{amplitude=0.23529412, frequencyHz=0.0, duration=40}], repeat=-1, mMagnitudeType=TYPE_EXTRA}}, attrs: VibrationAttributes: Usage=UNKNOWN Audio Usage= USAGE_UNKNOWN tags= Flags=0, reason: null, token: android.os.BinderProxy@1360e9c
2025-01-15 14:04:58.788  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:04:59.083  3618-3618  Compatibil...geReporter com.app.productscan                 D  Compat change id reported: 171228096; UID 11053; state: ENABLED
2025-01-15 14:04:59.094  3618-3618  ScrollView              com.app.productscan                 D  initGoToTop
2025-01-15 14:04:59.097  3618-3618  cvx.productscan         com.app.productscan                 W  Accessing hidden field Landroid/widget/ScrollView;->mScroller:Landroid/widget/OverScroller; (unsupported, reflection, allowed)
2025-01-15 14:04:59.171  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:04:59.399  3618-3618  ScrollView              com.app.productscan                 D  initGoToTop
2025-01-15 14:04:59.423  3618-3634  cvx.productscan         com.app.productscan                 I  Compiler allocated 4177KB to compile void android.widget.TextView.<init>(android.content.Context, android.util.AttributeSet, int, int)
2025-01-15 14:04:59.426  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:04:59.487  3618-4210  CameraView              com.app.productscan                 I  invokeOnAverageFpsChanged(0.0)
2025-01-15 14:04:59.647  3618-3618  ScrollView              com.app.productscan                 D  initGoToTop
2025-01-15 14:04:59.721  3618-3728  TrafficStats            com.app.productscan                 D  tagSocket(111) with statsTag=0xffffffff, statsUid=-1
2025-01-15 14:04:59.792  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:05:00.487  3618-4210  CameraView              com.app.productscan                 I  invokeOnAverageFpsChanged(0.0)
2025-01-15 14:05:01.365  1237-1237  CameraService           cameraserver                         E     Conflicts with: Device 0, client package com.app.productscan (PID 3618, score 0, state 2)
2025-01-15 14:05:01.487  3618-4210  CameraView              com.app.productscan                 I  invokeOnAverageFpsChanged(0.0)
2025-01-15 14:05:01.737  3618-3618  ViewRootIm...nActivity] com.app.productscan                 I  ViewPostIme pointer 0
2025-01-15 14:05:01.784  3618-3618  GestureDetector         com.app.productscan                 I  obtain mCurrentDownEvent. id: 1000438133 caller: com.swmansion.gesturehandler.core.ScaleGestureDetector.onTouchEvent:221 com.swmansion.gesturehandler.core.PinchGestureHandler.onHandle:67 com.swmansion.gesturehandler. [... cut ...]
2025-01-15 14:05:06.689  3618-3618  GestureDetector         com.app.productscan                 I  obtain mCurrentMotionEventRaw. action: 2 id: 318304691
2025-01-15 14:05:06.749  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:05:06.850  3618-4026  ReactNativeJS           com.app.productscan                 I  'Daily scans:', 0
2025-01-15 14:05:06.866  3618-3618  GestureDetector         com.app.productscan                 I  handleMessage LONG_PRESS
```

xulihang, 2025-01-20:
This is related to the barcode scanning library.

You can filter out misrecognized results by comparing multiple reads.

You can also use a frame processor plugin which uses a more advanced library. Here is a demo app comparing different libraries: https://github.com/tony-xlh/react-native-zxing-mlkit-dynamsoft

EmanuilM, 2025-01-20:
What do you mean by "comparing multiple reads"? What if on the first scan returns the wrong barcode? The UX of the app will be bad

xulihang, 2025-01-20:
For example, you get the following five reads in two seconds:

* 644924010013 
* 7614500010013
* 7614500010013
* 7614500010013
* 7614500010013

You can filter out the different ones.

closed 2025-01-20 by mrousavy

EmanuilM, 2025-01-20:
I will try this method.

@mrousavy Can you please share more details about this issue? Is this something related to native code in Kotlin?

Why should i do this filter it does not make any sense