margelo/react-native-vision-camera #3368: error: package com.mrousavy.camera.frameprocessor does not exist

sumreenali-servionsoft, 2025-01-16:
### How were you trying to build the app?

 "react-native-vision-camera": "^4.6.3",
 "react-native-worklets-core": "^1.5.0",
 "vision-camera-face-detector": "^0.1.8"
 
 
 
 i have installed these packages through npm install 
tried following solutions

1. clean project through  **npx react-native-clean-project**   
2.  running these commands 
brew uninstall --cask zulu11                            
brew install --cask adoptopenjdk/openjdk/adoptopenjdk11
3. cleaning gradlew and build watchman 
4. restarting metro bundler

5. i tried adding these dependencies in build.gradle
    implementation 'com.google.mlkit:face-detection:16.1.5'

    // ML Kit Vision Dependencies
    implementation 'com.google.mlkit:vision-common:17.3.0'

    // Worklets Core (if not already installed)
    implementation project(':react-native-worklets-core')

    // CameraX (used by react-native-vision-camera)
    implementation "androidx.camera:camera-core:1.1.0"
    implementation "androidx.camera:camera-camera2:1.1.0"
    implementation "androidx.camera:camera-lifecycle:1.1.0"
    implementation "androidx.camera:camera-view:1.1.0"

and in settings.gradle
include ':react-native-vision-camera'
project(':react-native-vision-camera').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-vision-camera/android')

my babel.config.js file

module.exports = function(api) {
  if (api.env("production")) {
    return {
      "presets": ["module:@react-native/babel-preset"],
      "plugins": [
        // "transform-flow-strip-types",
        "transform-remove-console",
        "react-native-reanimated/plugin"
      ]
    }
  }

  return {
    "presets": ["module:@react-native/babel-preset"],
    "plugins": [
      // "transform-flow-strip-types",
      "react-native-reanimated/plugin"
    ]
  }
};

this is my android/build.gradle file

buildscript {
    ext {
        buildToolsVersion = "35.0.0"
        minSdkVersion = 24
        compileSdkVersion = 34
        targetSdkVersion = 34
        ndkVersion = "26.1.10909125"
        kotlinVersion = "1.9.24"
[... 94 lines cut ...]
```

### VisionCamera Version

4.6.3

### Target platforms

iOS, Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sumreenali-servionsoft, 2025-01-16:
vision camera works fine but when I installed vision-camera-face-detection and react-native-worklets-core it started giving me above errors

theambidextrous, 2025-04-20:
the package name in the latest version is `frameprocessors`

thanhlucnguyen, 2025-06-19:
Let try with
`import com.mrousavy.camera.frameprocessors....`

dztran, 2025-06-25:
Has anyone got this to work? It does not matter what plugins I used, I got the same error messages when building my app with: 

CodeScannerProcessorPlugin.java:20: error: package com.mrousavy.camera.frameprocessor does not exist
import com.mrousavy.camera.frameprocessor.Frame;

I see the java codes for com.mrousavy.camera.frameprocessor in the app's node_modules directory, so the code is there, it is just a matter of telling gradle to look in there when to compile. I just can't figure out gradle after may different tries.