margelo/react-native-vision-camera #3372: 🔧 A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction

CarterSnich, 2025-01-19:
### How were you trying to build the app?

I tried to install, prebuild and run with the following:
```
npx expo install react-native-vision-camera
npx expo prebuild
npx expo run:android
```

I somehow got a different error (different from what I have provided below) after I changed `kotlinVersion = '1.8.0'` on `android/build.gradle` but I can't reproduce the different error anymore.

Furthermore, I'm getting errors when building the example app. Android Studio shows errors about a package that leads to node_modules. But, `npm install` shows dependency errors.

I don't know what to do anymore.

### Full build logs

```tsx
$ npx expo prebuild --clean
Warning! Your git working tree is dirty.
It's recommended to commit all your changes before proceeding, so you can revert the changes made by this command if necessary.

√ Would you like to proceed? ... yes

✔ Cleared android code
✔ Created native directory
✔ Updated package.json | no changes
» android: androidStatusBar.backgroundColor: Color conflicts with the splash.backgroundColor
» android: userInterfaceStyle: Install expo-system-ui in your project to enable this feature.
✔ Finished prebuild

Aia@LAPTOP-VC332DRM MINGW64 ~/Desktop/revel (master)
$ npx expo run:android
› Building app...
Configuration on demand is an incubating feature.

> Configure project :app
 ??  Applying gradle plugin 'expo-dev-launcher-gradle-plugin' (expo-dev-launcher@3.6.10)
 ??  Applying gradle plugin 'expo-updates-gradle-plugin' (expo-updates@0.24.13)

> Configure project :react-native-firebase_app
:react-native-firebase_app package.json found at C:\Users\Aia\Desktop\revel\node_modules\@react-native-firebase\app\package.json
:react-native-firebase_app:firebase.bom using default value: 33.7.0
:react-native-firebase_app:play.play-services-auth using default value: 21.2.0
:react-native-firebase_app package.json found at C:\Users\Aia\Desktop\revel\node_modules\@react-native-firebase\app\package.json
:react-native-firebase_app:version set from package.json: 21.7.0 (21,7,0 - 21007000)
:react-native-firebase_app:android.compileSdk using custom value: 34
:react-native-firebase_app:android.targetSdk using custom value: 34
:react-native-firebase_app:android.minSdk using custom value: 23
:react-native-firebase_app:reactNativeAndroidDir C:\Users\Aia\Desktop\revel\node_modules\react-native\android  

> Configure project :expo

Using expo modules
  - expo-av (13.10.6)
  - expo-camera (14.1.3)
  - expo-constants (15.4.6)
  - expo-dev-client (3.3.12)
  - expo-dev-launcher (3.6.10)
  - expo-dev-menu (4.5.8)
  - expo-eas-client (0.11.2)
  - expo-file-system (16.0.9)
  - expo-font (11.10.3)
  - expo-image (1.10.6)
  - expo-image-loader (4.6.0)
  - expo-image-picker (14.7.1)
  - expo-json-utils (0.12.3)
  - expo-keep-awake (12.8.2)
  - expo-linear-gradient (12.7.2)
  - expo-manifests (0.13.2)
[... 311 lines cut ...]
```

### VisionCamera Version

4.3.2

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [x] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

CarterSnich, 2025-01-20:
> Please double-check your Kotlin version in `android/build.gradle`. You currently have `kotlinVersion = '1.8.0'`, but the error logs indicate incompatibility with classes compiled using Kotlin version `2.0.0`. Try updating your Kotlin version or resolving the version conflicts in your dependencies.

Already tried changing `kotlinVersion ` to `1.8.0` and `2.0.0`, nothing changes. It seems like it is ignored when running `npx expo run:android`.

erenkulaksiz, 2025-01-20:
Hello, i finally temporarly solved the issue after immense hours of debug and I want to share my solution.
First, lets remove gradle alltogether (i had cache issues):
`sudo rm -rf ~/.gradle`
Then remove node modules on repo `rm -rf node_modules`
install node modules `npm install`
Downgrade vision camera library to version 4.3.2 as im using react-native version of 0.73.6
`npm i react-native-vision-camera@4.3.2`
if using expo, clean prebuild `npx expo prebuild --platform=android --clean` you can also run `./gradlew clean` on ./android folder to make sure no caches present
then clear metro cache and start the app `sudo rm -rf $TMPDIR/metro-cache` `npx expo run:android`

Hope this helps!

CarterSnich, 2025-01-21:
> Hello, i finally temporarly solved the issue after immense hours of debug and I want to share my solution. First, lets remove gradle alltogether (i had cache issues): `sudo rm -rf ~/.gradle` Then [...]

That worked! I think removing `~/.gradle` did it.

mysport12, 2025-01-23:
This patch helped me get past the build error. Kotlin version 2.0.21 and RN 0.77.0

```
diff --git a/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt b/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
index 862c03a14680566280adfb48dd14b02781844d8d..5a31c764a642c35493fc7b1b1b2e48ae01f2047c 100644
--- a/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
+++ b/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
@@ -90,7 +90,8 @@ class CameraSession(internal val context: Context, internal val callback: Callba
     }
   }
 
-  override fun getLifecycle(): Lifecycle = lifecycleRegistry
+  override val lifecycle: Lifecycle
+    get() = lifecycleRegistry
 
   /**
    * Configures the [CameraSession] with new values in one batch.
```

reyhanadp, 2025-01-23:
> This patch helped me get past the build error. Kotlin version 2.0.21 and RN 0.77.0
>
> ```
> diff --git a/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt [...]

thank you very much. you saved my time

gkasireddy202, 2025-01-23:
I am facing a build issue.
org.jetbrains.kotlin.gradle.tasks.CompilationErrorException: Compilation error

react-native:0.73.7
react:18.2.0.
react-native-vision-camera:4.6.3

saif482, 2025-02-10:
@gkasireddy202 did you find the solution for this? the patch doesn't work for me

gkasireddy202, 2025-02-10:
@saif482 - Please try with "react-native-vision-camera": "^4.3.2"

saif482, 2025-02-10:
> Hello, i finally temporarly solved the issue after immense hours of debug and I want to share my solution. First, lets remove gradle alltogether (i had cache issues): `sudo rm -rf ~/.gradle` Then [...]

This worked for me, thanks!

velrino, 2025-08-28:
> Hello, i finally temporarly solved the issue after immense hours of debug and I want to share my solution. First, lets remove gradle alltogether (i had cache issues): `sudo rm -rf ~/.gradle` Then [...]

[image]