margelo/react-native-vision-camera #3377: 🐛  Camera recording cannot withstand app backgrounding even when recording is paused

creitz, 2025-01-22:
### What's happening?

I wanted to pause the recording when the user backgrounds the app, and when the app is re-foregrounded, the user has the option to resume the recording. This is related to the recommendation in the documentation at https://react-native-vision-camera.com/docs/guides/lifecycle.

However, once the app is re-foregrounded, I receive a flood of the following error:

`Camera: runtime error [capture/unknown: Frame arrived, but AssetWriter status is failed!]`

I suspect that the issue is caused by the app being backgrounded, and that it seems to be a known issue on iOS that the `AVAssetWriter` cannot withstand app backgrounding:
https://developer.apple.com/forums/thread/672056

I believe the backgrounding behavior is altered by whether the app is being debugged/run from Xcode - this issue is only reproducible when _not_ debugging with Xcode.

### Reproduceable Code

```tsx
The reproducible code can be found in the branch of a fork here: https://github.com/creitz/react-native-vision-camera-example-background-bug/tree/background-bug-demo. I also opened a PR here for easy diff viewing: https://github.com/creitz/react-native-vision-camera-example-background-bug/pull/1

To reproduce the error:
1. Launch the app *but not from xcode*. You can build with xcode and install, but then quit the app and relaunch by tapping the icon.
2. Tap the record button - which now starts a recording session that will not stop until tapped again
3. Send the app to the background
4. Reopen the app, and you should observe the stream of errors
```

### Relevant log output

```shell
ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]
 ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]

These are logs from console - as stated, the bug cannot be reproduced when running from xcode.
```

### Camera Device

```json
{
  "maxZoom": 123.75,
  "position": "back",
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "sensorOrientation": "portrait",
  "minZoom": 1,
  "formats": [],
  "hasTorch": true,
  "minExposure": -8,
  "hasFlash": true,
  "name": "Back Camera",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "supportsRawCapture": false,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "neutralZoom": 1,
  "minFocusDistance": 20,
  "isMultiCam": false,
  "hardwareLevel": "full"
}
```

### Device

iPhone 16 Pro iOS 18.2.1

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [x] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

creitz, 2025-01-22:
>You might want to consider implementing checks that halt the recording when the app is backgrounded and only resume it when foregrounded again, as per the documentation.

Thank you maintenance hans, yes - that's precisely what I'd like to do...pause the recording when the app is backgrounded, and resume when the app is reforegrounded. But actually this error occurs even without resuming recording at all. It occurs just from re-foregrounding the app.

eranga-akva, 2025-03-03:
experiencing the same issue. Setting the 

> isActive

 to 

> false

 and 

> true

when app goes to background and comes to foreground does not seem to work as well.

dnielgonzlz, 2025-03-05:
Any updates on this? Experiencing the same issue.

creitz, 2025-03-07:
The only "workaround" I found was to stop the recording as soon as the app is backgrounded. Not ideal, but at least the video up until that point is not lost.

dnielgonzlz, 2025-03-10:
yeah i've implemented the same temporary solution, thanks!

kculmback-eig, 2025-06-05:
I'm running into this issue as well. When the app state changes to `background` while recording, I get the `ERROR  [capture/unknown: Frame arrived, but AssetWriter status is failed!]` when opening the app back up.

I've tried implementing stopping the recording when the app state changes to `background` (and also tried `inactive`), but upon opening the app I get a different error in the `onRecordingError` function: 

```
{
  "cause": null,
  "code": "capture/unknown",
  "message": "capture/unknown: An unknown recording error occured! -11847 Error Domain=AVFoundationErrorDomain Code=-11847 \"Operation Interrupted\" UserInfo={NSLocalizedDescription=Operation Interrupted, NSLocalizedRecoverySuggestion=Stop other operations and try again.}"
}
```

This is an example of what I've tried doing:
```tsx
useEffect(() => {
  const appStateListener = AppState.addEventListener('change', (state) => {
    // isRecording is a local variable we set to true when recording starts
    if (state === 'background' && isRecording) {
      cameraRef.current?.stopRecording()
    }
  })

  return () => {
    appStateListener.remove()
  }
}, [isRecording])
```

This is in an expo development app; I have not tried it in a non-development app yet.

@creitz or @dnielgonzlz are your workarounds still working? If so, are you able to share any samples on what you did to get it to work?

kculmback-eig, 2025-06-06:
@creitz switching the check to `appState !== 'active'` was the secret sauce that did the trick! Thank you for the suggestion.