margelo/react-native-vision-camera #3379: ✨ How to capture mutiple images from camera

gkasireddy202, 2025-01-23:
### What feature or enhancement are you suggesting?

Feature: I need to capture multiple images at a time and add those images to the array. Could you send that array to the server for a request?

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

The feature is like in WhatsApp or any social media app.

### Additional information

- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

gkasireddy202, 2025-01-23:
Thanks for your response.
Here is the scenario.

If the user continuously presses the camera on the device, all the images are pushed to the array. The user can see those images in the flat list.

The user can send that array to the server request.

mrousavy, 2025-02-17:
This is very simple. read the docs.

closed 2025-02-17 by mrousavy