margelo/react-native-vision-camera #3380: 🔧 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'

mertkahramanturk, 2025-01-24:
### How were you trying to build the app?

npx @react-native-community/cli@latest init AwesomeProject

npm i react-native-vision-camera --save

npm start - npm run android
I created the project with the commands and I get this error during the compilation phase. I tried clearing gardlew and reinstalling node_modules...

### Full build logs

```tsx
> Task :react-native-vision-camera:compileDebugKotlin FAILED

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors: OFF!                                                                                       
48 actionable tasks: 38 executed, 10 up-to-date

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor 

e: file:///C:/Users/mert.kahramanturk/Desktop/test/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'.
e: file:///C:/Users/mert.kahramanturk/Desktop/test/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3 'getLifecycle' overrides nothing.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 15s
error Failed to install the app. Command failed with exit code 1: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081    
e: file:///C:/Users/mert.kahramanturk/Desktop/test/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'. e: file:///C:/Users/mert.kahramanturk/Desktop/test/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3 'getLifecycle' overrides nothing. FAILURE: Build failed with an exception. * What went wrong: Execution failed for task ':react-native-vision-camera:compileDebugKotlin'. > A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction > Compilation error. See log for more details * Try: > Run with --stacktrace option to get the stack trace. > Run with --info or --debug option to get more log output. > Run with --scan to get full insights. > Get more help at https://help.gradle.org. BUILD FAILED in 15s.
```

### Project dependencies

```json
"dependencies": {
    "react": "18.3.1",
    "react-native": "0.77.0",
    "react-native-vision-camera": "^4.6.3"
  },
```

### VisionCamera Version

4.6.3

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zkteco-home, 2025-01-24:
same isssue in RN 0.77

zkteco-home, 2025-01-24:
> Guten Tag, Hans here! 🍻 Thank you for providing detailed information about your issue. It looks like you have encountered a build error related to `CameraSession`.
> 
> Before we can proceed, I'd suggest trying to build the [VisionCamera Example app](https://github.com/mrousavy/react-native-vision-camera/tree/main/example) to see if you experience the same issue there. Additionally, please ensure your development environment is properly set up by running `npx react-native doctor`.
> 
> If the issue persists, kindly check for any typoes in your code or dependencies, and make sure everything is up to date.
> 
> If you find a solution or gather more specific logs, please update your issue here. By the way, if you enjoy our work, consider sponsoring the project [here](https://github.com/sponsors/mrousavy), so we can continue to improve it!
> 
> > Note: If you think I made a mistake, please ping `@mrousavy` to take a look.

your example is just for RN0.75,it just does not work on RN0.77

mertkahramanturk, 2025-01-24:
RN0.76 and RN0.75 also work @zkteco-home

ManojJyaniMudulesSeventeen, 2025-01-24:
node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'.
node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3 'getLifecycle' overrides nothing.

same issue with RN-77

Mhp23, 2025-01-24:
To resolve the build issue, check out this [PR](https://github.com/mrousavy/react-native-vision-camera/pull/3382) or use the following patch. If you're not familiar with patching, refer to [patch-package](https://github.com/ds300/patch-package)

[react-native-vision-camera+4.6.3.patch](https://github.com/user-attachments/files/18541824/react-native-vision-camera%2B4.6.3.patch)

zkteco-home, 2025-01-25:
> RN0.76 and RN0.75 also work [@zkteco-home](https://github.com/zkteco-home)

yes,it works fine on RN0.73 0.74 and 0.75 0.76

gabriel-sisjr, 2025-01-25:
> To resolve the build issue, check out this [PR](https://github.com/mrousavy/react-native-vision-camera/pull/3382) or use the following patch. If you're not familiar with patching, refer to [...]

Works perfectly on RN 0.77.

Audrey-Ann, 2025-01-27:
Can we merge and release this fix/patch please :pray:

> To resolve the build issue, check out this [PR](https://github.com/mrousavy/react-native-vision-camera/pull/3382) or use the following patch. If you're not familiar with patching, refer to [...]

coycoylaniba, 2025-01-29:
is this fixed? im still having this issue on 0.77

gabriel-sisjr, 2025-01-29:
> is this fixed? im still having this issue on 0.77

Not merged yet. You can solve the problem manually or with a patch.

Dallas62, 2025-02-03:
@mrousavy Just ping you because impacting more and more people and PR is available ;)

Ridwan226, 2025-02-04:
> To resolve the build issue, check out this [PR](https://github.com/mrousavy/react-native-vision-camera/pull/3382) or use the following patch. If you're not familiar with patching, refer to [...]

it works

Mohamed-Ziyad, 2025-02-05:
Same issue not working in react-native 0.77 android.

AliRezaBeigy, 2025-02-14:
@mrousavy, could you please check this issue and the suggested solutions? I would greatly appreciate it.

commit 6501b639 (2025-02-14): fix: Fix React Native 77 Android build failure by `LifecycleOwner` (#3394)

Fixes #3380
closed 2025-02-14 by mrousavy