margelo/react-native-vision-camera #3386: 🐛 Zoom is only applied when it reaches maximum value (Android)

L0NG4R3S, 2025-01-28:
### What's happening?

The camera zoom does not work smoothly, it only reaches the maximum or minimum value.

Video: https://streamable.com/deo4ru

### Reproduceable Code

```tsx
import React, { useState } from 'react';
import { Text, View } from 'react-native';
import Animated, {
  Extrapolation,
  interpolate,
  runOnJS,
  useAnimatedProps,
  useSharedValue,
  withTiming,
} from 'react-native-reanimated';
import { GestureDetector, Gesture } from 'react-native-gesture-handler';
import { Camera, useCameraDevice, CameraProps } from 'react-native-vision-camera';

Animated.addWhitelistedNativeProps({
  zoom: true,
});
const AnimatedCamera = Animated.createAnimatedComponent(Camera);

const CameraComponent = () => {
  const device = useCameraDevice('back');
  const zoom = useSharedValue(device.neutralZoom);
  const [showZoom, setShowZoom] = useState(device.neutralZoom);

  console.log(JSON.stringify(device, (k, v) => (k === 'formats' ? [] : v), 2));

  const zoomOffset = useSharedValue(0);
  const gesture = Gesture.Pinch()
    .onBegin(() => {
      zoomOffset.value = zoom.value;
    })
    .onUpdate((event) => {
      const z = zoomOffset.value * event.scale;
      const newZoomValue = interpolate(
        z,
        [1, 10],
        [device.minZoom, device.maxZoom],
        Extrapolation.CLAMP,
      );
      zoom.value = withTiming(newZoomValue, { duration: 10 }); //withTiming avoids animation blinking in android
      runOnJS(setShowZoom)(newZoomValue);
    });

  const animatedProps = useAnimatedProps<CameraProps>(() => ({ zoom: zoom.value }), [zoom.value]);

  if (device == null) {
    return null;
  }

  return (
    <View style={{ flex: 1 }}>
      <GestureDetector gesture={gesture}>
        <AnimatedCamera
          style={{ flex: 1 }}
          device={device}
          isActive={true}
          animatedProps={animatedProps}
        />
      </GestureDetector>
      <Text
        style={{
          color: 'red',
[... 467 lines cut ...]
  "hasFlash": true,
  "maxExposure": 32,
  "name": "0 (BACK) androidx.camera.camera2"
}
```

### Device

Nokia C20, Android 11

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

L0NG4R3S, 2025-01-28:
@main

> Guten Tag, Hans here! 🍻
> 
> Thank you for ze detailed bug report. It seems you have designed a reproducible example and provided relevant log output, which is very helpful.
> 
> However, please ensure that the zoom values you are using fall within ze specified range of your camera device. According to your log, ze `maxZoom` is 4 and the `minZoom` is 1. Make sure your scaling remains within those limits.
> 
> If you're still experiencing issues, perhaps you could try isolating the problem further or testing on a different device. If you believe this is indeed a bug, let's keep it open for mrousavy to investigate!
> 
> Danke and good luck!
> 
> > Note: If you think I made a mistake, please ping `@mrousavy` to take a look.

Hello,

Tthe code works well on other Android devices, but I believe there is a bug in the library for this device or specific characteristics.

Previously using react-native-camera, which was deprecated in favor of this lib, the error did not occur until switching to react-native-vision-camera.

L0NG4R3S, 2025-02-11:
Any follow-up?

sayan-avantai, 2025-07-04:
### So you are right, that's why i have use this strategy 

**To prevent the zoom up-to a certain range **
  ```
const applyZoomToFitQR = frame => {
    const qrWidth = frame.width;
    if (!qrWidth || !device) return;

    const desiredQrWidth = screenWidth * 0.4;
    let requiredZoomFactor = desiredQrWidth / qrWidth;

    const targetZoom = interpolate(
      requiredZoomFactor,
      [1, 3],
      [device.minZoom, device.maxZoom],
      Extrapolation.CLAMP,
    );

    const roundedTargetZoom = Math.round(targetZoom * 10) / 10;
    zoom.value = withTiming(roundedTargetZoom, { duration: 500 });
  };
```