margelo/react-native-vision-camera #3389: 🔧 [React Native 0.77.0] Vision Camera - Build/Runtime Error on Installation

Manav763, 2025-01-30:
### How were you trying to build the app?

I have set up a new React Native project using React Native 0.77.0 and installed react-native-vision-camera. However, when I try to build and run the Android app using the command:
`yarn android` and build fails with an error.

**Steps to Reproduce:**

1. Initialize a new React Native project:
`npx @react-native-community/cli@latest init AwesomeProject --version 0.77.0`

2. Navigate to the project directory:
`cd AwesomeProject`

3. Install react-native-vision-camera:
`yarn add react-native-vision-camera`

4. Attempt to run the app on Android:
`yarn android`

5. The build fails with an error.

**Environment:**

React Native version: 0.77.0
react-native-vision-camera version: 4.6.3
Android Studio version: Android Studio Ladybug | 2024.2.1 Patch 2
Device/Emulator: Tested on Emulator
NodeJs version: 20.18.0
Yarn version: 1.22.22
Java version: 17.0.13

**Additional Context:**
1. I have verified that the setup instructions from the official documentation were followed correctly.
2. I have tried cleaning the project `(cd android && ./gradlew clean && cd ..)` and re-running yarn android, but the issue persists.
3. No modifications have been made to the default React Native project setup.

### Full build logs

```tsx
yarn run v1.22.22
$ react-native run-android
info A dev server is already running for this project on port 8081.
info Installing the app...

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at /Users/abc/workspace/AwesomeProject/node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core not found, Frame Processors are disabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-vision-camera:generateCodegenSchemaFromJavaScript

> Task :react-native-vision-camera:compileDebugKotlin FAILED

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors: OFF!
50 actionable tasks: 50 executed

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor

No modules to process in combine-js-to-schema-cli. If this is unexpected, please check if you set up your NativeComponent correctly. See combine-js-to-schema.js for how codegen finds modules.
e: file:///Users/abc/workspace/AwesomeProject/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'.
e: file:///Users/abc/workspace/AwesomeProject/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3 'getLifecycle' overrides nothing.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 7s
error Failed to install the app. Command failed with exit code 1: ./gradlew app:installDebug -PreactNativeDevServerPort=8081
No modules to process in combine-js-to-schema-cli. If this is unexpected, please check if you set up your NativeComponent correctly. See combine-js-to-schema.js for how codegen finds modules.
e: file:///Users/abc/workspace/AwesomeProject/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'.
e: file:///Users/abc/workspace/AwesomeProject/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3 'getLifecycle' overrides nothing. FAILURE: Build failed with an exception. * What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction > Compilation error. See log for more details * Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org. BUILD FAILED in 7s.
info Run CLI with --verbose flag for more details.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

### Project dependencies

```json
"dependencies": {
    "react": "18.3.1",
    "react-native": "0.77.0",
    "react-native-vision-camera": "^4.6.3"
  },
```

### VisionCamera Version

4.6.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

deepktp, 2025-01-30:
@Manav763 
can you try this 
[https://github.com/mrousavy/react-native-vision-camera/pull/3382/files](https://github.com/mrousavy/react-native-vision-camera/pull/3382/files)

you need to edit this file in node modules

karamanmertcan, 2025-01-30:
> [@Manav763](https://github.com/Manav763) can you try this https://github.com/mrousavy/react-native-vision-camera/pull/3382/files
> 
> you need to edit this file in node modules
> 
> and take a look at this
> 
> [#3380](https://github.com/mrousavy/react-native-vision-camera/issues/3380)

this works, thank you!

closed 2025-01-31 by Manav763