margelo/react-native-vision-camera #3395: ✨ Qr Code distance prompt

sayan-avantai, 2025-02-01:
### What feature or enhancement are you suggesting?

Please add a feature to the QR code scanner that prompts the user with a message like "At least 1 foot of distance is needed to scan the QR code." The required distance should be adjustable dynamically.

### What Platforms whould this feature/enhancement affect?

Android, iOS

### Alternatives/Workarounds

Right now, I am only showing the user a prompt without any distance measuring.

### Additional information

- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

sayan-avantai, 2025-02-01:
Distance= Focal Length×Real QR Size/Apparent QR Size
​

mrousavy, 2025-02-17:
Not going to be supported by QR Code Scanner. This is a perfect use case for custom Frame Processors.