margelo/react-native-vision-camera #3396: [BUILD ERROR] react-native-vision-camera^4.6.3 for "react-native": "^0.77.0"

harshitparkar-ardent, 2025-02-03:
### How were you trying to build the app?

Android build was throwing error in react native as below.

Error resolving plugin [id: 'com.facebook.react.settings']
> A problem occurred configuring project ':gradle-plugin:react-native-gradle-plugin'.
   > java.util.zip.ZipException: zip END header not found

So had to upgrade the react native package to 0.77.0 and that's when react-native-vision-camera throws error during build.

### Full build logs

```tsx
yarn run v1.22.22
$ react-native run-android
info A dev server is already running for this project on port 8081.
info Installing the app...
Configuration on demand is an incubating feature.

> Configure project :app
Reading env from: .env
Reading env from: .env
[react-native-background-geolocation] Purging debug resources in release build

> Configure project :react-native-firebase_app
:react-native-firebase_app package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_app:firebase.bom using default value: 33.7.0
:react-native-firebase_app:play.play-services-auth using default value: 21.2.0
:react-native-firebase_app package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_app:version set from package.json: 21.7.1 (21,7,1 - 21007001)
:react-native-firebase_app:android.compileSdk using custom value: 35
:react-native-firebase_app:android.targetSdk using custom value: 34
:react-native-firebase_app:android.minSdk using custom value: 24
:react-native-firebase_app:reactNativeAndroidDir /Users/harshitparkar/Documents/appName/node_modules/react-native/android

> Configure project :react-native-firebase_auth
:react-native-firebase_auth package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/auth/package.json
:react-native-firebase_app package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_auth:firebase.bom using default value: 33.7.0
:react-native-firebase_auth package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/auth/package.json
:react-native-firebase_auth:version set from package.json: 21.7.1 (21,7,1 - 21007001)
:react-native-firebase_auth:android.compileSdk using custom value: 35
:react-native-firebase_auth:android.targetSdk using custom value: 34
:react-native-firebase_auth:android.minSdk using custom value: 24
:react-native-firebase_auth:reactNativeAndroidDir /Users/harshitparkar/Documents/appName/node_modules/react-native/android

> Configure project :react-native-firebase_crashlytics
:react-native-firebase_crashlytics package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/crashlytics/package.json
:react-native-firebase_app package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_crashlytics:firebase.bom using default value: 33.7.0
:react-native-firebase_crashlytics package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/crashlytics/package.json
:react-native-firebase_crashlytics:version set from package.json: 21.7.1 (21,7,1 - 21007001)
:react-native-firebase_crashlytics:android.compileSdk using custom value: 35
:react-native-firebase_crashlytics:android.targetSdk using custom value: 34
:react-native-firebase_crashlytics:android.minSdk using custom value: 24
:react-native-firebase_crashlytics:reactNativeAndroidDir /Users/harshitparkar/Documents/appName/node_modules/react-native/android

> Configure project :react-native-firebase_database
:react-native-firebase_database package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/database/package.json
:react-native-firebase_app package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_database:firebase.bom using default value: 33.7.0
:react-native-firebase_database package.json found at /Users/harshitparkar/Documents/appName/node_modules/@react-native-firebase/database/package.json
:react-native-firebase_database:version set from package.json: 21.7.1 (21,7,1 - 21007001)
:react-native-firebase_database:android.compileSdk using custom value: 35
:react-native-firebase_database:android.targetSdk using custom value: 34
:react-native-firebase_database:android.minSdk using custom value: 24
:react-native-firebase_database:reactNativeAndroidDir /Users/harshitparkar/Documents/appName/node_modules/react-native/android

> Configure project :react-native-reanimated
Android gradle plugin: 8.7.2
[... 326 lines cut ...]
```

### VisionCamera Version

^4.6.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rishik-simplifix, 2025-02-03:
I'm also facing same issue after upgrading RN 0.76.5 to 0.77.0
I've also tried after clean all caches still issue is persist

rishik-simplifix, 2025-02-03:
I'm also facing the same issue after upgrading RN 0.76.5 to 0.77.0 I've also tried after cleaning all caches still issue is persist.

nilede-raj, 2025-02-03:
https://github.com/mrousavy/react-native-vision-camera/issues/3380#issuecomment-2613375036

https://github.com/mrousavy/react-native-vision-camera/pull/3382 Try this patch.

rishik-simplifix, 2025-02-03:
> [#3380 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/3380#issuecomment-2613375036)
> 
> [#3382](https://github.com/mrousavy/react-native-vision-camera/pull/3382) Try this patch.

It's works for me Thanks @nilede-raj

Sarthak074, 2025-02-04:
When will this issue be fixed :) ?

Villefor, 2025-02-06:
Also got this issue yesterday on RN 0.77

Saqib92, 2025-02-07:
same issue 0.77 was working in 0.76

christian-hess-94, 2025-02-07:
Having a similar issue on 0.76.6. Not react-native-vision-camera:compileDebugKotlin but :react-native-vision-camera:configureCMakeDebug

vaibhavdhake123, 2025-02-10:
Having a similar issue on 0.77.0. Not react-native-vision-camera:compileDebugKotlin but :react-native-vision-camera:configureCMakeDebug

Try all way solution but not fix Iusse

Marcin-Falkowski, 2025-02-18:
1. Install patch-package and postinstall` npm install patch-package postinstall-postinstall`
2. Edit files for react-native-vision-camera in `node_modules `with this commit [6501b63#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b](https://github.com/mrousavy/react-native-vision-camera/commit/6501b6398897031c211ef9e3fac1a207953e0a31#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b)
3. Generate patch `npx patch-package react-native-vision-camera`
4. 
You can also add script to your package.json scripts:

```
"scripts": {
  "postinstall": "patch-package"
}
```

Marcin-Falkowski, 2025-02-18:
@christian-hess-94  @vaibhavdhake123 

Problems with CMakeDebug requires a few checks:

1. Are you sure that you use correct gradle version?
2. Which version Cmake you have?
3. Which Java version you have?
4. Did you check versions of other required dependencies?

Create new Issue with full logs here please.

christian-hess-94, 2025-02-18:
Forgot to update, already solved my issue. I think it was by updating a bunch of other dependencies during the RN upgrade process. It basically solved itself

Marcin-Falkowski, 2025-03-05:
To be honest, I had to disable new architecture for Android when I noticed similar issues with CMake/Codegen. That was only option to create build. 

_gradle.properties_

`newArchEnabled=false`

Marcin-Falkowski, 2025-03-06:
My stack:

RN - 0.77,
Java 17,
Gradle 8.10.2,
Kotlin:  1.9.24,
Groovy:  3.0.22,
React: 18.3.1,
react-native-vision-camera: 4.6.3
node: v20.18.2

_gradle.properties_
```
# Project-wide Gradle settings.

# IDE (e.g. Android Studio) users:
# Gradle settings configured through the IDE *will override*
# any settings specified in this file.

# For more details on how to configure your build environment visit
# http://www.gradle.org/docs/current/userguide/build_environment.html

# Specifies the JVM arguments used for the daemon process.
# The setting is particularly useful for tweaking memory settings.
# Default value: -Xmx512m -XX:MaxMetaspaceSize=256m
org.gradle.jvmargs=-Xmx2048m -XX:MaxMetaspaceSize=512m

# When configured, Gradle will run in incubating parallel mode.
# This option should only be used with decoupled projects. More details, visit
# http://www.gradle.org/docs/current/userguide/multi_project_builds.html#sec:decoupled_projects
# org.gradle.parallel=true

# AndroidX package structure to make it clearer which packages are bundled with the
# Android operating system, and which are packaged with your app's APK
# https://developer.android.com/topic/libraries/support-library/androidx-rn
android.useAndroidX=true
# Automatically convert third-party libraries to use AndroidX
android.enableJetifier=true

# Use this property to specify which architecture you want to build.
# You can also override it from the CLI using
# ./gradlew <task> -PreactNativeArchitectures=x86_64
reactNativeArchitectures=armeabi-v7a,arm64-v8a,x86,x86_64
# Use this property to enable support to the new architecture.
# This will allow you to use TurboModules and the Fabric render in
# your application. You should enable this flag either if you want
# to write custom TurboModules/Fabric components OR use libraries that
# are providing them.
newArchEnabled=false
# Use this property to enable or disable the Hermes JS engine.
# If set to false, you will be using JSC instead.
hermesEnabled=true
VisionCamera_enableCodeScanner=true
```

scripts to install clean and dependencies install:
```
    "clean": "watchman watch-del-all && rm -rf yarn.lock package-lock.json node_modules android/app/build ios/Pods ios/Podfile.lock ~/Library/Developer/Xcode/DerivedData && yarn install && cd ios && pod install && cd .. && yarn start --reset-cache",

    "clean-android-cache": "cd android && ./gradlew clean",
```
I hope it works!

shadowsheep1, 2025-06-13:
> 🔴 Run Time Error when running app on 16 KB emulator on Android API 35
> 
> * RN: 0.77
> * react-native-vision-camera: "^4.6.4"
> 
> ```
>  E  Failed to load VisionCamera C++ library!
> java.lang.UnsatisfiedLinkError: dlopen failed: empty/missing DT_HASH/DT_GNU_HASH in "/data/app/~~0d4CTKCKMv0ej7Ehb499Eg==/{package_name}-sUKN7tsAleCAEENsgJohtA==/base.apk!/lib/arm64-v8a/libVisionCamera.so" (new hash type from the future?)
> at java.lang.Runtime.loadLibrary0(Runtime.java:1081)
> at java.lang.Runtime.loadLibrary0(Runtime.java:1003)
> at java.lang.System.loadLibrary(System.java:1765)
> at com.mrousavy.camera.react.CameraViewModule.<clinit>(CameraViewModule.kt:48)
> at com.mrousavy.camera.react.CameraPackage.createNativeModules(CameraPackage.kt:11)
> at com.facebook.react.ReactPackageTurboModuleManagerDelegate.initialize(ReactPackageTurboModuleManagerDelegate.java:98)
> at com.facebook.react.ReactPackageTurboModuleManagerDelegate.<init>(ReactPackageTurboModuleManagerDelegate.java:57)
> at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate.<init>(DefaultTurboModuleManagerDelegate.kt:35)
> at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate.<init>(Unknown Source:0)
> at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate$Builder.build(DefaultTurboModuleManagerDelegate.kt:62)
>  at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate$Builder.build(DefaultTurboModuleManagerDelegate.kt:42)
> at com.facebook.react.ReactPackageTurboModuleManagerDelegate$Builder.build(ReactPackageTurboModuleManagerDelegate.java:266)
> at com.facebook.react.runtime.ReactInstance.<init>(ReactInstance.java:184)
> at com.facebook.react.runtime.ReactHostImpl.lambda$getOrCreateReactInstanceTask$22(ReactHostImpl.java:1201)
> at com.facebook.react.runtime.ReactHostImpl.$r8$lambda$sX0vNfk_MttxfVwE2w9ALnXHrPI(Unknown Source:0)
> at com.facebook.react.runtime.ReactHostImpl$$ExternalSyntheticLambda41.then(D8$$SyntheticClass:0)
> at com.facebook.react.runtime.internal.bolts.Task$7.run(Task.java:423)
> at com.facebook.react.runtime.internal.bolts.Executors$ImmediateExecutor.execute(Executors.kt:41)
> at com.facebook.react.runtime.internal.bolts.Task.completeImmediately(Task.java:418)
> at com.facebook.react.runtime.internal.bolts.Task.continueWith(Task.java:288)
> at com.facebook.react.runtime.internal.bolts.Task.continueWith(Task.java:299)
> at com.facebook.react.runtime.internal.bolts.Task$5.then(Task.java:354)
> at com.facebook.react.runtime.internal.bolts.Task$5.then(Task.java:346)
> at com.facebook.react.runtime.internal.bolts.Task$8.run(Task.java:460)
> at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
> at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
> at java.lang.Thread.run(Thread.java:1012)
> ```

Let's see if [this](https://github.com/mrousavy/react-native-vision-camera/tree/feat/16kb-page-size) helps