margelo/react-native-vision-camera #3397: 🐛 Vision Camera Processor Frame Orientation

piroposantosdev, 2025-02-03:
### What's happening?

I know there is many other topics opened here for this point, and that there is a PR waiting for money support to insert Orientation in camera component, but I'd like to know if there is any way to fix this bug by myself?

I'm using a tflite model and camera processor to detect objects. But while the frame is not in the same orientation and screen proportion, the detection bounding box is not fitting the correct area as you can see below (tested in iphone 14 - Orientation is always showing landscape-right):

I tried fix it changing things in .swift but however i could change the display name to portrait in orientation, screen size doesn't changes, showing always landscape mode (1920x1080).

Can someone helps me saying where exactly should i change to fix it? I saw there is a commit here saying that we must change to -90 to identify as portrait, but where exactly should I change?

### Reproduceable Code

```tsx
Just a simple camera vision component with a processor (frame always return landscape-right in back camera)
```

### Relevant log output

```shell
Orientation: Landscape-right
```

### Camera Device

```json
I don't have this info
```

### Device

iPhone 11

### VisionCamera Version

Latest

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

piroposantosdev, 2025-02-03:
> Guten Tag, Hans here! 🍻
>
> Thank you for your detailed issue report. It looks like you are trying to resolve a framing issue with your camera processor. However, I see a couple of points that could be [...]

None of your points are fitting my question. There is no example page testing the processor with a tflite model.

And about the point of changing .swift files, i rollback all to original to prevent a mismatching

piroposantosdev, 2025-02-06:
> [image]

Hi, can you help me understand where should I change to fix portrait/landscape bug? I tried to change in many swift files and i was able to change the display name from landscape-right (iphone back camera) to display portrait, but dimensions were not changed, they keep in 1920x1080

mthinh, 2025-03-09:
@piroposantosdev did you manage to solve this ? 
facing the same issue , im not sure if the frame orientation and size are correct, 
tried this ios example of mediapipe
https://github.com/google-ai-edge/mediapipe-samples/tree/main/examples/object_detection/ios
the frame's orientation is portrait, and size should be 1080 for width, 1920 for height
while on vision camera frame seem return landscape-right or left