margelo/react-native-vision-camera #3399: Unable to scan QR code using V2 of react-native-vision-camera due to react-native-reanimated errors

shengl711, 2025-02-07:
### How were you trying to build the app?

I have tried different combinations in order to enable QR code when using react-native-vision-camera but neither of below options are working.   

When installing the latest version of react-native-vision-camera, I encounter below error.  As I need to stay on JDK 33 because of the many other dependencies in the project I am working on, I started tried option #1 below.  

Execution failed for task ':app:checkDebugAarMetadata'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
   > 7 issues were found when checking AAR metadata:

       1.  Dependency 'androidx.camera:camera-extensions:1.5.0-alpha03' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.

           :app is currently compiled against android-33.

           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 7.3.1 is 33.

           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdkVerion of at least 34.

Option number 1 which is to revert back to V2 and I had to install code-scanner as that library is required to scan QR code: 

  "react-native-vision-camera": "^2.15.6",
    "vision-camera-code-scanner": "^0.2.0"
    "react-native-reanimated": "^3.4.0",

In this option, I faced issue similar to #1937 and although that issue was closed, there's never a documented resolution but based on one developer's response, I believe 3.5.0 is needed.   So I tried below option. 

Option number 2: 

  "react-native-vision-camera": "^2.15.6",
    "vision-camera-code-scanner": "^0.2.0"
    "react-native-reanimated": "^3.5.0",

I still get this error saying frameprocessor is disabled.  On top of that, I get other syntax errors: 

> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: C:\Users\Betty\StudioProjects\Torus\node_modules
VisionCamera: Building with Hermes...
VisionCamera: Frame Processors are disabled because Reanimated 3.5.0+ is not installed. C++ part will not be built.
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.

C:\Users\Betty\StudioProjects\Torus\node_modules\react-native-reanimated\android\src\main\java\com\swmansion\reanimated\layoutReanimation\AnimationsManager.java:356: error: ')' expected
      if (props.get(Snapshot.TRANSFORM_MATRIX) instanceof ReadableNativeArray matrixArray) {

What I am seeking help with is to find out the proper versions required for react-native-renanimated when using V2 or 2.15.6 of react-native-vision-camera in order to enable QR code scanning

I also like help if there's a V3 of react-native-vision-camera where JDK 33 can still be used?  

### Full build logs

```tsx
> Task :react-native-reanimated:compileDebugJavaWithJavac FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.5.1/userguide/command_line_interface.html#sec:command_line_warnings
372 actionable tasks: 2 executed, 370 up-to-date

info 💡 Tip: Make sure that you have set up your development environment correctly, by running react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor

C:\Users\Betty\StudioProjects\Torus\node_modules\react-native-reanimated\android\src\main\java\com\swmansion\reanimated\layoutReanimation\AnimationsManager.java:356: error: ')' expected
      if (props.get(Snapshot.TRANSFORM_MATRIX) instanceof ReadableNativeArray matrixArray) {
[... 149 lines cut ...]
```

### VisionCamera Version

2.15.6

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Marcin-Falkowski, 2025-02-18:
@shengl711

I see that you have `^` next to dependency version. This error is due to reanimated i think and not compatibility Java project version to Reanimated. Which version you have installed in your project (package.lock or yarn.lock)? 

Try remove `^`, clean everything and try again. 
```
cd android
./gradlew clean
cd ..
npx react-native run-android
```
If the problem still exist, let us now your Java and Gradle version.

shengl711, 2025-02-18:
@Marcin-Falkowski - I am not sure I follow your recommendation.  The ^ is auto-added when I upgrade react-native-reanimated version.   There's ^ for few other libraries too so I am not sure if you are recommending another newer version which by the way, I did try to upgrade to the latest version and I still get the same syntax error.

shengl711, 2025-02-20:
@Marcin-Falkowski - I removed the ^ and using below version which is the latest I installed even in my last week's run and after running android clean, I still see the same build errors on top of the same error about qrcode won't be enabled because frame processor is disabled.   

> Configure project :react-native-vision-camera
VisionCamera: node_modules/ found at: C:\Users\Betty\StudioProjects\Torus\node_modules
VisionCamera: Building with Hermes...
VisionCamera: Frame Processors are disabled because Reanimated 3.5.0+ is not installed. C++ part will not be built.
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.

I wasn't sure if you were asking me to remove my node_modules folder and re-install AFTER I remove the ^ character?

Lock file still has the ^ character although I did remove it from json file.  Below is lock file content: 

"react-native": "0.72.3",
        "react-native-reanimated": "^3.16.7",
        "react-native-vision-camera": "^2.15.6",
        "vision-camera-code-scanner": "^0.2.0"

I am using Java 11 which is required for the private project I am contributing and I know gradle version is old which again is tied to the project and I don't want to change without understanding why I must change it.  I think it's below version based on build.gradle file: 

        classpath("com.android.tools.build:gradle:7.3.1")