margelo/react-native-vision-camera #3400: 🐛

aldunnaeyonus, 2025-02-07:
### What's happening?

I am having an android IOS issue, something is not adding up with Android, I am unable to get the Camera to obey overflow:'hidden', works perfect on Iphone, but not android.

[image]

### Reproduceable Code

```tsx
return (
 <Modal
        visible={modalQRCodeVisable}
        animationType="fade"
        transparent={true}
        onRequestClose={() => setmodalQRCodeVisable(false)}
      >
        <View style={style.centeredView}>
          
          <View style={style.qrmodalView}>
          {device == null ? (
              <Text>No Camera Device</Text>
            ) : (
              <Camera
                style={[
                  StyleSheet.absoluteFill]}
                device={device}
                isActive={true}
                codeScanner={codeScanner}
              />
            )}
          </View>
          <TouchableOpacity
            style={{
              marginTop: 20,
              flexDirection: "row",
              width: 250,
              backgroundColor: "rgba(234, 85, 4, 1)",
              borderRadius: 24,
              padding: 15,
              alignItems: "center",
              justifyContent: "center",
              marginbottom: 20,
            }}
            onPress={() => {
              setmodalQRCodeVisable(false);
              setisBarcodeScannerEnabled(false);
            }}
          >
            <Text
              style={{
                textTransform: "uppercase",
                fontSize: 20,
                fontWeight: 600,
                color: "#fff",
              }}
            >
              {i18n.t("Close")}
            </Text>
          </TouchableOpacity>
        </View>
      </Modal>
)

const style = StyleSheet.create({
 centeredView: {
    flex: 1,
    justifyContent: "center",
    alignItems: "center",
    overflow: "hidden",
  },
  qrmodalView: {
    width: ScreenWidth - 50,
    height: ScreenWidth - 50,
    backgroundColor: "white",
    borderRadius: 20,
    alignItems: "center",
    shadowColor: "#000",
    shadowOffset: {
      width: 0,
      height: 2,
    },
    shadowOpacity: 0.25,
    shadowRadius: 22,
    elevation: 7,
  }
}
```

### Relevant log output

```shell
No Logs
```

### Camera Device

```json
None
```

### Device

Samsung Flip 6

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [x] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ChristopherGabba, 2025-02-10:
Hey @aldunnaeyonus, just stumbled across your issue. I'm not a maintainer or anything but I had this issue as well. I used a prop @mrousavy provided called `androidViewType` and I set it to `texture-view`. See if this fixes it for you on Android? This is a long standing problem for me in my app with `expo-video` as well.

aldunnaeyonus, 2025-02-10:
Thank you so much.

closed 2025-02-10 by aldunnaeyonus