margelo/react-native-vision-camera #3401: 🐛 Task :react-native-vision-camera:compileDebugKotlin FAILED

Musheer0, 2025-02-07:
### What's happening?

### Bug Report  

**Description**  
I encountered an issue while trying to build my React Native project. The build fails with the following error:  

**Error Message:**  
> Task :react-native-vision-camera:compileDebugKotlin FAILED

> Task :react-native-vision-camera:configureCMakeDebug[x86]
C/C++: VisionCamera: Frame Processors: OFF!
52 actionable tasks: 42 executed, 10 up-to-date                                                                      

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor 

[Fatal Error] annotations-31.7.2.pom:2:90: The markup in the document following the root element must be well-formed.
[Fatal Error] annotations-31.7.2.pom:2:90: The markup in the document following the root element must be well-formed.
No modules to process in combine-js-to-schema-cli. If this is unexpected, please check if you set up your NativeComponent correctly. See combine-js-to-schema.js for how codegen finds modules.
e: file:///C:/Users/mubas/Music/awesomeapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'.
e: file:///C:/Users/mubas/Music/awesomeapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3 'getLifecycle' overrides nothing.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 33s
error Failed to install the app. Command failed with exit code 1: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081
[Fatal Error] annotations-31.7.2.pom:2:90: The markup in the document following the root element must be well-formed. [Fatal Error] annotations-31.7.2.pom:2:90: The markup in the document following the root element must be well-formed. No modules to process in combine-js-to-schema-cli. If this is unexpected, please check if you set up your NativeComponent correctly. See combine-js-to-schema.js for how codegen finds modules.
e: file:///C:/Users/mubas/Music/awesomeapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1 Class 'CameraSession' is not abstract and does not implement abstract member 'lifecycle'. e: file:///C:/Users/mubas/Music/awesomeapp/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3 'getLifecycle' overrides nothing. FAILURE: Build failed with an exception. * What went wrong: Execution failed for task ':react-native-vision-camera:compileDebugKotlin'. > A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction > Compilation error. See log for more details * Try: > Run with --stacktrace option to get the stack trace. > Run with --info or --debug option to get more log output. > Run with --scan to get full insights. > Get more help at https://help.gradle.org. BUILD FAILED in 33s.
info Run CLI with --verbose flag for more details.

---

### Environment  

- **React Native Version:** 0.77.0  
- **React Version:** 18.3.1  
- **React Native Vision Camera Version:** 4.6.3  
- **Android SDK Version:** 35  
- **Build Tools Version:** 35.0.0  
- **Min SDK Version:** 24  
- **Target SDK Version:** 34  
- **NDK Version:** 27.1.12297006  
- **Kotlin Version:** 2.0.21  
- **Gradle Plugin Versions:**  
  - `com.android.tools.build:gradle`  
  - `com.facebook.react:react-native-gradle-plugin`  
  - `org.jetbrains.kotlin:kotlin-gradle-plugin`  

---

### Steps to Reproduce  

1. Run `npm install` or `yarn install`.  
2. Run `npx react-native run-android`.  
3. Observe the error in the terminal.  

---

### Build.gradle Configuration  
[... 97 lines cut ...]
### Camera Device

```json
non
```

### Device

android emulate pixel 7a

### VisionCamera Version

^4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

leonelunderscore, 2025-02-07:
I have the same issue !

KESHAOYE, 2025-02-08:
I have the same issue too.

guzeradimaz, 2025-02-08:
Same issue +++

maximcoding, 2025-02-08:
LETS GO++

mschmid, 2025-02-09:
Same issue

Anuj2286, 2025-02-10:
Same issue

OTitaev, 2025-02-10:
Same issue

KESHAOYE, 2025-02-10:
The issue can be resolved by applying a patch.

saif482, 2025-02-10:
@KESHAOYE what is the patch? I am having same issue at RN v0.73.11

```
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.73.11",
    "react-native-vision-camera": "^4.3.2"
}
```

```
buildscript {
    ext {
        buildToolsVersion = "34.0.0"
        minSdkVersion = 24
        compileSdkVersion = 34
        targetSdkVersion = 34
        ndkVersion = "25.1.8937393"
        kotlinVersion = "1.8.0"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("org.jetbrains.kotlin:kotlin-gradle-plugin")
        classpath 'com.android.tools.build:gradle:4.2.2'
        classpath 'com.google.gms:google-services:4.4.2'
        classpath 'com.google.firebase:firebase-crashlytics-gradle:2.7.1'
        classpath('com.google.gms:google-services:4.3.3')
    }
}
```

KESHAOYE, 2025-02-10:
@saif482 This is the patch information generated by patch-package, you can give it a try.

diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
index 862c03a..5a31c76 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
@@ -90,7 +90,8 @@ class CameraSession(internal val context: Context, internal val callback: Callba
     }
   }
 
-  override fun getLifecycle(): Lifecycle = lifecycleRegistry
+  override val lifecycle: Lifecycle
+    get() = lifecycleRegistry
 
   /**
    * Configures the [CameraSession] with new values in one batch.

reyhanadp, 2025-02-10:
same issue here

Dennis-Neduvelil, 2025-02-10:
Solution: The Android build fails when adding react-native-vision-camera. This issue can be resolved by updating the code in `CameraSession.kt:`

`File: package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt`

Fix:

```
93- override fun getLifecycle(): Lifecycle = lifecycleRegistry  
93+ override val lifecycle: Lifecycle  
94+ get() = lifecycleRegistry  
```

mertkahramanturk, 2025-02-11:
Deleting node_modules and reinstalling it fixes it.

souhailtourjmen, 2025-02-12:
> The Android build fails when adding react-native-vision-camera. This issue can be resolved by updating the code in `CameraSession.kt:`
>
> `File: [...]

This Node.js script reads the CameraSession.kt file, replaces the specified line and saves the change. Run it with node script.js from the root of your project.  
folder scripts/Update_Camera_Session.js
-----------------------begin Script ------------------------
const fs = require('fs');
const path = require('path');
const filePath = path.join(
  __dirname,
  '../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt'
);

// Read file contents
fs.readFile(filePath, 'utf8', (err, data) => {
  if (err) {
    console.error('Erreur lors de la lecture du fichier :', err);
    return;
  }

  // Replace the specific line
  const updatedData = data.replace(
    'override fun getLifecycle(): Lifecycle = lifecycleRegistry',
    `override val lifecycle: Lifecycle  
  get() = lifecycleRegistry`
  );

  // Write changes to file
  fs.writeFile(filePath, updatedData, 'utf8', (err) => {
    if (err) {
      console.error("Erreur lors de l'écriture du fichier :", err);
      return;
    }
    console.log('Modification effectuée avec succès !');
  });
});
-----------------------end Script --------------------

in package.json 
 "scripts": {
  "android": "node scripts/Update_Camera_Session.js && react-native run-android",
........
}

shealan, 2025-02-16:
Running into the same issue on a new project. Any ideas?

Marcin-Falkowski, 2025-02-18:
Try this:

For newests RN versions:
1. Install patch-package and postinstall `npm install patch-package postinstall-postinstall`
2. Edit files for `react-native-vision-camera` in node_modules with this commit [6501b63#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b](https://github.com/mrousavy/react-native-vision-camera/commit/6501b6398897031c211ef9e3fac1a207953e0a31#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b)
3. Generate patch `npx patch-package react-native-vision-camera`

You can also add script to your package.json scripts:
```
"scripts": {
  "postinstall": "patch-package"
}
```

@saif482 to use latest version of react-native-vision-camera (v.4.6.3) - update your react native and whole stack.

falcondpr, 2025-08-19:
Following the steps I still have the problem

```bash
info A dev server is already running for this project on port 8081.
info Installing the app...

> Configure project :app
Reading env from: .env
**************************
*** Missing .env file ****
**************************

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ??
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at C:\dev\project-name\node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core not found, Frame Processors are disabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-vision-camera:generateCodegenSchemaFromJavaScript
> Task :react-native-vision-camera:compileDebugKotlin FAILED

[Incubating] Problems report is available at:
file:///C:/dev/project-name/android/build/reports/problems/problems-report.html

Deprecated Gradle features were used in this build, making it incompatible with Gradle 10.
You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.
For more on this, please refer to https://docs.gradle.org/9.0.0-rc-1/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

273 actionable tasks: 29 executed, 244 up-to-date
Consider enabling configuration cache to speed up this build:
https://docs.gradle.org/9.0.0-rc-1/userguide/configuration_cache_enabling.html

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor.
To read more about doctor command visit:
https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor

No modules to process in combine-js-to-schema-cli.
If this is unexpected, please check if you set up your NativeComponent correctly.
See combine-js-to-schema.js for how codegen finds modules.

File system event consumer thread stopped due to exception
java.nio.file.InvalidPathException: Illegal char <                                                   > at index 86:
C:\dev\project-name\android\ebug\mergeDebugResources\values-pt_values-pt.arsc.flat
        at java.base/sun.nio.fs.WindowsPathParser.normalize(Unknown Source)
        at java.base/sun.nio.fs.WindowsPathParser.parse(Unknown Source)
        at java.base/sun.nio.fs.WindowsPathParser.parse(Unknown Source)
        at java.base/sun.nio.fs.WindowsPath.parse(Unknown Source)
        at java.base/sun.nio.fs.WindowsFileSystem.getPath(Unknown Source)
        at java.base/java.nio.file.Path.of(Unknown Source)
        at java.base/java.nio.file.Paths.get(Unknown Source)
        at org.gradle.internal.watch.registry.impl.DefaultFileWatcherRegistry$1.handleChangeEvent(DefaultFileWatcherRegistry.java:83)
        at org.gradle.fileevents.internal.AbstractFileEventFunctions$ChangeEvent.handleEvent(AbstractFileEventFunctions.java:244)
        at org.gradle.internal.watch.registry.impl.DefaultFileWatcherRegistry.lambda$createAndStartEventConsumerThread$0(DefaultFileWatcherRegistry.java:78)
        at java.base/java.lang.Thread.run(Unknown Source)

e: file:///C:/dev/project-name/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:34:1
Class 'CameraSession' is not abstract and does not implement abstract member: val lifecycle: Lifecycle

e: file:///C:/dev/project-name/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:93:3
'getLifecycle' overrides nothing.

e: file:///C:/dev/project-name/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewManager.kt:31:5
Return type mismatch: expected 'MutableMap<String, Any>?', actual 'Map<String, Any>'.

e: file:///C:/dev/project-name/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewModule.kt:192:20
Unresolved reference 'currentActivity'.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to generate a Build Scan (Powered by Develocity).
> Get more help at https://help.gradle.org.

BUILD FAILED in 1m 10s
error Failed to install the app. Command failed with exit code 1: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081

```

The version of my package is 

```bash
...
    "react-native-vision-camera": "4.6.3",
...
```

node 20.19.4
npm 10.8.2

```bash
buildscript {
    ext {
        // buildToolsVersion = "33.0.2"
        minSdkVersion = 24
        compileSdkVersion = 34
        targetSdkVersion = 33
        ndkVersion = "27.1.12297006"
        kotlinVersion = "2.1.20"
    }
```

I use the windows SO

globemediaofficial, 2025-08-21:
I was able to patch the relevant errors on **Expo SDK 54 (React Native 0.81.0)** and successfully build the project. I’ve attached the patch file below for anyone who wants to use it:

[react-native-vision-camera+4.7.1.patch](https://github.com/user-attachments/files/21913924/react-native-vision-camera%2B4.7.1.patch)

globemediaofficial, 2025-08-21:
Ran into the same issue with Expo SDK 54 (React Native 0.81.0). Here is the fix (and patch file):

In `node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewManager.kt`, `getExportedCustomDirectEventTypeConstants` expects `MutableMap<String, Any>` but the current code returns `Map<String, Any>`. Add `toMutableMap()` at the end to fix it.
```jsx
  override fun getExportedCustomDirectEventTypeConstants(): MutableMap<String, Any>? {
    return MapBuilder.builder<String, Any>()
      .put(CameraViewReadyEvent.EVENT_NAME, MapBuilder.of("registrationName", "onViewReady"))
      .put(CameraInitializedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onInitialized"))
      .put(CameraStartedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onStarted"))
      .put(CameraStoppedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onStopped"))
      .put(CameraShutterEvent.EVENT_NAME, MapBuilder.of("registrationName", "onShutter"))
      .put(CameraErrorEvent.EVENT_NAME, MapBuilder.of("registrationName", "onError"))
      .put(CameraCodeScannedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onCodeScanned"))
      .put(CameraPreviewStartedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onPreviewStarted"))
      .put(CameraPreviewStoppedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onPreviewStopped"))
      .put(CameraOutputOrientationChangedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onOutputOrientationChanged"))
      .put(CameraPreviewOrientationChangedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onPreviewOrientationChanged"))
      .put(AverageFpsChangedEvent.EVENT_NAME, MapBuilder.of("registrationName", "onAverageFpsChanged"))
      .build()
      .toMutableMap() // <-- convert to MutableMap
  }
```

In `node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraViewModule.kt`, `canRequestPermission` references `currentActivity` which causes an error. Change it to `reactApplicationContext.currentActivity`.
```jsx
  private fun canRequestPermission(permission: String): Boolean {
    val activity = reactApplicationContext.currentActivity as? PermissionAwareActivity
    return activity?.shouldShowRequestPermissionRationale(permission) ?: false
  }
```

Patch file:
[react-native-vision-camera+4.7.1.patch](https://github.com/user-attachments/files/21914760/react-native-vision-camera%2B4.7.1.patch)

souhailtourjmen, 2025-08-21:
I ran into the same problem and fixed it by adding a small postinstall patch script.
The script automatically replaces the problematic lines in react-native-vision-camera after yarn install / npm install.

👉 Create a file scripts/patch-vision-camera.js:
`const fs = require("fs");
const path = require("path");

const visionCameraPath = path.join(
  __dirname,
  "../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera"
);

function patchFile(filePath, searchValue, replaceValue) {
  const absolutePath = path.resolve(filePath);

  if (!fs.existsSync(absolutePath)) {
    console.warn(`⚠️ File not found: ${absolutePath}`);
    return;
  }

  let content = fs.readFileSync(absolutePath, "utf8");

  if (content.includes(replaceValue)) {
    console.log(`✅ Already patched: ${filePath}`);
    return;
  }

  if (!content.includes(searchValue)) {
    console.warn(`⚠️ Could not find "${searchValue}" in ${filePath}`);
    return;
  }

  content = content.replace(searchValue, replaceValue);
  fs.writeFileSync(absolutePath, content, "utf8");

  console.log(`🔧 Patched: ${filePath}`);
}

// 1) Patch CameraViewManager.kt
patchFile(
  path.join(visionCameraPath, "react/CameraViewManager.kt"),
  "override fun getExportedCustomDirectEventTypeConstants(): MutableMap<String, Any>? =",
  "override fun getExportedCustomDirectEventTypeConstants(): Map<String, Any>? ="
);

// 2) Patch CameraViewModule.kt
patchFile(
  path.join(visionCameraPath, "react/CameraViewModule.kt"),
  "val activity = currentActivity as? PermissionAwareActivity",
  "val activity = reactApplicationContext.currentActivity as? PermissionAwareActivity"
);
`
Then add this line to your package.json so it runs automatically:
`"scripts": {
  "postinstall": "node scripts/patch-vision-camera.js"
}`

swtm-dev, 2025-08-22:
@souhailtourjmen Thanks man it's really working now 👍