margelo/react-native-vision-camera #3402: 🔧 useSkiaFrameProcessor crash app on Samsung s10

DonOzOn, 2025-02-07:
### How were you trying to build the app?

I tried implement react native vision camera with filter  like below code but it automatically close (crash) after about 15s. When i hide 
`     frameProcessor={frameProcessor}` its no longer crash. Im using latest [version.](url)
```
  const colorMatrix = [
    -0.578, 0.99, 0.588, 0, 0, 0.469, 0.535, -0.003, 0, 0, 0.015, 1.69, -0.703,
    0, 0, 0, 0, 0, 1, 0,
  ];

  const colorFilter = Skia.ColorFilter.MakeMatrix(colorMatrix);
  const paint = Skia.Paint();
  paint.setColorFilter(colorFilter);
  const frameProcessor = useSkiaFrameProcessor(
    (frame) => {
      'worklet'
      frame.render(paint);
    },
    [paint],
  );

   <Camera
              device={device}
              enableZoomGesture={true}
              exposure={exposure}
              focusable={true}
              format={format}
              frameProcessor={frameProcessor}
              isActive={true}
              onError={(error) => {
                console.log('error', error);
              }}
              onTouchEnd={onTouchEnd}
              onTouchStart={handleFocus}
              photo={true}
              pixelFormat={'rgb'}
              preview={true}
              ref={cameraRef}
              style={[
                {
                  alignSelf: 'center',
                  height:
                    aspectRatio > 1.7
                      ? width * aspectRatio - 100
                      : width * aspectRatio,
                  width: '100%',
                },
                borders.rounded_16,
              ]}
              zoom={zoomLevel}
            />
```

### Full build logs

```tsx
2025-02-08 00:20:32.495 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.495 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.495 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.495 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.496 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.496 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.744 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.745 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.745 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.746 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.746 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.747 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.747 11396-11430 System                  com.docam                            W  A resource failed to call close. 
2025-02-08 00:20:32.903 11396-12065 CameraView              com.docam                            I  invokeOnAverageFpsChanged(29.804727646454268)
[... 212 lines cut ...]
```

### VisionCamera Version

4.6.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DonOzOn, 2025-02-07:
https://eu-central.storage.cloudconvert.com/tasks/2db641a3-6bf1-4805-81e8-8aed3cdda151/IMG_7991.mp4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&X-Amz-Credential=cloudconvert-production%2F20250207%2Ffra%2Fs3%2Faws4_request&X-Amz-Date=20250207T174050Z&X-Amz-Expires=86400&X-Amz-Signature=7bb5fa598b991cf1b434ee7de15ee1e9923c7660c8a107547034372361c04c24&X-Amz-SignedHeaders=host&response-content-disposition=inline%3B%20filename%3D%22IMG_7991.mp4%22&response-content-type=video%2Fmp4&x-id=GetObject|

This is my record about my issue

jslok, 2025-04-03:
Same issue here on a Pixel 7. FPS drops then app crashes after about 15 seconds when using useSkiaFrameProcessor.

RN 0.75.4
react-native-vision-camera 4.6.4
@shopify/react-native-skia 1.12.3