margelo/react-native-vision-camera #3406: 🐛 FormatFilter not respecting priority order with photoAspectRatio and photoResolution combined

anderson-souza-involves, 2025-02-10:
### What's happening?

Hello, I'm facing some issues when trying to use the the format filter with `photoAspectRatio` combined with `photoResolution`.

Basically, I want to use a 4/3 aspect ratio with a maximum resolution of `width: 2582` and `height: 1936`, which will be the maximum resolution of a 5MP camera. 

The `photoAspectRatio` is the first property of my filter, and the `photoResolution` it's the last, so I was expecting that the filter would first get all available formats that have a 4/3 aspect ratio, and then, find the closest to the desired photoResolution, but it's not working on all devices.

```typescript
const filters: FormatFilter[] = [
      {
        photoAspectRatio: 4.0 / 3.0,
        videoAspectRatio: 4.0 / 3.0,
        photoResolution: {
          width: 2582,
          height: 1936,
        },
      },
    ]
const result = getCameraFormat(device, filters)
```

Most devices are getting the format correctly, but I have a Samsung A32 and in this device, when setting the `photoResolution` filter, it's getting a format with 16/9 aspect ratio because the resolution is closer to the `photoResolution` property.

In the Reproduceable Code, I created a simple Unit test that mocks a Samsung A32 device with the returned formats and execute the `getCameraFormat()` method.

The first test, when I don't set the photoResolution filter, runs correctly and the format with 4/3 resolution is returned.

But the second test, when I set the `photoResolution` as the last filter, the format returns a resolution of 2560x1440(16:9), which is the closest to the photoResolution and it's not respecting the aspect Ratio.

Is there a way for me to force the format to be as I described above, first selecting all with aspect ratio 4/3, then selecting the closest to the desired resolution?

Thanks in advance

### Reproduceable Code

```tsx
import { FormatFilter } from './getCameraFormat'
import { getCameraFormat } from './getCameraFormat'
import { CameraDevice } from '../types/CameraDevice'

describe('getCameraFormat', () => {
  const mockDevice: CameraDevice = {
    formats: [
      {
        videoStabilizationModes: ['off', 'cinematic'],
        autoFocusSystem: 'contrast-detection',
        photoWidth: 2560,
        supportsPhotoHdr: false,
        supportsDepthCapture: false,
        maxISO: 4000,
        minISO: 20,
        minFps: 5,
        videoWidth: 1920,
        supportsVideoHdr: false,
        videoHeight: 1080,
        fieldOfView: 82.66746985425544,
        maxFps: 30,
        photoHeight: 1440,
      },
      {
        videoStabilizationModes: ['off', 'cinematic'],
        autoFocusSystem: 'contrast-detection',
        photoWidth: 1920,
        supportsPhotoHdr: false,
        supportsDepthCapture: false,
        maxISO: 4000,
        minISO: 20,
        minFps: 5,
        videoWidth: 1920,
[... 509 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Android Samsung A32

### VisionCamera Version

4.3.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

anderson-souza-involves, 2025-02-10:
> Guten Tag, Hans here! ☕️
>
> Thanks for your detailed issue report. It looks like you’re encountering an unexpected behavior with the `getCameraFormat` function on the Samsung A32 when combining the [...]

Issue updated 👍🏻

mrousavy, 2025-02-17:
I think the filtering is correct, but you can always write your own filter function - just loop over `device.formats`.

If there's actually a bug in `useCameraFormat` please submit a PR.

closed 2025-02-17 by mrousavy

lc3t35, 2025-04-21:
I think there is a bug in useCameraFormat on android, getCameraFormat returns always this crazy values (ratio 1.33) for photo ok for video, whatever format is requested : 

formatCamera {"autoFocusSystem": "contrast-detection", "fieldOfView": 77.91879105237977, "maxFps": 30, "maxISO": 3200, "minFps": 12, "minISO": 50, "photoHeight": 3120, "photoWidth": 4160, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1080, "videoStabilizationModes": ["off"], "videoWidth": 1920}