margelo/react-native-vision-camera #3407: ✨ Support Proxy/Copying shared array data so we can use runAsync with data from resize plugin and similar.

FieldMarshallVague, 2025-02-12:
### What feature or enhancement are you suggesting?

Basically, I'm running into this problem:

https://github.com/mrousavy/react-native-vision-camera/issues/2520#issuecomment-2073868718

I'm getting a segmentation fault when running my tflite model inside a worklet that I'm passing the output of the resize plugin.  I've tried copying the data etc., but it all ends up the same way.  Segfault, SIGSEG, dead 11, something something.  I'm looking at patching the plugin myself, but this is way outside my experience and I don't know what I don't know (if you know what I mean).

I know it's possible to use runAtTargetFps, but that is janky and the sync approach is bad UX (either jank on target FPS or just <5fps in my case - using fast-tflite).

It would be great if the fix mrousavy suggests in the above link could be implemented.  I guess proxying would be ideal, if that means the frame data doesn't have to be duplicated?

If I make any progress on this myself, I'll report back here, but I've never done anything on Android native before (and guessing this might also be an issue on iOS?).  

I don't seem to be able to select 'both' in the platform dropdown, but I'm assuming this either isn't an issue on iOS or would also need to be done on that, too?

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

(as above) I know it's possible to use runAtTargetFps, but that is janky and all other approaches are bad UX (either jank on target FPS or just <10fps in my case).

### Additional information

- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

FieldMarshallVague, 2025-02-12:
^^ The issue I linked to has logs which are exemplary of the problem.  I've had different errors, but they all refer to the same underlying issue, afaict.

mrousavy, 2025-02-17:
This is way more complicated than it sounds. Check out Nitro's ArrayBuffer implementation to understand the constraints.

FieldMarshallVague, 2025-02-17:
> This is way more complicated than it sounds. Check out Nitro's ArrayBuffer implementation to understand the constraints.

@mrousavy Thanks for letting me know, and the tip.  I'd already started to realise it might be more than I could handle, and this pretty much confirms it (very new to native and no cpp experience).  Feel free to close this.