margelo/react-native-vision-camera #3413: 🔧  > Task :react-native-vision-camera:compileDebugKotlin FAILED

micromen1986, 2025-02-14:
### How were you trying to build the app?

npm run android

### Full build logs

```tsx
> Task :react-native-vision-camera:compileDebugKotlin FAILED
```

### Project dependencies

```json
"dependencies": {
    "@react-native-async-storage/async-storage": "^2.1.1",
    "@react-native-community/geolocation": "^3.4.0",
    "@react-native-community/netinfo": "^11.4.1",
    "@react-native-masked-view/masked-view": "^0.3.2",
    "@react-navigation/drawer": "^7.1.1",
    "@react-navigation/native": "^7.0.14",
    "@react-navigation/native-stack": "^7.2.0",
    "@react-navigation/stack": "^7.1.1",
    "formik": "^2.4.6",
    "moment": "^2.30.1",
    "react": "18.3.1",
    "react-native": "0.77.1",
    "react-native-biometrics": "^3.0.1",
    "react-native-blob-util": "^0.21.2",
    "react-native-date-picker": "^5.0.10",
    "react-native-device-info": "^14.0.4",
    "react-native-dotenv": "^3.4.11",
    "react-native-fast-image": "^8.6.3",
    "react-native-geocoding": "^0.5.0",
    "react-native-gesture-handler": "^2.23.1",
    "react-native-image-crop-picker": "^0.42.0",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-month-year-picker": "^1.9.0",
    "react-native-permissions": "^5.2.5",
    "react-native-reanimated": "^3.16.7",
    "react-native-safe-area-context": "^5.2.0",
    "react-native-screens": "^4.6.0",
    "react-native-toast-message": "^2.2.1",
    "react-native-vector-icons": "^10.2.0",
    "react-native-vision-camera": "^4.6.3",
    "react-native-webview": "^13.13.2"
  },
```

### VisionCamera Version

4.6.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

micromen1986, 2025-02-14:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.lambda$executeIfValid$1(ExecuteActionsTaskExecuter.java:130)
	at org.gradle.internal.Try$Failure.ifSuccessfulOrElse(Try.java:293)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeIfValid(ExecuteActionsTaskExecuter.java:128)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:116)
	at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter.java:46)
	at org.gradle.api.internal.tasks.execution.ResolveTaskExecutionModeExecuter.execute(ResolveTaskExecutionModeExecuter.java:51)
	at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:57)
	at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:74)
	at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:36)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:77)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:55)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:209)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:204)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:166)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:53)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:52)
	at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:42)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:331)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:318)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.lambda$execute$0(DefaultTaskExecutionGraph.java:314)
	at org.gradle.internal.operations.CurrentBuildOperationRef.with(CurrentBuildOperationRef.java:85)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:314)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:303)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.execute(DefaultPlanExecutor.java:459)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:376)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
	at org.gradle.internal.concurrent.AbstractManagedExecutor$1.run(AbstractManagedExecutor.java:48)
Caused by: org.gradle.workers.internal.DefaultWorkerExecutor$WorkExecutionException: A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
	at org.gradle.workers.internal.DefaultWorkerExecutor$WorkItemExecution.waitForCompletion(DefaultWorkerExecutor.java:287)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.lambda$waitForItemsAndGatherFailures$2(DefaultAsyncWorkTracker.java:130)
	at org.gradle.internal.Factories$1.create(Factories.java:31)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withoutLocks(DefaultWorkerLeaseService.java:335)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withoutLocks(DefaultWorkerLeaseService.java:318)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withoutLock(DefaultWorkerLeaseService.java:323)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForItemsAndGatherFailures(DefaultAsyncWorkTracker.java:126)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForItemsAndGatherFailures(DefaultAsyncWorkTracker.java:92)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForAll(DefaultAsyncWorkTracker.java:78)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForCompletion(DefaultAsyncWorkTracker.java:66)
[... cut ...]
	at org.gradle.internal.work.DefaultWorkerLeaseService.runAsWorkerThread(DefaultWorkerLeaseService.java:127)
	at org.gradle.internal.work.DefaultWorkerLeaseService.runAsWorkerThread(DefaultWorkerLeaseService.java:132)
	at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.runBatch(DefaultConditionalExecutionQueue.java:164)
	at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.run(DefaultConditionalExecutionQueue.java:133)
	... 2 more

BUILD FAILED in 1m 48s
456 actionable tasks: 147 executed, 309 up-to-date

numandev460, 2025-02-14:
same problem

GeorgeBelben, 2025-02-16:
For the record, I'm seeing the same thing after upgrading to 0.77.1, at a glance it appears to be a mismatch in kotlin versions, vision camera seems to require `1.7.20` whereas after upgrading, `2.0.21` is used

sammy-black, 2025-02-16:
Hey, the issue has been fixed.
I had to manually update their files in my node_modules folder.

Here is the commit link to the two files they updated: https://github.com/mrousavy/react-native-vision-camera/commit/6501b6398897031c211ef9e3fac1a207953e0a31#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b

tayyabrajpoot12, 2025-02-16:
@sammy-black I am facing the same issue I also updated node_modules but still getting the issue my react native version is 0.72.6 and react-native-vision-camera version is 4.6.3

Marcin-Falkowski, 2025-02-17:
> [@sammy-black](https://github.com/sammy-black) I am facing the same issue I also updated node_modules but still getting the issue my react native version is 0.72.6 and react-native-vision-camera version is 4.6.3

Probably react-native-vision-camera v4 is not compatible with your version of React Native and it's dependencies. Use older version or update your stack. Remember, that the 0.74 introduce new architecture, so most of popular libraries in newest versions will be adapted to it and it's can make lot of problems between newest libraries and olders RN's.

orzhtml, 2025-02-17:
> Hey, the issue has been fixed. I had to manually update their files in my node_modules folder.
> 
> Here is the commit link to the two files they updated: [6501b63#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b](https://github.com/mrousavy/react-native-vision-camera/commit/6501b6398897031c211ef9e3fac1a207953e0a31#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b)

Can you generate patches for me?

Marcin-Falkowski, 2025-02-17:
@orzhtml

1. Install patch-package and postinstall `npm install patch-package postinstall-postinstall`
2. Edit files for react-native-vision-camera in node_modules with this commit [6501b63#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b](https://github.com/mrousavy/react-native-vision-camera/commit/6501b6398897031c211ef9e3fac1a207953e0a31#diff-265d108e616a25a6325b349a2b497d527a1cc2c4a972787cc5ed9c5cd5a91c6b)
3. Generate patch  `npx patch-package react-native-vision-camera`

You can also add script to your `package.json` scripts: 
```
"scripts": {
  "postinstall": "patch-package"
}
```

mrousavy, 2025-02-17:
Update react-native.

closed 2025-02-17 by mrousavy

orzhtml, 2025-02-18:
> [@orzhtml](https://github.com/orzhtml)
>
> 1. Install patch-package and postinstall `npm install patch-package postinstall-postinstall`
> 2. Edit files for react-native-vision-camera in node_modules with [...]

thank you