margelo/react-native-vision-camera #3415: 🐛 is it work on react native 0.77 , if work then any setting , plz help

mayursancheti, 2025-02-15:
### What's happening?

i am making something below but not work , even if i change camerasession.kt two lines ( which has given on  https://github.com/mrousavy/react-native-vision-camera/issues/3389)  

`import React, { useState, useCallback, useEffect } from 'react';
import { StyleSheet, View, TouchableOpacity, Text, Image } from 'react-native';
import { Camera, useCameraDevices } from 'react-native-vision-camera';
import Icon from 'react-native-vector-icons/MaterialIcons';

const App = () => {
  const [hasPermission, setHasPermission] = useState(false);
  const [capturedImage, setCapturedImage] = useState(null);
  const devices = useCameraDevices();
  const device = devices.back;

  useEffect(() => {
	(async () => {
	  const cameraPermission = await Camera.requestCameraPermission();
	  const micPermission = await Camera.requestMicrophonePermission(); // Some devices require mic permission
	  setHasPermission(cameraPermission === 'authorized' && micPermission === 'authorized');
	})();
  }, []);

  const takePicture = async (camera) => {
    try {
      const photo = await camera.takePhoto({
        qualityPrioritization: 'quality',
        flash: 'off',
      });
      setCapturedImage(`file://${photo.path}`);
    } catch (error) {
      console.error('Failed to take picture:', error);
    }
  };

  const CameraOverlay = () => (
    <View style={styles.overlay}>
      <View style={styles.cornerOverlay}>
        <View style={styles.corner} />
        <View style={[styles.corner, { backgroundColor: 'green' }]} />
        <View style={[styles.corner, { backgroundColor: 'orange' }]} />
        <View style={[styles.corner, { backgroundColor: 'blue' }]} />
      </View>
    </View>
  );

  const DiagnoseTips = () => (
    <View style={styles.tipsContainer}>
      <Icon name="info-outline" size={24} color="white" />
      <Text style={styles.tipsText}>
        Get Closer to the crop and make sure the crop is clearly visible and it's not too dark or bright
      </Text>
    </View>
  );

  const CameraControls = ({ onCapture }) => (
    <View style={styles.controls}>
      <TouchableOpacity style={styles.controlButton}>
        <Icon name="photo-library" size={30} color="white" />
      </TouchableOpacity>
      <TouchableOpacity style={styles.captureButton} onPress={onCapture}>
        <View style={styles.captureButtonInner} />
      </TouchableOpacity>
      <TouchableOpacity style={styles.controlButton}>
        <Icon name="flash-on" size={30} color="white" />
      </TouchableOpacity>
    </View>
  );

  if (!hasPermission) {
    return (
      <View style={styles.container}>
        <Text>No camera permission</Text>
      </View>
    );
  }

  if (!device) {
    return (
      <View style={styles.container}>
        <Text>Loading camera...</Text>
      </View>
    );
  }

  return (
    <View style={styles.container}>
      {!capturedImage ? (
        <Camera style={styles.camera} device={device} isActive={true} photo={true} enableZoomGesture>
          <View style={styles.header}>
            <TouchableOpacity>
              <Icon name="close" size={24} color="white" />
            </TouchableOpacity>
            <Text style={styles.headerText}>How to Capture the Image?</Text>
          </View>
          <CameraOverlay />
          <DiagnoseTips />
          <CameraControls onCapture={() => takePicture(camera)} />
        </Camera>
      ) : (
        <View style={styles.container}>
          <Image source={{ uri: capturedImage }} style={styles.preview} />
          <View style={styles.buttonContainer}>
            <TouchableOpacity 
              style={styles.recaptureButton}
              onPress={() => setCapturedImage(null)}>
              <Text style={styles.buttonText}>Recapture</Text>
            </TouchableOpacity>
            <TouchableOpacity style={styles.diagnoseButton}>
              <Text style={styles.buttonText}>Diagnose</Text>
            </TouchableOpacity>
          </View>
        </View>
      )}
    </View>
  );
};

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
  camera: {
    flex: 1,
  },
  header: {
    flexDirection: 'row',
    padding: 20,
    alignItems: 'center',
  },
  headerText: {
    color: 'white',
    marginLeft: 20,
    fontSize: 16,
  },
  overlay: {
    flex: 1,
    justifyContent: 'center',
    alignItems: 'center',
  },
  cornerOverlay: {
    width: 300,
    height: 300,
    borderWidth: 2,
    borderColor: 'transparent',
    position: 'relative',
  },
  corner: {
    position: 'absolute',
    width: 40,
    height: 40,
    borderColor: 'yellow',
    borderWidth: 3,
  },
  tipsContainer: {
    flexDirection: 'row',
    backgroundColor: 'rgba(0,0,0,0.5)',
    padding: 15,
    margin: 20,
    borderRadius: 8,
    alignItems: 'center',
  },
  tipsText: {
    color: 'white',
    marginLeft: 10,
    flex: 1,
  },
  controls: {
    flexDirection: 'row',
    justifyContent: 'space-around',
    alignItems: 'center',
    padding: 20,
  },
  controlButton: {
    padding: 10,
  },
  captureButton: {
    width: 70,
    height: 70,
    borderRadius: 35,
    backgroundColor: 'white',
    justifyContent: 'center',
    alignItems: 'center',
  },
  captureButtonInner: {
    width: 60,
    height: 60,
    borderRadius: 30,
    backgroundColor: 'white',
    borderWidth: 2,
    borderColor: '#000',
  },
  preview: {
    flex: 1,
  },
  buttonContainer: {
    flexDirection: 'row',
    justifyContent: 'space-between',
    padding: 20,
    position: 'absolute',
    bottom: 0,
    left: 0,
    right: 0,
  },
  recaptureButton: {
    backgroundColor: 'transparent',
    padding: 15,
    borderRadius: 5,
    borderWidth: 1,
    borderColor: '#2196F3',
  },
  diagnoseButton: {
    backgroundColor: '#2196F3',
    padding: 15,
    borderRadius: 5,
  },
  buttonText: {
    color: 'white',
    fontSize: 16,
  },
});

export default App;`

### Reproduceable Code

```tsx
-
```

### Relevant log output

```shell
-
```

### Camera Device

```json
-
```

### Device

android

### VisionCamera Version

-

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2025-02-15 by maintenance-hans[bot]

mayursancheti, 2025-02-15:
I am testing on real device , also given all permisson , 
still showing no camera permission , is it work with react native 0.77

k-saparia, 2025-05-23:
🔴 Run Time Error when running app on 16 KB emulator on Android API 35

* RN: 0.77
* react-native-vision-camera: "^4.6.4"

```
 E  Failed to load VisionCamera C++ library!
java.lang.UnsatisfiedLinkError: dlopen failed: empty/missing DT_HASH/DT_GNU_HASH in "/data/app/~~0d4CTKCKMv0ej7Ehb499Eg==/{package_name}-sUKN7tsAleCAEENsgJohtA==/base.apk!/lib/arm64-v8a/libVisionCamera.so" (new hash type from the future?)
at java.lang.Runtime.loadLibrary0(Runtime.java:1081)
at java.lang.Runtime.loadLibrary0(Runtime.java:1003)
at java.lang.System.loadLibrary(System.java:1765)
at com.mrousavy.camera.react.CameraViewModule.<clinit>(CameraViewModule.kt:48)
at com.mrousavy.camera.react.CameraPackage.createNativeModules(CameraPackage.kt:11)
at com.facebook.react.ReactPackageTurboModuleManagerDelegate.initialize(ReactPackageTurboModuleManagerDelegate.java:98)
at com.facebook.react.ReactPackageTurboModuleManagerDelegate.<init>(ReactPackageTurboModuleManagerDelegate.java:57)
at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate.<init>(DefaultTurboModuleManagerDelegate.kt:35)
at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate.<init>(Unknown Source:0)
at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate$Builder.build(DefaultTurboModuleManagerDelegate.kt:62)
 at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate$Builder.build(DefaultTurboModuleManagerDelegate.kt:42)
at com.facebook.react.ReactPackageTurboModuleManagerDelegate$Builder.build(ReactPackageTurboModuleManagerDelegate.java:266)
at com.facebook.react.runtime.ReactInstance.<init>(ReactInstance.java:184)
at com.facebook.react.runtime.ReactHostImpl.lambda$getOrCreateReactInstanceTask$22(ReactHostImpl.java:1201)
at com.facebook.react.runtime.ReactHostImpl.$r8$lambda$sX0vNfk_MttxfVwE2w9ALnXHrPI(Unknown Source:0)
at com.facebook.react.runtime.ReactHostImpl$$ExternalSyntheticLambda41.then(D8$$SyntheticClass:0)
at com.facebook.react.runtime.internal.bolts.Task$7.run(Task.java:423)
at com.facebook.react.runtime.internal.bolts.Executors$ImmediateExecutor.execute(Executors.kt:41)
at com.facebook.react.runtime.internal.bolts.Task.completeImmediately(Task.java:418)
at com.facebook.react.runtime.internal.bolts.Task.continueWith(Task.java:288)
at com.facebook.react.runtime.internal.bolts.Task.continueWith(Task.java:299)
at com.facebook.react.runtime.internal.bolts.Task$5.then(Task.java:354)
at com.facebook.react.runtime.internal.bolts.Task$5.then(Task.java:346)
at com.facebook.react.runtime.internal.bolts.Task$8.run(Task.java:460)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
at java.lang.Thread.run(Thread.java:1012)
```

shadowsheep1, 2025-06-13:
> 🔴 Run Time Error when running app on 16 KB emulator on Android API 35
>
> * RN: 0.77
> * react-native-vision-camera: "^4.6.4"
>
> ```
>  E  Failed to load VisionCamera C++ library! [...]

Let's see if [this](https://github.com/mrousavy/react-native-vision-camera/tree/feat/16kb-page-size) helps