margelo/react-native-vision-camera #3425: 🔧 Build Error

ovi100, 2025-02-18:
### How were you trying to build the app?

I tried to build my app using react-native-vision-camera with the command 'npm run android' and it failed to build

### Full build logs

```tsx
> react-native run-android

info Installing the app...

> Configure project :react-native-reanimated
Android gradle plugin: 8.7.2
Gradle: 8.10.2

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at /home/shwapno/Desktop/projects/VisionCamera/node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core found, Frame Processors are enabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to true!

> Task :react-native-vision-camera:generateCodegenSchemaFromJavaScript
> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/home/shwapno/Desktop/projects/VisionCamera/node_modules/react-native-vision-camera/android/.cxx/Debug/s65j3pm1/arm64-v8a'
C/C++: : && /home/shwapno/Android/Sdk/ndk/27.1.12297006/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android24 --sysroot=/home/shwapno/Android/Sdk/ndk/27.1.12297006/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--no-undefined-version -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/s65j3pm1/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/MutableJByteBuffer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraScheduler.cpp.o  /home/shwapno/Android/Sdk/ndk/27.1.12297006/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/aarch64-linux-android/24/liblog.so  -landroid  /home/shwapno/.gradle/caches/8.10.2/transforms/d04e401f6db98d4307004fde50158b02/transformed/react-android-0.77.1-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /home/shwapno/.gradle/caches/8.10.2/transforms/d7d422c0963f92aa7b8a765f3bd54231/transformed/fbjni-0.7.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  /home/shwapno/.gradle/caches/8.10.2/transforms/d04e401f6db98d4307004fde50158b02/transformed/react-android-0.77.1-debug/prefab/modules/reactnative/libs/android.arm64-v8a/libreactnative.so  -latomic -lm && :
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::initialize(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>> const&, facebook::jsi::Runtime*, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>)
C/C++: >>> referenced by JVisionCameraProxy.cpp:45 (/home/shwapno/Desktop/projects/VisionCamera/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp:45)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::JVisionCameraProxy(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraProxy, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jsi::Runtime*, std::__ndk1::shared_ptr<facebook::react::CallInvoker> const&, facebook::jni::basic_strong_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*, facebook::jni::GlobalReferenceAllocator> const&))
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
C/C++: >>> referenced by JVisionCameraProxy.h:42 (/home/shwapno/Desktop/projects/VisionCamera/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.h:42)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp.o:(vision::JSharedArray::JSharedArray(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JSharedArray, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraProxy, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jni::alias_ref<facebook::jni::JByteBuffer>))
C/C++: >>> referenced by JFrameProcessor.cpp:37 (/home/shwapno/Desktop/projects/VisionCamera/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp:37)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
C/C++: ld.lld: error: undefined symbol: RNWorklet::RuntimeLifecycleMonitor::addListener(facebook::jsi::Runtime&, RNWorklet::RuntimeLifecycleListener*)
C/C++: >>> referenced by WKTRuntimeAwareCache.h:46 (/home/shwapno/Desktop/projects/VisionCamera/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTRuntimeAwareCache.h:46)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::RuntimeAwareCache<std::__ndk1::shared_ptr<facebook::jsi::Function>>::get(facebook::jsi::Runtime&))
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::JsiWorkletContext(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>> const&)
C/C++: >>> referenced by allocator.h:165 (/home/shwapno/Android/Sdk/ndk/27.1.12297006/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/__memory/allocator.h:165)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(void std::__ndk1::allocator<RNWorklet::JsiWorkletContext>::construct[abi:ne180000]<RNWorklet::JsiWorkletContext, char const (&) [13]>(RNWorklet::JsiWorkletContext*, char const (&) [13]))
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()
C/C++: >>> referenced by WKTJsiWorklet.h:82 (/home/shwapno/Desktop/projects/VisionCamera/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:82)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(RNWorklet::JsiWorklet::JsiWorklet(facebook::jsi::Runtime&, std::__ndk1::shared_ptr<facebook::jsi::Function>))
C/C++: ld.lld: error: undefined symbol: RNWorklet::RuntimeLifecycleMonitor::removeListener(facebook::jsi::Runtime&, RNWorklet::RuntimeLifecycleListener*)
C/C++: >>> referenced by WKTRuntimeAwareCache.h:37 (/home/shwapno/Desktop/projects/VisionCamera/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTRuntimeAwareCache.h:37)
[... cut ...]
### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Sachiinkk, 2025-02-28:
is there any solution for this ?

jimbofoo, 2025-02-28:
This build error on Android is still happening randomly. Trying to investigate this issue, it seems
the generated cmake config file: react-native-worklets-coreConfig.cmake  seems to be missing the
IMPORTED_LOCATION section, which causes it to quietly not link the worklets library.

cat ./.cxx/Debug/471rl62z/prefab/x86_64/prefab/lib/x86_64-linux-android/cmake/react-native-worklets-core/react-native-worklets-coreConfig.cmake
```
if(NOT TARGET react-native-worklets-core::rnworklets)
add_library(react-native-worklets-core::rnworklets SHARED IMPORTED)
set_target_properties(react-native-worklets-core::rnworklets PROPERTIES
    INTERFACE_INCLUDE_DIRECTORIES "/home/foo/myapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets"
    INTERFACE_LINK_LIBRARIES ""
)
endif()
```

tremblerz, 2025-03-03:
For me this gets fixed by removing node modules directory and android build folders. When you rebuild it afterwards, it should work fine. Would be great if the problem can be spotted and fixed because it takes time to remove folders and rebuild.

hstorz, 2025-03-12:
I also got this build error when switching between old and new arch in Android. As mentioned above, cleaning build dir and node_modules helps but I also had to delete my .gradle folder in the user home directory.

zrina1314, 2025-03-31:
@jimbofoo Can you solve the problem in your way?

jimbofoo, 2025-03-31:
> [@jimbofoo](https://github.com/jimbofoo) Can you solve the problem in your way?

No, what seems to cause this is some kind of race condition. The cmake build of react-native-visition-camera completes before the react-native-worklets-core library. This causes prefab to not find the library (the libraries array is empty here: https://github.com/google/prefab/blob/176f5b1d6fc90297271b44c5452a6ebd9a98c39f/api/src/main/kotlin/com/google/prefab/api/Module.kt#L139  ) which causes the prefab config.cmake generator to think this is a header only library. This is why it sometimes work, sometimes it doesn't.

jimbofoo, 2025-05-06:
Related:

https://github.com/mrousavy/react-native-vision-camera/issues/1708
https://github.com/mrousavy/react-native-vision-camera/issues/1770
https://github.com/mrousavy/react-native-vision-camera/issues/1913
https://github.com/mrousavy/react-native-vision-camera/issues/2290
https://github.com/mrousavy/react-native-vision-camera/issues/2013

https://github.com/margelo/react-native-worklets-core/issues/242
https://github.com/margelo/react-native-worklets-core/issues/235

https://stackoverflow.com/questions/78123397/react-native-error-regards-to-react-native-vision-camera-and-react-native-reanim

jimbofoo, 2025-06-02:
Linking the worklets library explicit as a workaround works here:

```
diff --git a/node_modules/react-native-vision-camera/android/CMakeLists.txt b/node_modules/react-native-vision-camera/android/CMakeLists.txt
index 17097c1..c3baf4d 100644
--- a/node_modules/react-native-vision-camera/android/CMakeLists.txt
+++ b/node_modules/react-native-vision-camera/android/CMakeLists.txt
@@ -63,6 +63,18 @@ target_link_libraries(
         fbjni::fbjni                        # <-- fbjni
 )
 
+if(CMAKE_BUILD_TYPE MATCHES "Debug")
+  target_link_libraries(
+        ${PACKAGE_NAME}
+        "${NODE_MODULES_DIR}/react-native-worklets-core/android/build/intermediates/cmake/debug/obj/${ANDROID_ABI}/librnworklets.so"
+  )
+else()
+  target_link_libraries(
+        ${PACKAGE_NAME}
+        "${NODE_MODULES_DIR}/react-native-worklets-core/android/build/intermediates/cmake/release/obj/${ANDROID_ABI}/librnworklets.so"
+  )
+endif()
+
 # Link react-native (different prefab between RN 0.75 and RN 0.76)
 if(ReactAndroid_VERSION_MINOR GREATER_EQUAL 76)
     target_link_libraries(

```