margelo/react-native-vision-camera #3434: 🐛 QR Code only detected in top-right corner on Samsung A23

olivertylsar, 2025-02-26:
### What's happening?

**Issue Description**
On the Samsung A23, the camera has trouble detecting QR codes when using the example app's CodeScannerPage. No additional props were passed to the Camera component.

Through testing, we discovered that the QR code is recognized only when positioned in the top-right corner of the frame. In all other positions, the detection fails. However, the same QR code is instantly detected anywhere in the frame when using the native camera app.

**Related Issue**
A similar issue was reported here: [#3248](https://github.com/mrousavy/react-native-vision-camera/issues/3248). However, in our case, the issue can be consistently reproduced in the example app.

The suggested workaround of specifying a format did not resolve the issue and instead resulted in a black screen.

**Reproduction**
Here is a video showing the issue on a Samsung A23:

https://github.com/user-attachments/assets/03e77b13-bbb4-45bc-8e6d-8e292a37f204

For comparison, the same code works correctly on a Samsung S21, where the QR code is detected anywhere in the frame:
🔗 [Samsung S21 - QR Code Detection Working](https://github.com/user-attachments/assets/a54112d8-98be-44f2-830f-9daed5609c45)

**Additional Notes**
The issue is reproducible in the example app without modifications.
The same QR code is detected properly by the native camera app, regardless of its position.
Other devices, such as the Samsung S21, do not exhibit this issue.
Specifying a format as a workaround caused a black screen rather than resolving the problem.
This behavior suggests a potential issue with how the camera processes frames on the Samsung A23. Any insights or potential fixes would be greatly appreciated. Thanks! 🚀

### Reproduceable Code

```tsx
// 1. Use a simple default back camera
  const device = useCameraDevice('back')

  // 2. Only activate Camera when the app is focused and this screen is currently opened
  const isFocused = useIsFocused()
  const isForeground = useIsForeground()
  const isActive = isFocused && isForeground

  // 3. (Optional) enable a torch setting
  const [torch, setTorch] = useState(false)

  // 4. On code scanned, we show an aler to the user
  const isShowingAlert = useRef(false)
  const onCodeScanned = useCallback((codes: Code[]) => {
    console.log(`Scanned ${codes.length} codes:`, codes)
    const value = codes[0]?.value
    if (value == null) return
    if (isShowingAlert.current) return
    showCodeAlert(value, () => {
      isShowingAlert.current = false
    })
    isShowingAlert.current = true
  }, [])

  // 5. Initialize the Code Scanner to scan QR codes and Barcodes
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: onCodeScanned,
  })

  return (
    <View style={styles.container}>
      {device != null && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isActive}
          codeScanner={codeScanner}
          torch={torch ? 'on' : 'off'}
          enableZoomGesture={true}
[... 169 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy A23 5G (Android 13) NOT OK, Samsung Galaxy S21 (Android 14) OK

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

capognavalerio, 2025-03-18:
My customers with same phone have the same issue. Any updates about @mrousavy ? Thanks

olivertylsar, 2025-03-24:
Same issue was reported in a Flutter camera library. 
https://github.com/juliansteenbakker/mobile_scanner/issues/698

They were able to resolve it by changing the image format from `YUV` to `NV21` before passing it to the MLKit.
https://github.com/juliansteenbakker/mobile_scanner/issues/1116

mauripavan, 2025-04-03:
@olivertylsar @mrousavy I'm having the same issue here. Specifying the format as a fix, worked well before upgrading RN version to 0.76 and react-native-vision-camera version. 

After the upgrade, my A23's users are experiencing a black screen as Oliver mentioned, and QR codes are only detected in the top right corner.

olivertylsar, 2025-04-10:
I tried these tips from https://github.com/mrousavy/react-native-vision-camera/issues/3248#issuecomment-2789562786 and in my case, just passing `photo={true}` to the `Camera` component fixes the issue. 

I'm not sure how exactly this prop impacts this issue, based on the prop annotation it should not be relevant, but it looks like it is working just fine for Samsung A23 - immediate QR detection anywhere in the frame + no black screen (no need to pass a format which otherwise still causes it). 

I haven't tested it on other devices yet since I just found out, but looks promising. 🤞

zwerg44, 2025-05-16:
Hello @olivertylsar,
I have the same issue after update from `4.5.3` to `4.6.4`
With `4.5.3` we used hardcoded `640x480` format for `A23`
Did `photo={true}` helped for you without hardcoding `640x480` format for `A23`, right?

olivertylsar, 2025-05-16:
@zwerg44 Passing `photo={true}` indeed fixed the issue for the A23 model. The `format` prop workaround didn't work for me and was causing black screen. 

But I recommend to use `photo={true}` only for this specific device, it didn't work well on iOS (the frame went black sometimes).

olivertylsar, 2025-06-17:
Same problem occurs on **Realme C67**. QR code is only detected in the top-right corner on this device. I assume the workaround will work for this device as well.

@zwerg44 I saw you listed all model names for the Samsung A23 here https://github.com/mrousavy/react-native-vision-camera/issues/3248#issuecomment-2420031911. Do you have a reference how to find out the actual model name for Realme C67 to include it into the problematic models without having the physical device? I'm thinking about using the `photo` prop for all android devices as these two might not be the only ones having this issue.

zwerg44, 2025-06-17:
Hi @olivertylsar 

I got the list from this flutter app: https://github.com/digitalfabrik/entitlementcard/pull/1524/files
I saw link to this PR mentioned in flutter camera lib issue discussion about A23.
And I suspect it's actually not the full list of A23 devices.

On my side I have an internal app, and I just have quite a lot of users with `SM-A236B`

I guess you can try:
- to dig device statistics from your firebase or other analytics
- google play console has device catalog
- or just google it

camboYY, 2026-01-30:
same problem with Hornor Play 10