margelo/react-native-vision-camera #3441: 🔧  react-native-vision-camera ios build error with react native 0.78

Loop2030, 2025-02-28:
### How were you trying to build the app?

I was getting below error while i am trying to run the react native app in ios simulator through xcode. i have tried many solutions but still facing same issue. I tried this as well: https://react-native-vision-camera.com/docs/guides/troubleshooting but no use.
please help me to resolve this issue.
npx u/react-native-community/cli info

System:
  OS: macOS 15.3.1
  CPU: (8) arm64 Apple M2
  Memory: 121.33 MB / 8.00 GB
  Shell:
    version: "5.9"
    path: /bin/zsh
Binaries:
  Node:
    version: 18.20.4
    path: ~/.nvm/versions/node/v18.20.4/bin/node
  Yarn:
    version: 1.22.22
    path: /opt/homebrew/bin/yarn
  npm:
    version: 10.7.0
    path: ~/.nvm/versions/node/v18.20.4/bin/npm
  Watchman:
    version: 2024.12.02.00
    path: /opt/homebrew/bin/watchman
Managers:
  CocoaPods:
    version: 1.16.2
    path: /Users/maheshmuttinti/.rbenv/shims/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 24.2
      - iOS 18.2
      - macOS 15.2
      - tvOS 18.2
      - visionOS 2.2
      - watchOS 11.2
  Android SDK:
    API Levels:
      - "30"
      - "33"
      - "34"
      - "35"
    Build Tools:
      - 31.0.0
      - 34.0.0
      - 35.0.0
    System Images:
      - android-35 | Google Play ARM 64 v8a
    Android NDK: Not Found
IDEs:
  Android Studio: 2024.2 AI-242.23339.11.2421.12550806
  Xcode:
    version: 16.2/16C5032a
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 17.0.13
    path: /opt/homebrew/opt/openjdk@17/bin/javac
  Ruby:
    version: 3.3.3
    path: /Users/maheshmuttinti/.rbenv/shims/ruby
npmPackages:
  "@react-native-community/cli":
    installed: 15.1.2
    wanted: 15.1.2
  react:
    installed: 19.0.0
[... 89 lines cut ...]
```

### VisionCamera Version

4.6.4

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Loop2030, 2025-02-28:
> Guten Tag, Hans here! 🍻
>
> You have provided quite a bit of detail, which is good! However, it looks like there are some build errors that need to be addressed. I suggest you try building your project [...]

[image]

Arnie098, 2025-03-03:
Same Issue Here

jeanraymonddaher, 2025-03-06:
is the library compatible with 0.78 in theory ? Thanks

kishanbh, 2025-06-05:
i'm also facing the same issue, any solution?