margelo/react-native-vision-camera #3444: 🐛 [unknown/unknown: Camera is closed.] when taking photos

francesco-clementi-92, 2025-03-04:
### What's happening?

I want to take a picture, show it and if something is not detected restore the camera.

### Reproduceable Code

```tsx
export default () => {
  const navigation = useNavigation<any>();
  const camera = useRef<Camera>(null);
  const isFocused = useIsFocused();
  const {hasPermission, requestPermission} = useCameraPermission();
  const [isInitialized, setIsInitialized] = useState(false);
  const [isTakingPhoto, setIsTakingPhoto] = useState(false);
  const [photoUri, setPhotoUri] = useState<string | null>(null);
  const device = useCameraDevice('back');

  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
  },[hasPermission, requestPermission]);

  const takePhoto = async () => {
    if (isInitialized && camera.current) {
      try {
        setIsTakingPhoto(true);
        const file = await camera.current.takePhoto();
        setPhotoUri(`file://${file.path}`);
        // if photo match, navigate away, otherwise throw error
        
      } catch (e) {
        setIsTakingPhoto(false);
        console.error(e);
      }
    }
  };

  if (isTakingPhoto && photoUri !== null) {
    return (<>
    <Image source={{uri: photoUri}} style={{flex: 1}} />
    <ActivityIndicator />
    </>);
  }

  return device !== undefined && hasPermission ? (
      <>
        <Camera
          ref={camera}
          onInitialized={() => setIsInitialized(true)}
          onStopped={() => console.error('Camera stopped')}
          onPreviewStopped={() => console.error('Camera preview stopped')}
          style={[StyleSheet.absoluteFill]}
          device={device}
          isActive={isFocused}
          photo={true}
          photoQualityBalance="speed"
        />
        <View
          style={styles.buttonContainer}
        >
          <Button onPress={takePhoto} style={styles.button}>
            <View />
          </Button>
        </View>
      </>
  ) : (
    <Card onPress={() => Linking.openSettings()}>
      <Card.Content>
        <Text style={styles.settingsTextColor} variant="titleLarge">
[... 158 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Redmi note 9 pro

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

phi0411141, 2025-04-13:
@francesco-clementi-92 I had Camera is closed issue and fixed it by not setting isActive={false} (pause) when taking the photo, alway let isActive={true}

cccolo, 2025-04-22:
Hi, we have the same issue in different Samsung devices... @francesco-clementi-92 do you have some fix yet?

francesco-clementi-92, 2025-04-22:
Hi @cccolo , the workaround I implemented is to add a dynamic key in the camera component and change it everytime I need to reuse it

cccolo, 2025-04-22:
> Hi [@cccolo](https://github.com/cccolo) , the workaround I implemented is to add a dynamic key in the camera component and change it everytime I need to reuse it

thanks for you response, do you have some example? thanks!

cccolo, 2025-04-22:
@francesco-clementi-92 something like this??

`<Camera
          key={isIdFront ? 'front' : 'back'}
....`

francesco-clementi-92, 2025-04-23:
> [@francesco-clementi-92](https://github.com/francesco-clementi-92) something like this??
> 
> `<Camera key={isIdFront ? 'front' : 'back'} ....`

Correct.

I used it like this:
```
const [cameraKey, setCameraKey] = useState(1);
...
  const reset = () => {
    setPhotoUri(null);
    setCameraKey((prev) => prev + 1);
  }
...
 <Camera
          ref={camera}
          key={`camera-${cameraKey}`}
```

cccolo, 2025-04-23:
@francesco-clementi-92 thanks!!!

flochtililoch, 2025-06-12:
I'm reproducing this error with the example app provided in this repo, on a Samsung S9+ (SM-G96U1) running Android 10.

@mrousavy Any idea what might be wrong? 
<details>
<summary>See a capture of the output of`adb logcat` right before/during/right after the error here</summary>

```
06-12 17:23:02.275 16694 19000 D OpenGLRenderer: mPreviousPosition == bounds[l=-42.00 t=84.00 r=1482.00 b=2793.00]
06-12 17:23:02.288  3663 19268 V LocSvc_HIDL_DebugReportService_jni: DebugReportService_getReport
06-12 17:23:02.288  3663 19268 D LocSvc_HIDL_DebugReportService_jni: DebugReportService_getReport, Get Latest 1 sec data.
06-12 17:23:02.288  3663 19268 V LocSvc_HIDL_DebugReportService_jni: [DebugReportService_getReport][942] [HC] =>> [HS]
06-12 17:23:02.288   824   875 E LocSvc_HIDL_DebugReportService: Invalid count: 0
06-12 17:23:02.288   824   875 E LocSvc_HIDL_DebugReportService: Invalid count: 0
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateEphemerisStatusListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateFixStatusListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateEpiReportListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateBestLocationListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateGpsTimeReportListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateXoStateReportListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateRfStateReportListObjV1_1
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateErrorRecoveriesListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populatePdrReporListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateSvHealthReportListObj
06-12 17:23:02.289  3663 19268 V LocSvc_HIDL_DebugReportService_jni: populateXtraReportListObj
06-12 17:23:02.292 16694 19000 D OpenGLRenderer: mPreviousPosition == bounds[l=-42.00 t=84.00 r=1482.00 b=2793.00]
06-12 17:23:02.298 16694 16694 I CameraView: invokeOnShutter(PHOTO)
06-12 17:23:02.298 16694 19034 D ExamplePlugin: 3840 x 2160 Image with format #35. Logging 5 parameters:
06-12 17:23:02.299 16694 19034 D ExamplePlugin:   -> 42.0 (java.lang.Double)
06-12 17:23:02.299 16694 19034 D ExamplePlugin:   -> true (java.lang.Boolean)
06-12 17:23:02.299 16694 19034 D ExamplePlugin:   -> {test=0.0, second=test} (java.util.HashMap)
06-12 17:23:02.299 16694 19034 D ExamplePlugin:   -> hello! (java.lang.String)
06-12 17:23:02.299 16694 19034 D ExamplePlugin:   -> [another test, 5.0] (java.util.ArrayList)
06-12 17:23:02.300 16694 19034 D ExampleKotlinPlugin: 3840 x 2160 Image with format #35. Logging 5 parameters:
06-12 17:23:02.300 16694 19034 D ExampleKotlinPlugin:   -> 42.0 (java.lang.Double)
06-12 17:23:02.300 16694 19034 D ExampleKotlinPlugin:   -> true (java.lang.Boolean)
06-12 17:23:02.300 16694 19034 D ExampleKotlinPlugin:   -> {test=0.0, second=test} (java.util.HashMap)
06-12 17:23:02.300 16694 19034 D ExampleKotlinPlugin:   -> hello! (java.lang.String)
06-12 17:23:02.300 16694 19034 D ExampleKotlinPlugin:   -> [another test, 5.0] (java.util.ArrayList)
06-12 17:23:02.302 16694 19038 I ReactNativeJS: 18155678778826: 3840x2160 yuv Frame (landscape-right)
06-12 17:23:02.305   911   911 W BufferQueueLayer: [SurfaceView - com.mrousavy.camera.example/com.mrousavy.camera.example.MainActivity@cc4976d@0#0] Timestamp 18155645184087 seems implausible relative to expectedPresent 12293909714750
06-12 17:23:02.307   849  1790 D SensorListener: Sensor(8) valid data is not coming yet!
06-12 17:23:02.307   849  1790 I SS_3A   :  INFO: AEC: TsAec_process_get_aec_info: 1422: [1][192]algo_out Av=1.168,g=70.367,e_t=0.033,LLS=0x7,Bv=-4.164,Ev=0.793,PEv=0.836,Cvgd=1,lux=414,FS=0,gam=0.0,lls=0x7,drc=1.000
06-12 17:23:02.309 16694 19000 D OpenGLRenderer: mPreviousPosition == bounds[l=-42.00 t=84.00 r=1482.00 b=2793.00]
06-12 17:23:02.322   832   894 I flicker_log: 17.29, 17.10, 30.65, 34.88
06-12 17:23:02.322   832   894 I flicker_log: 303, 34.88
06-12 17:23:02.322   832   894 I flicker_log: 396, 34.06
06-12 17:23:02.322   832   894 I flicker_log: 467, 33.74
06-12 17:23:02.322   832   894 D SensorsBio: BioSensor_common::readEvents, FAR = -1
06-12 17:23:02.325 16694 19000 D OpenGLRenderer: mPreviousPosition == bounds[l=-42.00 t=84.00 r=1482.00 b=2793.00]
06-12 17:23:02.337   849  1787 D SensorListener: Sensor(8) valid data is not coming yet!
06-12 17:23:02.338   849  1787 I SS_3A   :  INFO: AEC: TsAec_process_get_aec_info: 1422: [1][193]algo_out Av=1.168,g=70.367,e_t=0.033,LLS=0x7,Bv=-4.145,Ev=0.793,PEv=0.855,Cvgd=1,lux=414,FS=0,gam=0.0,lls=0x7,drc=1.000
06-12 17:23:02.338   911   911 W BufferQueueLayer: [SurfaceView - com.mrousavy.camera.example/com.mrousavy.camera.example.MainActivity@cc4976d@0#0] Timestamp 18155678778826 seems implausible relative to expectedPresent 12293943109750
06-12 17:23:02.341 16694 19000 D OpenGLRenderer: mPreviousPosition == bounds[l=-42.00 t=84.00 r=1482.00 b=2793.00]
06-12 17:23:02.358 16694 19000 D OpenGLRenderer: mPreviousPosition == bounds[l=-42.00 t=84.00 r=1482.00 b=2793.00]
06-12 17:23:02.363   824  1383 I LocSvc_ApiV02: <--- globalEventCb line 245 QMI_LOC_EVENT_NMEA_IND_V02
06-12 17:23:02.364   824  1383 I chatty  : uid=1000(system) LocApiMsgTask identical 2 lines
[... cut ...]
06-12 17:23:02.826  3281  4484 I CameraLatencyHistogram:      94.53   3.98   1.00   0.00   0.00   0.50   0.00   0.00   0.00   0.00 (%)
06-12 17:23:02.826  3281  4484 D Camera3-Device: disconnectImpl: mInterfaceLock.unlock()
06-12 17:23:02.826  3281  4484 D Camera3-Device: disconnectImpl: mInterfaceLock.lock()
06-12 17:23:02.826  3281  4484 D Camera3-Device: CameraPerf: close E
06-12 17:23:02.827 16694 19000 D OpenGLRenderer: mPreviousPosition == bounds[l=-42.00 t=84.00 r=1482.00 b=2793.00]
```
</details>

@francesco-clementi-92 can you detail under which condition you call  `reset()` in your example?

felipejh, 2025-06-19:
Same issue here. It happens only in one specific device. Two users have a Pixel 8 Pro. The issue happens only with one of them.

I've tried the workarounds above with no success.

phazei, 2026-02-12:
OMG! I solved it :D

Found two fixes for camera issues on Pixel 8 (likely affects other Pixels too):

---

**Black screen / HDR crash:** The format reports `supportsVideoHdr: true`, but the device can't use HDR concurrently with other use cases (photo + video). The error is `Unable to resolve supported dynamic range... Constrained set of concurrent dynamic ranges: SDR 8-bit only`.

**Fix:** only pass `videoHdr={true}` to the Camera component when the user explicitly enables it, rather than always passing the format's capability. Additionally, gate `photoHdr` and `videoHdr` to their respective capture types so they're never active simultaneously.

```tsx
// Before (always on based on format capability):
videoHdr={format?.supportsVideoHdr}
photoHdr={format?.supportsPhotoHdr}

// After (gated by capture type and user toggle):
videoHdr={format?.supportsVideoHdr && captureType === 'video' && controls.hdr}
photoHdr={format?.supportsPhotoHdr && captureType === 'photo' && controls.hdr}
```

---

**Night mode / lowLightBoost photo capture crash:** With `lowLightBoost` enabled and both `photo` and `video` use cases active, taking a photo throws `Capture request is cancelled because camera is closed`. The video streaming pipeline conflicts with the low-light photo capture.

**Fix:** disable the video use case when in photo mode with night mode enabled. The brief camera reconfiguration when toggling night mode is acceptable UX since it's a deliberate user action.

```tsx
// Before (both always active):
photo={photo}
video={video}

// After (scoped to active capture type, video disabled for night mode photos):
photo={photo && captureType === 'photo'}
video={video && !(captureType === 'photo' && controls.nightMode)}
```

---