margelo/react-native-vision-camera #3451: Fish-eye (ultra-wide-angle) not working in android RN 0.72

Maazy3, 2025-03-07:
### What's happening?

I am trying to set zoom to x0.5(ultra wide camera) as same as the one on native camera.But the neutral zoom is always 1. On mobile native camera it displays 0.5x and can zoom-out but on android it doesn't work.
I have read the documentation that says for 0.5x default will be greater than mini-zoom but in my case its always 1
 
I have tried on Samsung A52 ,one plus8. Its working fine on ios but not for android

Device & Environment Details:
Device: Samsung A52, OnePlus8
OS Version: Android 14
Camera Library: react-native-ivision-camera ( version 4.0.0-beta.13)
React Native Version: 0.72.6
targetSdkVersion = 34
minSdkVersion = 24

### Reproduceable Code

```tsx
const cameraRef = useRef<Camera | null>(null);
  const device = useCameraDevice('front', {
    physicalDevices: [
      'wide-angle-camera',
      'ultra-wide-angle-camera',
      'telephoto-camera',
    ],
  });

      <Camera
        ref={cameraRef}
        style={[styles.camera]}
        device={device!}
        resizeMode="cover"
        enableZoomGesture
        photo={true}
        zoom={zoom}
      />
```

### Relevant log output

```shell
None
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "limited",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -20,
  "name": "1 (FRONT) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung A-52, One Plus 8

### VisionCamera Version

4.0.0-beta.13

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Maazy3, 2025-03-08:
**UPDATED**

After some more debugging i have found that 

  const devices = Camera.getAvailableCameraDevices();
  const backCameras = devices.filter((d) => d.position === 'back');

my device returns ultra wide camera but the min zoom and neutral both are 1 which are incorrect

[
  {
     "formats":[],
     "hardwareLevel":"full",
     "hasFlash":true,
     "hasTorch":true,
     "id":"0",
     "isMultiCam":false,
     "maxExposure":20,
     "maxZoom":8,
     "minExposure":-20,
     "minFocusDistance":10,
     "minZoom":1,
     "name":"0 (BACK) androidx.camera.camera2",
     "neutralZoom":1,
     "physicalDevices":[
        "wide-angle-camera"
     ],
     "position":"back",
     "sensorOrientation":"landscape-left",
     "supportsFocus":true,
     "supportsLowLightBoost":false,
     "supportsRawCapture":false
  },
  {
     "formats":[],
     "hardwareLevel":"limited",
     "hasFlash":false,
     "hasTorch":false,
     "id":"2",
     "isMultiCam":false,
     "maxExposure":20,
     "maxZoom":8,
     "minExposure":-20,
     "minFocusDistance":0,
     "minZoom":1,
     "name":"2 (BACK) androidx.camera.camera2",
     "neutralZoom":1,
     "physicalDevices":[
        "ultra-wide-angle-camera"
     ],
     "position":"back",
     "sensorOrientation":"landscape-left",
     "supportsFocus":true,
     "supportsLowLightBoost":false,
     "supportsRawCapture":false
  }
]

 const device = getCameraDevice(devices2, 'back', {
    physicalDevices: [
      'ultra-wide-angle-camera',
      'wide-angle-camera',
      'telephoto-camera',
    ],
  });

and when i use devices with this it only returns the one with wide angle camera and not the ultra wide angle camera. I am not sure what i am missing.

{
   "formats":[],
   "hardwareLevel":"full",
   "hasFlash":true,
   "hasTorch":true,
   "id":"0",
   "isMultiCam":false,
   "maxExposure":20,
   "maxZoom":8,
   "minExposure":-20,
   "minFocusDistance":10,
   "minZoom":1,
   "name":"0 (BACK) androidx.camera.camera2",
   "neutralZoom":1,
   "physicalDevices":[
      "wide-angle-camera"
   ],
   "position":"back",
   "sensorOrientation":"landscape-left",
   "supportsFocus":true,
   "supportsLowLightBoost":false,
   "supportsRawCapture":false
}

yuborama, 2025-03-22:
> **UPDATED**
>
> After some more debugging i have found that
>
> const devices = Camera.getAvailableCameraDevices(); const backCameras = devices.filter((d) => d.position === 'back');
>
> my device returns [...]

I was reading old issues and Discord, if I understood correctly, it could be that Google is enabling the "ultra-wide-angle-camera" and "telephoto-camera" only for the native camera, making it impossible to access them.
https://github.com/mrousavy/react-native-vision-camera/issues/1511#issuecomment-1759419457

kerituni12, 2025-03-27:
The same issue on Iphone 11 Promax, even with demo ShadowLens app. It can't zoom to 0.5x
But if I using another camera photo app like Soda it can zoom to 0.5x.

rojuanmanuel, 2025-07-02:
I recently developed an application with React Native Vision Camera and I'm having the same problem on iOS 18. I've been testing on an iPhone 16 Pro Max, and although it has a 0.5x zoom, it always reaches 1x. This problem is repeated in the example application and even in its ShadowLens application.

lehoi2195, 2025-11-07:
Hi @Maazy3, just wanted to say a huge thanks 🙏 for sharing this solution!
I ran into the same issue where some Android devices didn’t show the wide-angle camera in react-native-vision-camera, and your approach worked perfectly.
I’ve also optimized your code into a lighter version for anyone who wants a more minimal implementation — keeping your original logic but trimming unnecessary overhead.
Really appreciate your contribution, it saved me a lot of time! 🚀

```jsx
import { useIsFocused } from '@react-navigation/native'
import { useRef, useState } from 'react'
import { Platform, SafeAreaView, StyleSheet, Text, TouchableOpacity, View, Dimensions } from 'react-native'
import { Camera, CameraDevice, getCameraDevice, useCameraDevice } from 'react-native-vision-camera'

const isAndroid = Platform.OS === 'android'
const isIOS = Platform.OS === 'ios'
const SCREEN_W = Dimensions.get('window').width

const CameraScreen: React.FC<any> = () => {
  const cameraRef = useRef<Camera | null>(null)
  const isFocused = useIsFocused()
  const isForeground = true

  const isActive = isFocused && isForeground
  const physicalCameras = Camera.getAvailableCameraDevices()
  const backCameras = physicalCameras?.filter((d) => d.position === 'back')
  let supportedDevice: CameraDevice | undefined = useCameraDevice('back', {
    physicalDevices: ['wide-angle-camera', 'ultra-wide-angle-camera', 'telephoto-camera'],
  })
  const hasWideAngleSupport = supportedDevice?.physicalDevices.includes('ultra-wide-angle-camera')
  let ultraWideCamera: CameraDevice | undefined
  let ultraWideDevice: CameraDevice | undefined

  if (isAndroid && !hasWideAngleSupport) {
    ultraWideCamera = backCameras?.find((item) => item?.physicalDevices.includes('ultra-wide-angle-camera'))
  }
  if (ultraWideCamera && isAndroid) {
    ultraWideDevice = getCameraDevice([ultraWideCamera], 'back')
  }
  const { minZoom, maxZoom, neutralZoom } = supportedDevice || {}
  const ultraWideZoom = ultraWideCamera && isAndroid ? ultraWideCamera?.minZoom : minZoom
  const defaultZoom = neutralZoom
  const zoom2xFactor = isIOS ? 0.125 : 0.198
  const zoom2x = Math.min(neutralZoom! * 10, maxZoom!) * zoom2xFactor
  const [zoom, setZoom] = useState(defaultZoom)
  const [isUltraWideSelected, setIsUltraWideSelected] = useState(false)

  const zoomLevels = Platform.select({
    ios: [
      ...(hasWideAngleSupport ? [{ zoomValue: ultraWideZoom, label: 0.5 }] : []),
      { zoomValue: defaultZoom, label: 1 },
      { zoomValue: zoom2x, label: 2 },
    ],
    android: [
      ...(ultraWideCamera || hasWideAngleSupport
        ? [
            {
              zoomValue: hasWideAngleSupport ? ultraWideZoom : 0.5,
              label: hasWideAngleSupport ? ultraWideZoom?.toFixed(1) : 0.5,
            },
          ]
        : []),
      { zoomValue: defaultZoom, label: 1 },
      { zoomValue: zoom2x, label: 2 },
      // { zoomValue: zoom2x < 2 ? 3 : 2, label: zoom2x < 2 ? 3 : 2 },
    ],
  })

  const handleZoomChange = (zoomValue: number | string) => {
    if (isAndroid && ultraWideCamera && Number(zoomValue) < 1) {
      setIsUltraWideSelected(true)
    } else if (!hasWideAngleSupport) {
      setIsUltraWideSelected(false)
    }
    setZoom(Number(zoomValue) || defaultZoom)
  }

  const renderCameraView = () => {
    const cameraDevice = isAndroid && isUltraWideSelected && ultraWideCamera ? ultraWideDevice : supportedDevice

    if (!cameraDevice) {
      return (
        <View style={styles.noCameraView}>
          <Text>Camera not support</Text>
        </View>
      )
    }
    return (
      <View>
        <View>
          <Camera
            zoom={zoom}
            photo={true}
            ref={cameraRef}
            enableZoomGesture
            resizeMode="cover"
            isActive={isActive}
            style={styles.camera}
            device={cameraDevice!}
          />
        </View>
        <View style={styles.zoomOptionsView}>
          {zoomLevels?.map((item: any) => (
            <TouchableOpacity onPress={() => handleZoomChange(item?.zoomValue)} style={styles.zoomOption}>
              <Text>{item?.label}x</Text>
            </TouchableOpacity>
          ))}
        </View>
      </View>
    )
  }

  return (
    <SafeAreaView style={styles.safeAreaView}>
      <View>{renderCameraView()}</View>
    </SafeAreaView>
  )
}
const styles = StyleSheet.create({
  safeAreaView: {
    flex: 1,
    width: '100%',
    height: '100%',
    paddingTop: 20,
    backgroundColor: '#000',
  },
  camera: {
    width: SCREEN_W,
    height: SCREEN_W * (4 / 3),
    overflow: 'hidden',
  },
  noCameraView: {
    height: 20,
    justifyContent: 'center',
    alignItems: 'center',
    paddingHorizontal: 20,
  },
  zoomOptionsView: {
    alignSelf: 'center',
    display: 'flex',
    flexDirection: 'row',
    columnGap: 20,
    marginBottom: 20,
  },
  zoomOption: {
    backgroundColor: '#fff',
    borderRadius: 20,
    width: 40,
    height: 40,
    justifyContent: 'center',
    alignItems: 'center',
    marginHorizontal: 12,
  },
})
export default CameraScreen
```
"react-native-vision-camera": "4.7.2",
"react-native": "0.78.3",