margelo/react-native-vision-camera #3456: ✨ [Question] Is it possible to use community frame processor plugins on still images?

tuncaulubilge, 2025-03-12:
### What feature or enhancement are you suggesting?

This library has a great community around it that built various plugins based on MLKit, such as background removal, object detection, text detection etc. Is it possible to utilise these plugins on still images as well?

### What Platforms whould this feature/enhancement affect?

Android, iOS

### Alternatives/Workarounds

No existing workarounds apart from building a custom native module

### Additional information

- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2025-03-14:
maybe after the nitro rewrite

mrousavy, 2025-03-14:
then every Image can be a Frame