margelo/react-native-vision-camera #3481: 🐛 Camera crashes when rendered is second time when iPad app launched on Mac

kacienk, 2025-04-01:
### What's happening?

I tried to launch my RN application on my MacBook via Xcode using target: My Mac (Designed for iPad). I use built-in camera for scanning QR codes in my application. When I open the screen where the `<Camera>` is rendered, the camera works as expected. However, the app crashes when I navigate to a different screen and then re-open the screen with the  `<Camera>`. This issue does not occur on mobile devices.

I believe the issue is caused because of this line https://github.com/mrousavy/react-native-vision-camera/blob/b982ecacea1d2a7cc521948f622cbc497b6d9856/package/ios/Core/CameraSession.swift#L153

When I commented out this line and rebuilt the project, the application worked as expected. The reason might be that tracking and setting device orientation is not applicable for MacBooks.

### Reproduceable Code

```tsx
const qrReader = ({ hasPermission }: { hasPermission: boolean }) => {
  const camera = useRef<Camera>(null);
  const device = useCameraDevice('back');
  const codeScanner = useCodeScanner({
    /**/
  });

  // whether app has permission tousee camera
  return (
    <View>
      {!!device && !!hasPermission && (
        <Camera
          ref={camera}
          codeScanner={codeScanner}
          device={device}
          isActive
        />
      )}
    </View>
  );
};
```

### Relevant log output

```shell
Unfortunately, for some reason Xcode started to give me (afaik) unrelated stack traces when I launch the camera for the first time. However, launching the application not via Xcode, but directly from the dock, allows to go through the scenario described above and observe the crash when I re-open the screen with camera.
```

### Camera Device

```json
{
  "neutralZoom": 1,
  "isMultiCam": false,
  "minExposure": -8,
  "name": "Kamera HD FaceTime",
  "hasFlash": false,
  "id": "...",
  "hasTorch": false,
  "minFocusDistance": 0,
  "hardwareLevel": "full",
  "formats": [],
  "maxZoom": 67.5,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "minZoom": 1,
  "supportsFocus": false,
  "sensorOrientation": "portrait",
  "position": "back",
  "maxExposure": 8,
  "supportsRawCapture": false,
  "supportsLowLightBoost": false
}
```

### Device

MacBook M1 macOS 15.3.2

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [x] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

angelos3lex, 2025-07-16:
We also have similar crash on macbooks only on released build (macOS 17.5.0):

[image]

imamrobani, 2025-08-02:
same issue, here

mrousavy, 2026-04-21:
Hey! Sorry for the late reply and thank you for your issue report.
I was hyper-focused the entire winter to build the next major version of VisionCamera - and I think [VisionCamera V5](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v5.0.0) fixes this! 🥳

Closing, but feel free to open a new issue if this still needs attention in V5! :)