margelo/react-native-vision-camera #3488: 🐛 Error While Updating Property 'codeScannerOptions'

shoxter, 2025-04-05:
### What's happening?

When trying to launch the Camera component while trying to scan barcodes, I'm getting the following error:

[image]

### Reproduceable Code

```tsx
const cameraDevice = useCameraDevice('back');
const codeScanner = useCodeScanner({
	codeTypes: ['code-128', 'code-39', 'code-93', 'codabar', 'ean-13', 'ean-8', 'itf', 'itf-14', 'upc-e', 'upc-a', 'qr', 'pdf-417', 'aztec', 'data-matrix'],
	onCodeScanned: (codes) => {
		console.log(`Scanned ${codes.length} codes!`)
	}
});

<Camera
	style={{ flex: 1 }}
	device={cameraDevice}
	isActive={true}
	codeScanner={codeScanner}
/>

codeScanner =
 {"codeTypes": ["code-128", "code-39", "code-93", "codabar", "ean-13", "ean-8", "itf", "itf-14", "upc-e", "upc-a", "qr", "pdf-417", "aztec", "data-matrix"], "onCodeScanned": [Function anonymous]}
```

### Relevant log output

```shell
It errors out before starting so I'm not sure if this is something I should be trying to do.
```

### Camera Device

```json
{
  "name": "0 (BACK) androidx.camera.camera2",
  "maxExposure": 9,
  "hasFlash": true,
  "id": "0",
  "position": "back",
  "isMultiCam": false,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "maxZoom": 10,
  "sensorOrientation": "landscape-left",
  "minFocusDistance": 5,
  "supportsRawCapture": false,
  "hasTorch": true,
  "supportsLowLightBoost": false,
  "formats": [],
  "supportsFocus": true,
  "minZoom": 1,
  "neutralZoom": 1,
  "minExposure": -9,
  "hardwareLevel": "full"
}
```

### Device

Android Emulator & Pixel 9 Pro (both fail same error)

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [x] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

shoxter, 2025-04-05:
Updated with ADB logs.

truongtv22, 2025-05-05:
Same issue. Is there a solution yet?

Jamal-Ud-Din-hub, 2025-06-21:
+1 with type `gs1-data-bar`

PABourdais, 2025-10-28:
> Getting the same thing with any of the `gs1` variants.
> 
> `gs1` variants aren't supported by [MLKit](https://developers.google.com/android/reference/com/google/mlkit/vision/barcode/common/Barcode) which is what this package uses. The following scanner types seem to be iOS only:
> 
> * gs1-data-bar
> * gs1-data-bar-limited
> * gs1-data-bar-expanded
> * itf-14
> 
> Could we update the docs to let users know this is the case?

Removing itf-14 on Android fix the issues.