margelo/react-native-vision-camera #3491: 🐛 [session/invalid-output-configuration: Failed to configure the Camera Session because the output/stream configurations are invalid!]

devanshisrivastava20, 2025-04-07:
### What's happening?

Hi I am using this library in my React Native project for barcode scanning only. 

I am getting this error intermittently from the library specifically on Android after I launch the Camera and grant access. Earlier it was happening continuously but after fixing it with the App Background/Foreground and ref management logic it started working once the camera is open. 
I have 3 points to launch this camera - 1) App quick link 2) Home Page Icon for Barcode scanner 3) Icon shown on Search Field when its in focus

I can add more details as needed

### Reproduceable Code

```tsx
import React, { useState, useRef, useEffect } from 'react';
import { useIsFocused } from '@react-navigation/native';
import { Button, Header, withTheme } from 'react-native-elements';
import { View, AppState, Platform, Text, Linking, StyleSheet } from 'react-native';
import { useCameraDevice, Camera, useCodeScanner, useCameraPermission } from 'react-native-vision-camera';

import Styles from './ScanBarcode.style';
import WithLoader from '../hoc/loader';

export const isAppInBackgroundMode = (appState: any) => {
  if (appState != null && typeof appState.current === 'string' && appState?.current?.match(/inactive|background/)) {
    return true;
  }
  return false;
};

export const ScanBarcode = (props) => {
  const {
    theme,
    navigation,
    searchResultData,
  } = props;

  const {
    messageWrapper,
    scanMessage,
    scanArea,
    blankArea,
    sideBlankArea,
    middleArea,
    middleAreaContainer,
  } = Styles(theme);

  const device = useCameraDevice('back');
  const permissionRequested = useRef(false);

  const { hasPermission, requestPermission } = useCameraPermission();

  const [scanned, setScanned] = useState(false);

  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
    if (hasPermission) {
      requestPermission();
    }
  }, []);

  const isFocused = useIsFocused();
  const appState = useRef(AppState.currentState);

  const _handleAppStateChange = (nextAppState: any) => {
    appState.current = nextAppState;
    const isBackground = isAppInBackgroundMode(appState);
    if (!isBackground && !permissionRequested.current) {
      requestPermission();
      permissionRequested.current = true;
    }
  };

  React.useEffect(() => {
    const listener = AppState.addEventListener('change', _handleAppStateChange);
    return () => {
      listener?.remove();
    };
  }, []);

  React.useEffect(() => {
    if (searchResultData) {
      //code
    }
  }, [searchResultData]);

  const handleBarCodeScanned = useCodeScanner({
    codeTypes: [
      'code-128',
      'code-39',
      'code-93',
      'codabar',
      'ean-13',
      'ean-8',
      'itf',
      'upc-e',
      'upc-a',
      'qr',
      'pdf-417',
      'aztec',
      'data-matrix',
    ],
    onCodeScanned: codes => {
      setScanned(true);
      const code = codes && codes[0];
      let data: string = code.value;
      if (isValidBarcode(data)) {
        data = data?.length === 12 ? `0${data}` : data;
        setScanned(true);
        //code
      } else {
       //code
      }
    },
  });

  if (hasPermission === null) {
    return <View />;
  }

  if (!isFocused) {
    return null;
  }

  const handleOpenSetting = () => {
    navigation.pop();
    Linking.openSettings();
  };

  const isAndroid = Platform.OS === 'android';
  const isRefForAndroid = isAndroid ? permissionRequested.current : true;
  const shouldShowCamera = hasPermission === true ? isRefForAndroid : false;

  return (
    <View style={{ flex: 1 }}>
      <Header />
      {shouldShowCamera ? (
        <React.Fragment>
          <View style={messageWrapper}>
            <Text style={scanMessage}>
              {'scanBarcodeScreenMessage'}
            </Text>
          </View>
          <View style={{ flex: 1 }}>
            <Camera
              device={device}
              style={StyleSheet.absoluteFill}
              isActive={true}
              codeScanner={scanned ? undefined : handleBarCodeScanned}
            />
            <View style={scanArea}>
              <View style={blankArea} />
              <View style={middleAreaContainer}>
                <View style={sideBlankArea} />
                <View style={middleArea} />
                <View style={sideBlankArea} />
              </View>
              <View style={blankArea} />
            </View>
          </View>
        </React.Fragment>
      ) : null}

      {!hasPermission ? (
        <View style={messageWrapper}>
          <Text style={scanMessage}>{'noCameraAccess'}</Text>
          <Button
            title={'openSettings'}
            type='clear'
            onPress={() => handleOpenSetting()}
          />
        </View>
      ) : null}
    </View>
  );
};

export default withTheme(WithLoader(ScanBarcode));
```

### Relevant log output

```shell
{ [session/invalid-output-configuration: Failed to configure the Camera Session because the output/stream configurations are invalid!]
                                                                                                      name: 'session/invalid-output-configuration',
                                                                                                      _code: 'session/invalid-output-configuration',
                                                                                                      _message: 'Failed to configure the Camera Session because the output/stream configurations are invalid!',
                                                                                                      _cause: 
                                                                                                       { stacktrace: 'java.util.concurrent.TimeoutException: Future[androidx.camera.core.impl.utils.futures.ListFuture@cbbc556] is not done within 5000 ms.\n\tat androidx.camera.core.impl.utils.futures.Futures.lambda$makeTimeoutFuture$1(Futures.java:437)\n\tat androidx.camera.core.impl.utils.futures.Futures$$ExternalSyntheticLambda7.call(D8$$SyntheticClass:0)\n\tat androidx.camera.core.impl.utils.executor.HandlerScheduledExecutorService$HandlerScheduledFuture.run(HandlerScheduledExecutorService.java:243)\n\tat android.os.Handler.handleCallback(Handler.java:958)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loopOnce(Looper.java:257)\n\tat android.os.Looper.loop(Looper.java:368)\n\tat android.os.HandlerThread.run(HandlerThread.java:67)\n',
                                                                                                         message: 'Future[androidx.camera.core.impl.utils.futures.ListFuture@cbbc556] is not done within 5000 ms.' } }
```

### Camera Device

```json
{
  "hardwareLevel": "full",
  "hasFlash": true,
  "hasTorch": true,
  "id": "0",
  "isMultiCam": false,
  "maxExposure": 12,
  "maxZoom": 10,
  "minExposure": -12,
  "minFocusDistance": 8,
  "minZoom": 1,
  "name": "0 (BACK) androidx.camera.camera2",
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "position": "back",
  "sensorOrientation": "landscape-left",
  "supportsFocus": true,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false
}
```

### Device

Android 11 - Samsung M13, Android 14 - OnePlus 9R

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

lc3t35, 2025-04-08:
I have the same issue with 4.6.4 on android (works fine on ios), expo 52.0.41, "react-native": "0.76.7",

devanshisrivastava20, 2025-04-11:
I made some changes to the code—now the issue only occurs the first time Camera asks for permission, but everything works fine after that.

**Specific to Android**

export const useAppStateListener = () => {
  const appState = useRef(AppState.currentState);
  const [appStateVisible, setAppStateVisible] = React.useState(appState.current);
  const _handleAppStateChange = (nextAppState: any) => {
    appState.current = nextAppState;
    setAppStateVisible(appState.current);
  };
  React.useEffect(() => {
    const listener = AppState.addEventListener('change', _handleAppStateChange);
    return () => {
      listener.remove();
    };
  }, []);
  return appStateVisible;
};

export const ScanBarcode = (props) => {
  const {
    theme,
    navigation,
    searchResultData,
  } = props;

  const device = useCameraDevice('back');
  const { hasPermission, requestPermission } = useCameraPermission();

  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
  }, []);

const isFocused = useIsFocused();
const currentAppState = useAppStateListener();

  return (
    <View style={{ flex: 1 }}>
      <Header />
      {hasPermission ? (
        <React.Fragment>
          <View style={messageWrapper}>
            <Text style={scanMessage}>
              {'scanBarcodeScreenMessage'}
            </Text>
          </View>
          <View style={{ flex: 1 }}>
            <Camera
              device={device}
              style={StyleSheet.absoluteFill}
              isActive={currentAppState === 'active'}
              codeScanner={scanned ? undefined : handleBarCodeScanned}
            />
           
        </React.Fragment>
      ) : null}
);
}

lc3t35, 2025-04-12:
I was using Camera.getCameraPermissionStatus / requestCameraPermission and for Microphone -> moved to useCameraPermission and useMicrophonePermission
checking both in the initial useEffect

```
useEffect(() => {
    if (!hasCameraPermission) {
      requestCameraPermission();
    }
    if (!hasMicrophonePermission) {
      requestMicrophonePermission();
    }
}, []), 
```

no more crash !
Thank you @devanshisrivastava20 !

aron-skaleet, 2025-05-04:
Hello guys, I got the same message

 ERROR  [session/invalid-output-configuration: Failed to configure the Camera Session because the output/stream configurations are invalid!]

Because I copy the same code in my app

```
  return (
    <SafeAreaView style={styles.container}>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        frameProcessorFps={2}
        {...props}
        codeScanner={isScanning ? codeScanner : undefined}
      />
      <View style={styles.infoContainer}>
        <Text style={styles.infoText}>Point the camera at a code</Text>
      </View>
    </SafeAreaView>
  );
```

But after removing the <SafeAreaView style={styles.container}> and </SafeAreaView> the camera works and the app scans correctly.
Note: I remove the SafeAreaView because my component is already in an existing <View></View> code.

I hope it can help you !

brunnosena, 2025-07-23:
👀

aehp20, 2025-07-23:
> [@aron-skaleet](https://github.com/aron-skaleet) "But after removing the .... ", removing what please ?

Hello @lc3t35 
I have removed ```<SafeAreaView style={styles.container}>```in my original code because my component is already contained for another View parent component.
So finally, my code is:
```
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={isActive && isFocused && appState === "active"}
      torch={flashOn ? "on" : "off"}
      codeScanner={isScanning ? codeScanner : undefined}
    />
  );
```
I hope it can help you.

joaquinvaz, 2025-08-04:
If you have this issue, check the re-mounting, just isolate the component and try from there, we had this issue and I resolved it with that.
For example we had it conditionally display based on a value. We removed that conditional display and it worked correctly from there. You can make use of isActive prop too.

joaquinvaz, 2025-08-04:
@NoahVandy just tried it and it's working fine, didn't apply the video functionality but I put the flag

in styles I have just height set up
try:
```
{
height: some number
}
``

```
      <Camera
        style={styles.container}
        codeScanner={codeScanner}
        device={device}
        isActive={true}
        photo={false}
        video={true}
        torch={props.flash ? 'on' : 'off'} />
```

ginixsan, 2025-12-07:
I tried using https://github.com/mrousavy/react-native-vision-camera/issues/2862#issuecomment-2604144751 and it seems to fix the problem