margelo/react-native-vision-camera #3496: 🐛 On app install `device` is null and `<Camera />` is stick on the top-left corner!

mgulfam0722, 2025-04-15:
### What's happening?

When app is first installed, firstly, the `device` returned from `useCameraPermission()` hook is `undefined` and upon reload the expo server, the `device` variable is there however the `<Camera />` is stick on top-left corner of the screen.

**1. When app is installed after uninstall**
[image]

**2. When I hit `r` to reload the expo server, the `device` in not undefined anymore, but the `<Camera />` is stick on the top-left corner of the screen, no idea why.**
[image]

### Reproduceable Code

```tsx
import { layout } from '@/styles/common';
import { Button, Dimensions, Image, StyleSheet, Text, View } from 'react-native';
import { useRouter } from 'expo-router';
import { useRef } from 'react';
import { Camera, useCameraDevice, useCameraPermission } from 'react-native-vision-camera';
import Svg, { Mask, Rect } from 'react-native-svg';

const { width } = Dimensions.get('window');
const frameX = (width - (width * 0.9)) / 2;
const frameY = (300 - 300 * 0.9) / 2;

export default function Scan() {
    const router = useRouter();
    const { hasPermission, requestPermission } = useCameraPermission()
    const cameraRef = useRef<Camera>(null);
    const device = useCameraDevice('back');
    console.log(JSON.stringify(device, (k, v) => k === "formats" ? [] : v, 2))
    return (
        <View style={layout.fill}>
            <View style={styles.darkContainer}>
                <View style={styles.instructionContainer}>
                    <Text style={styles.cameraScanText}>Camera scan</Text>
                    <Text style={styles.instructionText}>Hold your card under camera to scan</Text>
                </View>
                <View style={styles.cameraContainer}>
                    {!hasPermission ? (
                        <View>
                            <Text style={styles.permissionPromptText}>
                                We need your permission to show the camera
                            </Text>
                            <Button onPress={requestPermission} title="grant permission" />
                        </View>
                    ) : (
                        device && (
                            <View
                                style={{
                                    flex: 1,
                                }}
                            >
                                <Camera
                                    style={StyleSheet.absoluteFill}
                                    ref={cameraRef}
                                    device={device}
                                    isActive={true}
                                    resizeMode="cover"
                                    photo={true}
                                />

                                {/* Masked Overlay */}
                                <Svg style={StyleSheet.absoluteFill}>
                                    <Mask id="mask">

                                        {/* This covers the whole screen in white. White in a mask means “keep this part visible.” */}
                                        <Rect
                                            x="0"
                                            y="0"
                                            width={'100%'}
[... 355 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```

### Device

Android Pixel Emulator

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [x] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ngxhuyhoang, 2025-04-17:
Same issue

ngxhuyhoang, 2025-04-21:
@mgulfam0722 
Do you have any temp solution while wait this bug to be fixed ? I have same issue

mgulfam0722, 2025-04-21:
Yes, so for the `<Camera />` UI fix is to set `overflow: 'hidden'` in the parent view of `<Camera />` component and use a physical device for `null` device error. The device `null` error only seems to be present in android emulators not physical device.

Let me know if you need any help.

Cheers.

mgulfam0722, 2025-04-21:
@ngxhuyhoang just in-case, below is my updated code which solves the UI issue of the `<Camera />` component.

```
import { layout } from '@/styles/common';
import { Button, Dimensions, StyleSheet, Text, View } from 'react-native';
import { useRouter } from 'expo-router';
import { useRef } from 'react';
import { Camera, useCameraDevice, useCameraPermission } from 'react-native-vision-camera';
import Svg, { Mask, Rect } from 'react-native-svg';

const { width } = Dimensions.get('window');
const frameX = (width - (width * 0.9)) / 2;
const frameY = (300 - 300 * 0.9) / 2;

export default function Scan() {
    const router = useRouter();
    const { hasPermission, requestPermission } = useCameraPermission()
    const cameraRef = useRef<Camera>(null);
    const device = useCameraDevice('back');
    console.log(JSON.stringify(device, (k, v) => k === "formats" ? [] : v, 2))
    return (
        <View style={layout.fill}>
            <View style={styles.darkContainer}>
                <View style={styles.instructionContainer}>
                    <Text style={styles.cameraScanText}>Camera scan</Text>
                    <Text style={styles.instructionText}>Hold your card under camera to scan</Text>
                </View>
                <View style={styles.cameraContainer}>
                    {!hasPermission ? (
                        <View>
                            <Text style={styles.permissionPromptText}>
                                We need your permission to show the camera
                            </Text>
                            <Button onPress={requestPermission} title="grant permission" />
                        </View>
                    ) : (
                        device && (
                            <View
                                style={{
                                    flex: 1,
                                    overflow: 'hidden'
                                }}
                            >
                                <Camera
                                    style={StyleSheet.absoluteFill}
                                    ref={cameraRef}
                                    device={device}
                                    isActive={true}
                                    resizeMode="cover"
                                    photo={true}
                                />

                                {/* Masked Overlay */}
                                <Svg style={StyleSheet.absoluteFill}>
                                    <Mask id="mask">

                                        {/* This covers the whole screen in white. White in a mask means “keep this part visible.” */}
                                        <Rect
                                            x="0"
                                            y="0"
                                            width={'100%'}
                                            height={'100%'}
                                            fill="white"
                                        />

                                        {/* This draws a black rounded rectangle in the middle-ish of the screen. Black in a mask means “cut a hole here.” */}
                                        <Rect
                                            x={frameX}
                                            y={frameY}
                                            width={'90%'}
                                            height={'90%'}
                                            rx={10}
                                            ry={10}
                                            fill="black"
                                        />
                                    </Mask>
                                    <Rect
                                        width={'100%'}
                                        height={'100%'}
                                        fill="#000000C7"
                                        mask="url(#mask)"
                                    />
                                </Svg>
                            </View>
                        )
                    )}
                </View>
                <View style={styles.guidePromptContainer}>
                    <View
                        style={{
                            flexDirection: 'row',
                            justifyContent: 'center',
                            alignItems: 'center',
                        }}
                    >
                        <Text style={[styles.bodyText, { color: 'white' }]}>Point on the </Text>
                        <Text style={[styles.bodyText, { color: '#2E9ECE' }]}>Emirates ID</Text>
                    </View>
                </View>
            </View>
        </View>
    );
}

const styles = StyleSheet.create({
    darkContainer: {
        backgroundColor: '#000000C7',
        flex: 1,
        borderRadius: 10,
        marginTop: 20,
        justifyContent: 'space-evenly',
        alignItems: 'center',
    },
    instructionContainer: {
        justifyContent: 'center',
        alignItems: 'center',
        flex: 1,
    },
    cameraScanText: {
        fontSize: 20,
        color: '#2BABA1',
        marginTop: 20,
    },
    instructionText: {
        color: 'white',
    },
    permissionPromptText: {
        color: 'white',
        marginBottom: 10,
    },
    bodyText: {
    },
    flashPromptContainer: {
        flexDirection: 'row',
        justifyContent: 'center',
        alignItems: 'center',
    },
    camera: {
        flex: 1,
    },
    guidePromptContainer: {
        justifyContent: 'space-between',
        paddingVertical: 10,
        alignItems: 'center',
        flex: 1,
    },
    cameraContainer: {
        height: 290,
        width: '100%',
    },
});

```

LPitonakova, 2025-04-30:
I am experiencing the same issue on a real device (Google Pixel 5, with the latest OS). The camera view simply ignores the Y position of the parent container. It also ignores "top" when I try to set its style as position: "absolute", top: 300. This did not happen in previous versions of this library. I have just upgraded to the latest version.

navtoj, 2025-07-19:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/3237

TheFifthDivision, 2026-08-27:
For anyone still on v4 hitting the `device` = null / empty `getAvailableCameraDevices()` part of this issue — we root-caused it in v4.7.3 and it's a startup race in `CameraDevicesManager.kt` (Android):

1. The constructor kicks off `ProcessCameraProvider`/`ExtensionsManager` initialization **asynchronously** on a camera executor.
2. `getConstants()` runs when JS first touches the module — usually **before** that init finishes — and `getDevicesJson()` early-returns an empty array whenever `cameraProvider`/`extensionsManager` are still null.
3. `initialize()` emits one `CameraDevicesChanged` event, but that also happens before init completes (and the `AvailabilityCallback` never re-fires because `deviceIds` is pre-seeded with `cameraManager.cameraIdList`).
4. Nothing re-emits after "Successfully initialized!", so JS keeps the empty list forever.

That's why emulators reproduce it ~100% (slow CameraX init) and physical devices only occasionally (cold start right into a camera screen).

One-line fix — re-emit once the provider is actually ready:

```diff
         Log.i(TAG, "Initializing ExtensionsManager...")
         extensionsManager = ExtensionsManager.getInstanceAsync(reactContext, cameraProvider!!).await(executor)
         Log.i(TAG, "Successfully initialized!")
+        sendAvailableDevicesChangedEvent()
       } catch (error: Throwable) {
```

Verified on react-native-vision-camera 4.7.3 (RN 0.81 / new arch interop, Expo SDK 54): with the patch, `useCameraDevice('back')` resolves as soon as init completes on an emulator that previously always returned an empty device list. Easy to apply via patch-package until you can move to v5.