margelo/react-native-vision-camera #3498: 🐛 Bug: CameraView Event Name Errors & QR Code Scan Timeout after Upgrading to RN 0.78.2

sanjayradadiya, 2025-04-16:
### What's happening?

After upgrading React Native and related packages, the app throws multiple event name errors related to CameraView. Additionally, QR code scanning fails with an ImageReader timeout.

🧩 Current Setup (After Upgrade)
```
{
  "react": "19.0.0",
  "react-native": "0.78.2",
  "react-native-vision-camera": "4.6.4",
  "react-native-worklets-core": "1.5.0"
}
```

🕐 Previous Working Setup
```
{
  "react": "18.1.0",
  "react-native": "0.71.1",
  "react-native-vision-camera": "3.8.2",
  "react-native-worklets-core": "0.3.0"
}
```

### Reproduceable Code

```tsx
Create a new React Native 0.78.2 project and setup react-native-vision-camera Run the app — the error appears immediately.
```

### Relevant log output

```shell
Android logcat

E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraStarted'->'onStarted'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraError'->'onError'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraViewReady'->'onViewReady'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraAverageFpsChanged'->'onAverageFpsChanged'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraShutter'->'onShutter'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraOutputOrientationChanged'->'onOutputOrientationChanged'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraInitialized'->'onInitialized'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraCodeScanned'->'onCodeScanned'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraPreviewOrientationChanged'->'onPreviewOrientationChanged'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraStopped'->'onStopped'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraPreviewStarted'->'onPreviewStarted'
E  Direct event name for 'CameraView' doesn't correspond to the naming convention, expected 'topEventName'->'onEventName', got 'topCameraPreviewStopped'->'onPreviewStopped

when attempting to scan a QR code:

[ImageReader-640x480f23m4-29343-0](id:729f00000003,api:4,p:1157,c:29343) waitForFreeSlotThenRelock: timeout [mDequeueTimeout=1000ms]
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

samsung SM-M146B

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sanjayradadiya, 2025-04-16:
Thank you for the reply!
The VisionCamera example uses an older version of React Native (0.75.4).

MarcelinaZiolkiewicz, 2025-04-30:
Same issue here:

{
  "react": "18.3.1",
  "react-native": "0.76.7",
  "react-native-vision-camera": "4.6.3",
}

CaptainJeff, 2025-08-20:
Any update on this? Getting the same issue