margelo/react-native-vision-camera #3502: 🐛 Build fails with CMake error: Target "ReactAndroid::reactnativejni" not found when using EAS (Expo) and VisionCamera

oussamakh96, 2025-04-23:
### What's happening?

When building the app using eas build -p android, the build fails at the CMake configuration step for react-native-vision-camera. The error indicates that ReactAndroid::reactnativejni is not found during add_library() in CMakeLists.txt.

Can you provide an official workaround or patch to correctly link ReactAndroid::reactnativejni when using VisionCamera with CMake?

"react-native": "0.76.9",
"expo": "^52.0.0",
"react-native-vision-camera": "4.5.2",
eas-cli/16.3.2   
 win32-x64   
node-v20.18.2

### Reproduceable Code

```tsx
<Camera
                isMirrored={false}
                ref={cameraRef}
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={true}
                photo={true}
                video={false}
                enableZoomGesture={true}
                flash={flashState ? "on" : "off"}
                onUIRotationChanged={(orientation) => {
                  setOrientationDegre(orientation);
                }}
              />
```

### Relevant log output

```shell
FAILURE: Build failed with an exception.
* What went wrong:
Execution failed for task ':react-native-vision-camera:configureCMakeDebug[arm64-v8a]'.
> [CXX1429] error when building with cmake using /home/expo/workingdir/build/node_modules/react-native-vision-camera/android/CMakeLists.txt: -- The C compiler identification is Clang 17.0.2
  -- The CXX compiler identification is Clang 17.0.2
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /home/expo/Android/Sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/linux-x86_64/bin/clang - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/expo/Android/Sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Configuring done
  -- Generating done
  
  C++ build system [configure] failed while executing:
      /home/expo/Android/Sdk/cmake/3.22.1/bin/cmake \
        -H/home/expo/workingdir/build/node_modules/react-native-vision-camera/android \
        -DCMAKE_SYSTEM_NAME=Android \
        -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
        -DCMAKE_SYSTEM_VERSION=24 \
        -DANDROID_PLATFORM=android-24 \
        -DANDROID_ABI=arm64-v8a \
        -DCMAKE_ANDROID_ARCH_ABI=arm64-v8a \
        -DANDROID_NDK=/home/expo/Android/Sdk/ndk/26.1.10909125 \
        -DCMAKE_ANDROID_NDK=/home/expo/Android/Sdk/ndk/26.1.10909125 \
        -DCMAKE_TOOLCHAIN_FILE=/home/expo/Android/Sdk/ndk/26.1.10909125/build/cmake/android.toolchain.cmake \
        -DCMAKE_MAKE_PROGRAM=/home/expo/Android/Sdk/cmake/3.22.1/bin/ninja \
        "-DCMAKE_CXX_FLAGS=-O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all" \
        -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/6w4i6ge2/obj/arm64-v8a \
        -DCMAKE_RUNTIME_OUTPUT_DIRECTORY=/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/6w4i6ge2/obj/arm64-v8a \
        -DCMAKE_BUILD_TYPE=Debug \
        -DCMAKE_FIND_ROOT_PATH=/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/.cxx/Debug/6w4i6ge2/prefab/arm64-v8a/prefab \
        -B/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/.cxx/Debug/6w4i6ge2/arm64-v8a \
        -GNinja \
        -DANDROID_STL=c++_shared \
        -DNODE_MODULES_DIR=/home/expo/workingdir/build/node_modules \
        -DENABLE_FRAME_PROCESSORS=OFF
    from /home/expo/workingdir/build/node_modules/react-native-vision-camera/android
  VisionCamera: Frame Processors: OFF!
  CMake Error at CMakeLists.txt:26 (add_library):
    Target "VisionCamera" links to target "ReactAndroid::reactnativejni" but
    the target was not found.  Perhaps a find_package() call is missing for an
    IMPORTED target, or an ALIAS target is missing?
  
  
  CMake Generate step failed.  Build files cannot be regenerated correctly. : com.android.ide.common.process.ProcessException: -- The C compiler identification is Clang 17.0.2
  -- The CXX compiler identification is Clang 17.0.2
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /home/expo/Android/Sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/linux-x86_64/bin/clang - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /home/expo/Android/Sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Configuring done
  -- Generating done
  
  C++ build system [configure] failed while executing:
      /home/expo/Android/Sdk/cmake/3.22.1/bin/cmake \
        -H/home/expo/workingdir/build/node_modules/react-native-vision-camera/android \
        -DCMAKE_SYSTEM_NAME=Android \
        -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
        -DCMAKE_SYSTEM_VERSION=24 \
        -DANDROID_PLATFORM=android-24 \
        -DANDROID_ABI=arm64-v8a \
        -DCMAKE_ANDROID_ARCH_ABI=arm64-v8a \
        -DANDROID_NDK=/home/expo/Android/Sdk/ndk/26.1.10909125 \
        -DCMAKE_ANDROID_NDK=/home/expo/Android/Sdk/ndk/26.1.10909125 \
        -DCMAKE_TOOLCHAIN_FILE=/home/expo/Android/Sdk/ndk/26.1.10909125/build/cmake/android.toolchain.cmake \
        -DCMAKE_MAKE_PROGRAM=/home/expo/Android/Sdk/cmake/3.22.1/bin/ninja \
        "-DCMAKE_CXX_FLAGS=-O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all" \
        -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/6w4i6ge2/obj/arm64-v8a \
        -DCMAKE_RUNTIME_OUTPUT_DIRECTORY=/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/6w4i6ge2/obj/arm64-v8a \
        -DCMAKE_BUILD_TYPE=Debug \
        -DCMAKE_FIND_ROOT_PATH=/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/.cxx/Debug/6w4i6ge2/prefab/arm64-v8a/prefab \
        -B/home/expo/workingdir/build/node_modules/react-native-vision-camera/android/.cxx/Debug/6w4i6ge2/arm64-v8a \
        -GNinja \
        -DANDROID_STL=c++_shared \
        -DNODE_MODULES_DIR=/home/expo/workingdir/build/node_modules \
        -DENABLE_FRAME_PROCESSORS=OFF
    from /home/expo/workingdir/build/node_modules/react-native-vision-camera/android
  VisionCamera: Frame Processors: OFF!
  CMake Error at CMakeLists.txt:26 (add_library):
    Target "VisionCamera" links to target "ReactAndroid::reactnativejni" but
    the target was not found.  Perhaps a find_package() call is missing for an
    IMPORTED target, or an ALIAS target is missing?

  CMake Generate step failed.  Build files cannot be regenerated correctly.
```

### Camera Device

```json
i don't have this info
```

### Device

Android

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [x] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bernborgess, 2025-04-23:
I'm having the same issue in 4.6.3

bernborgess, 2025-04-23:
> I'm having the same issue in 4.5.3

4.6.4 works again

monishkh, 2025-05-12:
> > I'm having the same issue in 4.5.3
> 
> 4.6.4 works again

did you solve issue