margelo/react-native-vision-camera #3505: 🐛 FieldOfView returned differently on the same physical device but with different version of React-Native-Vision-Camera

sekhuat, 2025-04-25:
### What's happening?

I am update my existing app that use react-native-vision-camera version 2.16.1 to 4.6.4. I found that the camera device fieldOfView value returned by getAvailableCameraDevices is not the same with version 2.16.1. Shouldn't the fieldOfView remain the same with the version upgrade since I am running the app in the identical physical device?

Below is the fieldOfView for each camera types with different versions of react-native-vision-camera:

Ver 2.16.1
wide-angle-camera: 66.56778180194215
ultra-wide-angle-camera: 79.35449698485297
telephoto-camera: 34.1279315276554

Ver 4.6.4
wide-angle-camera: 78.83660891980851
ultra-wide-angle-camera: 92.17162256451329
telephoto-camera: 42.005046621405626

The new version of react-native-vision-camera returned the fieldOfView with a large value for all camera types. My app is a viewfinder app for film photographer. It simulate the "actual" view the photographer can see from their film camera. The fieldOfView return by version 2.16.1 seem to be correct and using it I can simulate the actual view of the film camera in the mobile app. However, the fieldOfView returned by 4.6.4 will resulted a more wider view as compared to the actual film camera.

Why the fieldOfView returned by newer version different from the older version? It should be the same for the same physical mobile device right? Please advise, thank you!

### Reproduceable Code

```tsx
// Same code using both version 2.16.1 and 4.6.4
const cameraDevices = await Camera.getAvailableCameraDevices();
```

### Relevant log output

```shell
No log
```

### Camera Device

```json
[
  {
    "formats": [],
    "sensorOrientation": "landscape-left",
    "hardwareLevel": "limited",
    "maxZoom": 10,
    "minZoom": 1,
    "maxExposure": 4,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera",
      "wide-angle-camera",
      "wide-angle-camera",
      "telephoto-camera",
      "ultra-wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": true,
    "minFocusDistance": 0,
    "minExposure": -4,
    "name": "0 (BACK) androidx.camera.camera2",
    "hasFlash": true,
    "hasTorch": true,
    "position": "back",
    "id": "0"
  },
  {
    "formats": [],
    "sensorOrientation": "landscape-right",
    "hardwareLevel": "limited",
    "maxZoom": 6,
    "minZoom": 1,
    "maxExposure": 4,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera",
      "wide-angle-camera",
      "ultra-wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": true,
    "minFocusDistance": 0,
    "minExposure": -4,
    "name": "1 (FRONT) androidx.camera.camera2",
    "hasFlash": false,
    "hasTorch": false,
    "position": "front",
    "id": "1"
  },
  {
    "formats": [],
    "sensorOrientation": "landscape-left",
    "hardwareLevel": "limited",
    "maxZoom": 10,
    "minZoom": 1,
    "maxExposure": 4,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": false,
    "minFocusDistance": 0,
    "minExposure": -4,
    "name": "2 (BACK) androidx.camera.camera2",
    "hasFlash": false,
    "hasTorch": false,
    "position": "back",
    "id": "2"
  },
  {
    "formats": [],
    "sensorOrientation": "landscape-right",
    "hardwareLevel": "limited",
    "maxZoom": 6,
    "minZoom": 1,
    "maxExposure": 4,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": false,
    "minFocusDistance": 0,
    "minExposure": -4,
    "name": "3 (FRONT) androidx.camera.camera2",
    "hasFlash": false,
    "hasTorch": false,
    "position": "front",
    "id": "3"
  },
  {
    "formats": [],
    "sensorOrientation": "landscape-left",
    "hardwareLevel": "limited",
    "maxZoom": 10,
    "minZoom": 1,
    "maxExposure": 4,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": false,
    "minFocusDistance": 0,
    "minExposure": -4,
    "name": "4 (BACK) androidx.camera.camera2",
    "hasFlash": false,
    "hasTorch": false,
    "position": "back",
    "id": "4"
  },
  {
    "formats": [],
    "sensorOrientation": "landscape-left",
    "hardwareLevel": "limited",
    "maxZoom": 10,
    "minZoom": 1,
    "maxExposure": 4,
    "supportsLowLightBoost": false,
    "neutralZoom": 1,
    "physicalDevices": [
      "telephoto-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": false,
    "minFocusDistance": 0,
    "minExposure": -4,
    "name": "6 (BACK) androidx.camera.camera2",
    "hasFlash": false,
    "hasTorch": false,
    "position": "back",
    "id": "6"
  }
]
```

### Device

Huawei Mate 30 Pro

### VisionCamera Version

2.16.1 and 4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [x] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sekhuat, 2025-04-25:
> Guten Tag, Hans here. 🍻
>
> Thank you for your detailed report! It looks like you're encountering a real concern with the field of view values between versions. However, please ensure that you provide [...]

The issue is not a compilation or run time error that generate error message in the log. It is just the result returned by the `Camera.getAvailableCameraDevices();` method differ from what it was expected, therefore there is no log to provide. Thanks for your kind understanding.

sekhuat, 2025-04-26:
Hi @mrousavy please help to take a look on the issue I had reported. Two different versions of react-native-vision-camera (2.16.1 & 4.6.4) return different fieldOfView on a same physical phone. Thanks!

mrousavy, 2025-04-28:
V2 is two years old, it's not a surprise that there are some breaking changes.

closed 2025-04-28 by mrousavy