margelo/react-native-vision-camera #3508: 🔧 java.lang.NullPointerException: group.displayName must not be null

nikhilgautam810, 2025-05-01:
### How were you trying to build the app?

import { ActivityIndicator, StyleSheet, Text, View } from 'react-native'
import React from 'react'
import DynamicViewContainer from '../../../components/Other/DynamicViewContainer'
import { Camera, useCameraDevice } from 'react-native-vision-camera'

const index = () => {
    const device = useCameraDevice('back')
    if (device == null) return <ActivityIndicator size={'large'} color={colors.light.primary} />
    return (
        <DynamicViewContainer>
            <View style={styles.container}>
                <Camera
                    style={StyleSheet.absoluteFill}
                    device={device}
                    isActive={true}
                />
            </View>
        </DynamicViewContainer>
    )
}

export default index

const styles = StyleSheet.create({
    container: {
        flex: 1,
    }
})

### Full build logs

```tsx
No modules to process in combine-js-to-schema-cli. If this is unexpected, please check if you set up your NativeComponent correctly. See combine-js-to-schema.js for how codegen finds modules.

FAILURE: Build failed with an exception.

* What went wrong:
java.lang.NullPointerException: group.displayName must not be null

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 3m 49s
error Failed to install the app. Command failed with exit code 1: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081
No modules to process in combine-js-to-schema-cli. If this is unexpected, please check if you set up your NativeComponent correctly. See combine-js-to-schema.js for how codegen finds modules. FAILURE: Build failed with an exception. * What went wrong: java.lang.NullPointerException: group.displayName must not be null * Try: > Run with --stacktrace option to get the stack trace. > Run with --info or --debug option to get more log output. > Run with --scan to get full insights. > Get more help at https://help.gradle.org. BUILD FAILED in 3m 49s.
```

### Project dependencies

```json
"dependencies":{
    "react-native-vision-camera": "^4.6.4",
    "react": "19.0.0",
    "react-native": "0.78.2",
}
```

### VisionCamera Version

4.6.4

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nikhilgautam810, 2025-05-01:
I did everything according to the documentation.

GustavoSantiago113, 2025-05-07:
Hello! I was having the same issue. When I changed my gradle version from 8.12 to 8.13 in my file android\gradle\wrapper\gradle-wrapper.properties, and rebuilt the gradle by running

cd android
./gradlew clean

It worked just fine