margelo/react-native-vision-camera #3517: 🐛 Memory Leak in `useSkiaFrameProcessor` with `react-native-vision-camera` leading to app crash

rafaelrcamargo, 2025-05-08:
### What's happening?

There appears to be a memory leak when using `useSkiaFrameProcessor` from `react-native-skia` in conjunction with `react-native-vision-camera`. A simple Skia image shader running within the frame processor causes the app's memory usage to continuously increase until the app eventually crashes.

I have recorded a timelapse video using Xcode's memory profiler demonstrating the memory increasing consistently until the app crashes.

[image]

https://github.com/user-attachments/assets/a604df0d-003f-4de2-bfa6-db08988ff6a2

> The Skia shader used is a very simple `imageFilter` example [from the docs](https://react-native-vision-camera.com/docs/guides/skia-frame-processors#skia-apis), suggesting the leak might be in the bridge or a core interaction between the libraries rather than the shader logic itself.

### Reproduceable Code

```tsx
const device = useCameraDevice("back");
  const format = useCameraFormat(device, [{ photoResolution: "max" }]);

  // ...

  const invertColorsFilter = Skia.RuntimeEffect.Make(`
    uniform shader image;
    half4 main(vec2 pos) {
      vec4 color = image.eval(pos);
      return vec4((1.0 - color).rgb, 1.0);
    }
  `)!
  const shaderBuilder = Skia.RuntimeShaderBuilder(invertColorsFilter)
  const imageFilter = Skia.ImageFilter.MakeRuntimeShader(shaderBuilder, null, null)
  const paint = Skia.Paint()
  paint.setImageFilter(imageFilter)
  const frameProcessor = useSkiaFrameProcessor((frame) => {
    "worklet";
    frame.render(paint)
  }, [paint]);

<Camera
  audio={false}
  device={device}
  enableFpsGraph
  format={format}
  fps={format?.maxFps ?? 30}
  frameProcessor={frameProcessor}
  outputOrientation="preview"
  photo
  pixelFormat="rgb"
  video={false}
/>
```

### Relevant log output

```shell
// This, looping forever 👇
00:05:13.449: [info] 📸 VisionCamera.didSetProps(_:): Updating 0 props: []
00:05:13.449: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
00:05:13.449: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: false, outputsChanged: false, videoStabilizationChanged: false, orientationChanged: false, formatChanged: false, sidePropsChanged: false, torchChanged: false, zoomChanged: false, exposureChanged: false, audioSessionChanged: false, locationChanged: false)
```

### Camera Device

```json
{
  "minFocusDistance": 15,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "formats": [],
  "position": "back",
  "maxZoom": 123.75,
  "maxExposure": 8,
  "sensorOrientation": "portrait",
  "supportsRawCapture": false,
  "name": "Back Camera",
  "hasFlash": true,
  "neutralZoom": 1,
  "minExposure": -8,
  "supportsFocus": true,
  "hasTorch": true,
  "isMultiCam": false,
  "hardwareLevel": "full",
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0"
}
```

### Device

iPhone 15 (iOS 18.5)

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [x] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rafaelrcamargo, 2025-05-08:
Related issue on https://github.com/Shopify/react-native-skia/issues/2964

Admiral-Enigma, 2025-05-15:
+1 useSkiaFrameProcessor seems unusable at the moment

adriaanslechten, 2025-05-18:
+1, following

edesaras, 2025-05-19:
Is there a workaround we could use in the meantime?

rash-27, 2025-05-21:
facing the similar issue in ios, but i am able to process the frames using skia in android, 
Also getting a black screen in the front camera for both android and ios..

edesaras, 2025-05-21:
I've found a workaround that might be helpful in the meantime. Instead of using skiaframeprocessor i use regular frameprocessor set the react state and draw on Canvas from skia:

``` javascript
import { useFrameProcessor } from 'react-native-vision-camera';
import { inferenceLogic} from './inferenceLogic'; // your TFLite inference logic
import { Worklets } from 'react-native-worklets-core';

export const useInferenceLogic = (onInference: (detections: any[]) => void) => {
  const onInferenceJS= Worklets.createRunOnJS(onInference)
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const results = inferenceLogic(frame); // e.g., runs TFLite
    onInferenceJS(results);   // pass to JS side
  }, []);

  return { frameProcessor };
};
```

``` javascript
import { Canvas } from '@shopify/react-native-skia';
import { Camera, useCameraDevice } from 'react-native-vision-camera';
import { useInferenceLogic  } from './useInferenceLogic ';
import React, { useState } from 'react';

export default function App() {
  const device = useCameraDevice('back');
  const [results, setResults] = useState<any[]>([]);
  const { frameProcessor } = useInferenceLogic (setResults);

  return (
    <View style={{ flex: 1 }}>
      {device && (
        <>
          <Camera
            style={{ flex: 1 }}
            device={device}
            isActive={true}
            frameProcessor={frameProcessor}
            pixelFormat="rgb"
          />
          <Canvas style={{ position: 'absolute', width, height }}>
            // ... your skia logic
          </Canvas>
        </>
      )}
    </View>
  );
}

```

songxiaoliang, 2025-05-22:
@edesaras nice, Can you give some detailed code, it's been bugging me for a long time now. thanks!

songxiaoliang, 2025-05-26:
@rafaelrcamargo do you fix it ?

songxiaoliang, 2025-06-06:
@seanpcoyle same problem, how fix it? thanks

mdolr, 2025-07-24:
Hi, had same issue even with an empty skia frame processor, here is a repro with a simple setup like this

```tsx
import { useEffect } from "react";
import { StyleSheet } from "react-native";
import { Text } from "~/components/ui/text";
import {
  Camera,
  Frame,
  useCameraPermission,
  useCameraDevice,
  useSkiaFrameProcessor,
} from "react-native-vision-camera";

export default function CameraScreen() {
  const device = useCameraDevice("front");
  const { hasPermission, requestPermission } = useCameraPermission();

  useEffect(() => {
    if (!hasPermission) requestPermission();
  }, [hasPermission, requestPermission]);

  const frameProcessor = useSkiaFrameProcessor((frame: Frame) => {
    "worklet";
  }, []);

  if (!hasPermission) {
    return <Text>No permission</Text>;
  }

  if (device == null) {
    return <Text>No device</Text>;
  }

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      frameProcessor={frameProcessor}
      enableFpsGraph={true}
      enableZoomGesture={true}
      fps={30}
      onError={(e) => {
        console.error("Camera error", e);
      }}
    />
  );
}

```

and package versions:
```json
{
  "@shopify/react-native-skia": "^2.1.1",
  "react-native-worklets-core": "^1.6.0",
  "react-native-vision-camera": "^4.7.1",
}
```

3azam0, 2025-07-24:
"@shopify/react-native-skia": "1.5.0",
    "react-native-vision-camera": "^4.5.3",
     "react-native-worklets-core": "1.5.0",
these versions are working wfine with me android and ios but you have to enable the new arch

toonverbeek, 2025-09-04:
Update:I made some progress on the issue. 

It turns out that running `npx expo prebuild --clean` generates a version of `project.pbxproj` that leads to the memory leak. I don't understand why this happens, but after removing unused expo plugins from my `app.json` and running `npx expo prebuild --clean` again, the crashes stopped. On initial load it spikes up to 5GB+ usage but then it stabilizes itself to around 2.4GB of usage (I'm doing some complex masking). I also changed the `deploymentTarget` to `ios 18.0` (from 15.0) and this seems to also help, but I'm not entirely sure why (my suspicion is that the culprit lies within re-generating the `project.pbxproj`).

maxximee, 2025-10-03:
Hi @toonverbeek  
Talking about ios: using skia 2.2.21 I don't have the memory leak anymore, however the camera live view freezes after a second, so it's still not usable sadly. I also needed to patch vision-camera because of Canvas size error, which fixed the error but it's still freezing. Looking into it.

frodriguez-hu, 2025-10-03:
@maxximee I think it is still a memory leak.

azimgd, 2025-10-22:
What worked for us is lowering resolution, since calling `makeNonTextureImage` from `useSkiaFrameProcessor` at 60fps is quite expensive. Once photo resolution is adjusted, results began getting gc'd properly.

```diff
+ const format = useCameraFormat(device, [{ photoResolution: { width: 2048, height: 2048 } }]);

const frameProcessor = useSkiaFrameProcessor(...);

<Camera
+  format={format}
+  fps={30}
+  frameProcessor={frameProcessor}
```

chrishiguto, 2025-11-06:
Hey @maxximee does this fix the issue you were having with `makeImageFromNativeBuffer` on your end? I'm trying to fix this issue and any info you could provide would be very helpful. Thanks!

` ERROR Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera`