margelo/react-native-vision-camera #3518: 🔧 Build fails after installing `react-native-worklets-core`

mgulfam0722, 2025-05-08:
### How were you trying to build the app?

I tried to build my android app after installing `react-native-worklets-core` for frame processing, but the moment I install this library an error occurs and build fails. The app works fine with `react-native-vision-camera` installed alone.

```
// ./android/build.gradle
buildscript {
    ext {
        buildToolsVersion = "35.0.0"
        minSdkVersion = 24
        compileSdkVersion = 35
        targetSdkVersion = 35
        ndkVersion = "26.1.10909125"
        // ndkVersion = "27.1.12297006"
        kotlinVersion = "2.0.21"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("org.jetbrains.kotlin:kotlin-gradle-plugin")
    }
}

apply plugin: "com.facebook.react.rootproject"

```

### Full build logs

```tsx
$ npm run android

> TestApp@0.0.1 android
> react-native run-android

info A dev server is already running for this project on port 8081.
info Installing the app...

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ??
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at C:\Users\AG Computers\Desktop\TestApp\node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core found, Frame Processors are enabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-vision-camera:generateCodegenSchemaFromJavaScript

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a]
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...

> Task :react-native-vision-camera:configureCMakeDebug[armeabi-v7a]
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...

> Task :react-native-vision-camera:configureCMakeDebug[x86]
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...

> Task :react-native-vision-camera:configureCMakeDebug[x86_64]
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...

> Task :react-native-worklets-core:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `C:\Users\AG Computers\Desktop\TestApp\node_modules\react-native-worklets-core\android\.cxx\Debug\6c2u6043\arm64-v8a'
[... 373 lines cut ...]
```

### VisionCamera Version

^4.6.4

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

saadalla, 2025-05-10:
Hello Hans,
I would rule out the missing dependency as the app runs well for both android and iOS.
it is when we generate the  android ".aab" that issue arise.

The NDK used is ndkVersion = "27.1.12297006".
is this a problem with the react-native-worklets-core or vision camera ?
will downgrading vision camera package solve the issue ?

It is also weird since the file claimed to be missing during the ".aab" [libVisionCamera.so]
Execution failed for task ':react-native-vision-camera:externalNativeBuildRelease'.
> java.io.FileNotFoundException: 
/Users/development/Bellagio/node_modules/react-native-vision-camera/android/build/intermediates/cxx/RelWithDebInfo/3t212w65/obj/armeabi-v7a/libVisionCamera.so (No such file or directory)

does actually exist but on a different folder
/Users/saadalla/development/Bellagio/node_modules/react-native-vision-camera/android/build/intermediates/cxx/RelWithDebInfo/3t212w65/obj/armeabi-v7a
so it seems the vision-camera linker is removing my user name "saadalla" from the path.

could this be the problem ? how to fix it ?

saadalla, 2025-05-10:
ok guys I was able to fix the issue - courtesy of our friend ChatGPT

1. Add a forced nodeModules override via Gradle

In your root android/gradle.properties, add this line to explicitly pass the correct absolute path to node_modules:
VISION_CAMERA_NODE_MODULES_PATH=/Users/saadalla/development/Bellagio/node_modules

Then, inside node_modules/react-native-vision-camera/android/build.gradle, change this:
def nodeModules = findNodeModules(projectDir)
To:
def nodeModules = project.hasProperty("VISION_CAMERA_NODE_MODULES_PATH") ?
        project.getProperty("VISION_CAMERA_NODE_MODULES_PATH") :
        findNodeModules(projectDir)
✅ This makes the node_modules path deterministic and avoids any old cache issues.

 2. Delete everything related to .cxx and native builds
 
Run this from the root of your project:
rm -rf node_modules/react-native-vision-camera/android/.cxx
rm -rf node_modules/react-native-vision-camera/android/build
rm -rf ~/.gradle/caches

3. Try a clean release build

cd android
./gradlew clean
./gradlew bundleRelease --no-daemon --refresh-dependencies --stacktrace

Hope this can help someone out there,
it did save me a lot of heartache and endless debugging hours

yksa, 2025-05-19:
Is there any update?