margelo/react-native-vision-camera #3524: 🐛 Domain=AVFoundationErrorDomain Code=-11800 | The operation could not be completed | Error Domain=NSOSStatusErrorDo- main Code=-10868

renishmithani, 2025-05-13:
### What's happening?

When recording a short video, facing this error, and the audio is not coming in the video.

[image]

### Reproduceable Code

```tsx
// Parent component function for video recording
 const onLongPressIn = async () => {
    try {
      if (!cameraPermission || !microphonePermission) {
        return;
      }
      if (ref.current) {
        onToogleVideo();
        ref.current.recordVideo({
          fileType: 'mp4',
          flash: 'off',
          videoCodec: 'h265',
          onRecordingError: () => {
            setIsVideo(false);
          },
          onRecordingFinished: async video => {
            if (video.duration >= 1) {
              const thumbnail = await generaeThumbail({
                url: `file://${video.path}`,
                format: 'jpeg',
              });
              console.log(thumbnail.path, ' videoThumbnail');
              console.log(JSON.stringify(video), null, 2, ' video');
              dispatch({
                thumbnail: `${thumbnail.path}`,
                path: `file://${video.path}`,
                is_camera: false,
                type: getFileType(video.path).type,
                format: getFileType(video.path).format,
              });
              setIsVideo(false);
            } else {
              SnackBar({
                type: 'success',
                text1: 'video is too short for post',
              });
              setIsVideo(false);
            }
          },
        });
      }
    } catch (error) {}
  };

  const onPressOut = () => {
    if (isvideo && !isCameraError) {
      onFinishVideo();
    } else {
      setIsVideo(false);
      setIsCameraError(false);
      // if (isCameraError) {
      //   SnackBar({
      //     type: 'error',
      //     text1: 'Camera session have some issue',
      //   });
      // }
      onFinishVideo();
    }
  };

  const onFinishVideo = useCallback(async () => {
[... 463 lines cut ...]
  "minZoom": 1,
  "hasTorch": true,
  "hardwareLevel": "full"
}
```

### Device

iPhone 16 pro max (iOS 18), iPhone Xs (iOS 18)

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

renishmithani, 2025-05-14:
@mrousavy Here are the logs from Xcode while recording video. JFYI this issue is occurring in only iOS device and audio is not coming in video

{
  "isMultiCam": true,
  "maxZoom": 16,
  "hasTorch": true,
  "minFocusDistance": 12,
  "position": "back",
  "name": "Back Dual Camera",
  "minExposure": -8,
  "hasFlash": true,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:3",
  "sensorOrientation": "portrait",
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "formats": [],
  "hardwareLevel": "full"
}
10:24:29.164: [info] 📸 VisionCamera.didSetProps(_:): Updating 30 props: [onInitialized, cameraId, position, enableBufferCompression, preview, onOutputOrientationChanged, onStarted, onCodeScanned, top, right, isActive, video, onViewReady, onError, onStopped, isMirrored, onPreviewOrientationChanged, onPreviewStopped, enableFrameProcessor, onPreviewStarted, enableZoomGesture, left, audio, bottom, outputOrientation, height, width, zoom, onShutter, photo]
10:24:29.168: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: portrait...
10:24:29.168: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: portrait...
10:24:29.168: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
10:24:29.173: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true, locationChanged: true)
10:24:29.173: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Input Device...
10:24:29.173: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:3...
10:24:29.174: [info] 📸 VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
10:24:29.174: [info] 📸 VisionCamera.configureOutputs(configuration:): Configuring Outputs...
10:24:29.174: [info] 📸 VisionCamera.configureOutputs(configuration:): Adding Photo output...
10:24:29.176: [info] 📸 VisionCamera.configureOutputs(configuration:): Adding Video Data output...
10:24:29.177: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: portrait...
10:24:29.177: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: portrait...
10:24:29.177: [info] 📸 VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
10:24:29.180: [info] 📸 VisionCamera.setTargetOutputOrientation(_:): Setting target output orientation from device to preview...
10:24:29.183: [info] 📸 VisionCamera.getPixelFormat(for:): Available Pixel Formats: ["420v", "420f", "BGRA"], finding best match... (pixelFormat="yuv", enableHdr={false}, enableBufferCompression={true})
10:24:29.183: [info] 📸 VisionCamera.getPixelFormat(for:): Using PixelFormat: 420f...
10:24:29.541: [info] 📸 VisionCamera.init(frame:session:): Preview Layer started previewing.
10:24:29.542: [info] 📸 VisionCamera.configure(_:): Beginning AudioSession configuration...
10:24:29.542: [info] 📸 VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
10:24:29.542: [info] 📸 VisionCamera.configureAudioSession(configuration:): Adding Audio input...
10:24:29.543: [info] 📸 VisionCamera.configureAudioSession(configuration:): Adding Audio Data output...
10:24:29.543: [info] 📸 VisionCamera.configure(_:): Beginning Location Output configuration...
10:24:29.543: [info] 📸 VisionCamera.configure(_:): Committed AudioSession configuration!
10:24:29.544: [info] 📸 VisionCamera.configure(_:): Finished Location Output configuration!
{
  "isMultiCam": true,
  "maxZoom": 16,
  "hasTorch": true,
  "minFocusDistance": 12,
  "position": "back",
  "name": "Back Dual Camera",
  "minExposure": -8,
  "hasFlash": true,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:3",
  "sensorOrientation": "portrait",
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "telephoto-camera"
  ],
[... 77 lines cut ...]
}
{
  "isMultiCam": true,
  "maxZoom": 16,
  "hasTorch": true,
  "minFocusDistance": 12,
  "position": "back",
  "name": "Back Dual Camera",
  "minExposure": -8,
  "hasFlash": true,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:3",
  "sensorOrientation": "portrait",
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "formats": [],
  "hardwareLevel": "full"
}

qper228, 2025-05-16:
having the same issue, it works only once I launch the app first time another time same error

RohitDianApps, 2025-05-19:
Having the same issue, it works only once I launch the app first time another time same error

somibuon, 2025-05-23:
Do you use react-native-video?
Try to add disableAudioSessionManagement={true} to the Video component
If that doesn't work, try to downgrade the react-native-video version to 6.10.2 which is the latest version doesn't have audioSessionManager.

therveux, 2025-05-29:
@somibuon  I can confirm that after downgrading react-native-video to version 6.10.2, the error disappeared. I had recently upgraded to version 6.14, so when I saw your message, I immediately thought of that. Do we have to wait for a fix on the react-native-video side before being able to upgrade?

closed 2025-05-30 by renishmithani

RohitDianApps, 2025-05-30:
That worked perfectly!
Thank you!

rfontes-scene, 2025-06-11:
For other people encountering this issue, these are the issue reports on react-native-video:
https://github.com/TheWidlarzGroup/react-native-video/issues/4535
https://github.com/TheWidlarzGroup/react-native-video/issues/4484