margelo/react-native-vision-camera #3529: 🐛 After Scanning QR code ,Half Screen get Black

shubh-neorox, 2025-05-15:
### What's happening?

I implemented one QR scanner component ,once the user scan that code , after reading that data I am navigating user to another screen ,navigation is getting successful but it shows black screen half of the display and when press back button or changing orientension of device it vanish.already using the same component at one place it working , but this module getting this issue only in release mode in debug mode it works like a charm.

### Reproduceable Code

```tsx
import React, { useState, useEffect, useRef } from "react";
import {
  View,
  Text,
  StyleSheet,
  TouchableOpacity,
  Image,
  SafeAreaView,
  ScrollView,
  Alert,
  Platform,
  ActivityIndicator,
} from "react-native";
import { useNavigation, useRoute } from "@react-navigation/native";
import {
  responsiveHeight,
  responsiveWidth,
  responsiveSize,
} from "../../layout";
import { s, vs, ms, mvs } from "react-native-size-matters";
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useCodeScanner,
} from "react-native-vision-camera";
import NoInternetBanner from "../../components/common/NoInternetBanner";
import FONTS from "../../constants/fonts";
import Colors from "../../constants/colors";
import { useIsFocused } from '@react-navigation/native';
import AsyncStorage from "@react-native-async-storage/async-storage";
import apiService from "../../api/apiServices";
import { decryptAES256 } from "../../services/encryptionService";

// Scanner Component - Separate component for better isolation and cleanup
const Scanner = ({ onScanned, onCancel }) => {
  const device = useCameraDevice("back");
  const [isMounted, setIsMounted] = useState(true);
  
  useEffect(() => {
    return () => {
      // Ensure cleanup when unmounting
      setIsMounted(false);
    };
  }, []);
  
  const codeScanner = useCodeScanner({
    codeTypes: ["qr"],
    onCodeScanned: (codes) => {
      if (!isMounted) return;
      
      if (codes.length > 0 && codes[0]?.value) {
        onScanned(codes[0].value);
      }
    },
  });

  if (!device) {
    return (
      <View style={styles.scannerContainer}>
        <Text style={styles.scannerText}>Camera not available</Text>
      </View>
    );
[... 707 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Moto G64

### VisionCamera Version

^4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

shubh-neorox, 2025-05-16:
@mrousavy

h1nson, 2025-05-26:
IOS 18.5 get black screen.  works on IOS 18.3

1280103995, 2025-07-10:
I also have this problem on Android