margelo/react-native-vision-camera #3537: 🐛 Not able to scan QR code in some android device

hardikpatelks, 2025-05-22:
### What's happening?

QR code scanning fails on Samsung Galaxy A23 running Android 14. The camera initialises and renders correctly, but no QR codes are detected. The same implementation works fine on other devices like Samsung S21 and Pixel 6.

### Reproduceable Code

```tsx
'use strict';

import React, {useState, useEffect, useRef, useCallback} from 'react';
import PropTypes from 'prop-types';
import {
  StyleSheet,
  Dimensions,
  Vibration,
  Animated,
  Easing,
  View,
  Text,
  TouchableWithoutFeedback,
  Alert,
} from 'react-native';
import {
  Camera,
  useCameraDevice,
  useCodeScanner,
} from 'react-native-vision-camera';
import {Helper} from '../../apis';

const XQRCodeScanner = props => {
  const {
    onRead,
    vibrate,
    reactivate,
    reactivateTimeout,
    cameraTimeout,
    fadeIn,
    showMarker,
    customMarker,
    containerStyle,
    cameraStyle,
    cameraContainerStyle,
    markerStyle,
    topViewStyle,
    bottomViewStyle,
    topContent,
    bottomContent,
    cameraProps,
  } = props;

  const [scanning, setScanning] = useState(false);
  const [isCameraActivated, setIsCameraActivated] = useState(true);
  const [fadeInOpacity, setFadeInOpacity] = useState(new Animated.Value(0));
  const [isAuthorized, setIsAuthorized] = useState(false);
  const [isAuthorizationChecked, setIsAuthorizationChecked] = useState(false);
  const [disableVibrationByUser, setDisableVibrationByUser] = useState(false);

  const timer = useRef(null);
  const _scannerTimeout = useRef(null);

  useEffect(() => {
    setIsAuthorized(true);
    setIsAuthorizationChecked(true);

    if (fadeIn) {
      Animated.sequence([
        Animated.delay(1000),
        Animated.timing(fadeInOpacity, {
          toValue: 1,
          easing: Easing.inOut(Easing.quad),
          useNativeDriver: true,
        }),
      ]).start();
    }

    return () => {
      if (_scannerTimeout.current !== null) {
        clearTimeout(_scannerTimeout.current);
      }
      if (timer.current !== null) {
        clearTimeout(timer.current);
      }
    };
  }, []);

  const _setScanning = value => setScanning(value);

  const _setCamera = value => {
    setIsCameraActivated(value);
    setScanning(false);
    setFadeInOpacity(new Animated.Value(0));

    if (value && fadeIn) {
      Animated.sequence([
        Animated.delay(10),
        Animated.timing(fadeInOpacity, {
          toValue: 1,
          easing: Easing.inOut(Easing.quad),
          useNativeDriver: true,
        }),
      ]).start();
    }
  };

  const device = useCameraDevice('back');

  const onCodeScanned = useCallback(codes => {
    if (!scanning && !disableVibrationByUser) {
      if (vibrate) {
        Vibration.vibrate();
      }
      const value = codes[0]?.value;
      _setScanning(true);
      onRead({data: value});
      if (reactivate) {
        _scannerTimeout.current = setTimeout(
          () => _setScanning(false),
          reactivateTimeout,
        );
      }
    }
  }, []);

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: onCodeScanned,
  });

  const _renderCameraComponent = () => {
    return (
      <>
        {device ? (
          <Camera
            style={[styles.camera, cameraStyle]}
            device={device}
            isActive={true}
            codeScanner={codeScanner}
          />
        ) : (
          <View
            style={{
              flex: 1,
              alignItems: 'center',
              justifyContent: 'center',
            }}>
            <Text
              style={{
                textAlign: 'center',
                fontSize: 16,
              }}>
              {'Camera device not found'}
            </Text>
          </View>
        )}
      </>
    );
  };

//code

export default XQRCodeScanner;
```

### Relevant log output

```shell
no error log found in console
```

### Camera Device

```json
did not any logs
```

### Device

Samsung A23

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hardikpatelks, 2025-05-22:
@mrousavy  thanks for the reply, but as I do not have the device in which I have facing an issue I am not able to create the logs, the problem is, the package is working fine with most of the devices and as of now I just found it has a issue with Samsun A23 android OS 14 modal only with 3 different users. apart of it it is working fine and able to scan the QR code with other device with android os 14 as well

olivertylsar, 2025-05-23:
@hardikpatelks It's a known issue, the scan works only in the top-right corner of the frame on this device. See this issue for a workaround.

https://github.com/mrousavy/react-native-vision-camera/issues/3434