margelo/react-native-vision-camera #3538: 🔧 BUILD FAILED  VisionCamera: Linking react-native-worklets...   ninja: error: manifest 'build.ninja' still dirty after 100 tries

mayursancheti, 2025-05-24:
### How were you trying to build the app?

I tried to add worklet , also do npm install for both react native vision camera and npm i react-native-worklets-core, 
both installed successfully , then i tried 

const detectObjects = (frame: any) => {
		// This is a placeholder implementation
		// You should replace this with your actual object detection logic
		return [{
			boundingBox: {
				width: frame.width * 0.5,  // Example values
				height: frame.height * 0.5
			}
		}];
	};

const frameProcessor = useFrameProcessor((frame) => {
		'worklet'
		const objects = detectObjects(frame) // you need to import a real ML detector!
		if (objects.length > 0) {
			setshowDistanceLabel(true);
		  const { width, height } = objects[0].boundingBox
		  const size = width * height
	  
		  const screenSize = frame.width * frame.height
		  const fillRatio = size / screenSize
	  
		  const status = fillRatio > 0.3 ? 'perfect' : 'tooFar';
		  runOnJS(setdistanceStatus)(status);
		}
	}, []);

    // Function to capture image from camera
    const takePicture = async () => {
		if (camera.current) {
			try {
				setIsLoading(true);
	
				const photo = await camera.current.takePhoto({
					enableAutoRedEyeReduction: true,
					flash: flashMode as 'off' | 'on'
				});
	
				const uri = `file://${photo.path}`;
				setSelectedImage(uri);
	
				// === Save to Pictures/JainsConnect ===
				if (Platform.OS === 'android') {
					const granted = await PermissionsAndroid.request(
						PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE
					);
	
					if (granted === PermissionsAndroid.RESULTS.GRANTED) {
						const destDir = `${RNFS.PicturesDirectoryPath}/JainsConnect`;
	
						// Create folder if not exists
						if (!(await RNFS.exists(destDir))) {
							await RNFS.mkdir(destDir);
						}
	
						const fileName = `IMG_${Date.now()}.jpg`;
						const destPath = `${destDir}/${fileName}`;
	
						// Copy captured image to Pictures/JainsConnect
						await RNFS.copyFile(photo.path, destPath);
	
						// Make it visible in gallery
						await RNFS.scanFile(destPath);
	
						console.log('Saved to Pictures/JainsConnect:', destPath);
[... 1048 lines cut ...]
```

### VisionCamera Version

^4.6.4

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Febriansyah11, 2025-06-09:
I don't know if this is the same problem or not, but the gist is about NINJA

```
> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/Users/macbookair/Documents/Github/paydia-app-react/node_modules/react-native-vision-camera/android/.cxx/Debug/4j336o3d/arm64-v8a'
C/C++: : && /Users/macbookair/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android24 --sysroot=/Users/macbookair/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--no-undefined-version -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/4j336o3d/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/MutableJByteBuffer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraScheduler.cpp.o  /Users/macbookair/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/24/liblog.so  -landroid  /Users/macbookair/.gradle/caches/8.10.2/transforms/e95427115ce7267d1a973341407d86a4/transformed/react-android-0.76.9-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /Users/macbookair/.gradle/caches/8.10.2/transforms/dcd99e1c7b1d87c5144c0f93086b06e9/transformed/fbjni-0.6.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  /Users/macbookair/.gradle/caches/8.10.2/transforms/e95427115ce7267d1a973341407d86a4/transformed/react-android-0.76.9-debug/prefab/modules/reactnative/libs/android.arm64-v8a/libreactnative.so  -latomic -lm && :
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::initialize(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>> const&, facebook::jsi::Runtime*, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>)
C/C++: >>> referenced by JVisionCameraProxy.cpp:45 (/Users/macbookair/Documents/Github/paydia-app-react/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp:45)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::JVisionCameraProxy(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraProxy, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jsi::Runtime*, std::__ndk1::shared_ptr<facebook::react::CallInvoker> const&, facebook::jni::basic_strong_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*, facebook::jni::GlobalReferenceAllocator> const&))
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::JsiWorkletContext(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>> const&)
C/C++: >>> referenced by shared_ptr.h:279 (/Users/macbookair/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__memory/shared_ptr.h:279)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(std::__ndk1::__shared_ptr_emplace<RNWorklet::JsiWorkletContext, std::__ndk1::allocator<RNWorklet::JsiWorkletContext>>::__shared_ptr_emplace[abi:v170000]<char const (&) [13]>(std::__ndk1::allocator<RNWorklet::JsiWorkletContext>, char const (&) [13]))
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
C/C++: >>> referenced by JFrameProcessor.cpp:37 (/Users/macbookair/Documents/Github/paydia-app-react/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp:37)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
[... cut ...]
  clang++: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.
  
  C++ build system [build] failed while executing:
      /Users/macbookair/Library/Android/sdk/cmake/3.22.1/bin/ninja \
        -C \
        /Users/macbookair/Documents/Github/paydia-app-react/node_modules/react-native-vision-camera/android/.cxx/Debug/4j336o3d/arm64-v8a \
        VisionCamera
    from /Users/macbookair/Documents/Github/paydia-app-react/node_modules/react-native-vision-camera/android
```

stephan-wolf-ais, 2025-06-20:
Is there anything we can do here? We have the same problem after "yarn add..." I tried to clean by "gradlew clean" and also removing the build folder from the android folder. Nothing helps. Is there an intermediate build folder anywhere else?

Regards,
Stephan

Laionluis, 2025-06-24:
I had the same problem, after investigating further the problem in my case was the length of the project path which was too long.
I moved and renamed the project folder to have fewer characters and after that it worked.