margelo/react-native-vision-camera #3547: 🐛 App getting rejected because of MLKit having links to Iranian website.

oscar-gallog, 2025-06-06:
### What's happening?

## Problem:
The appstore is rejecting any app using react-native-vision-camera for using Google MLKit, the binary has links to iranian websites. 

## How to reproduce:
Go to your pods, and look for MLKitTextRecognitionCommon.
`cd ios/Pods/MLKitTextRecognitionCommon/Frameworks/MLKitTextRecognitionCommon.framework`

Use strings to inspect the binary:
strings MLKitTextRecognitionCommon | grep -i 'nic.ir'
You'll see output such as:
```
// ir : http://www.nic.ir/Terms_and_Conditions_ir,_Appendix_1_Domain_Rules
// Also see http://www.nic.ir/Internationalized_Domain_Names
// Two <iran>.ir entries added at request of <tech-team@nic.ir>, 2010-04-16
// ir.md : https://nic.ir.md/
// Submitted by Ali Soizi <info@nic.ir.md>
// ir : http://www.nic.ir/Terms_and_Conditions_ir,_Appendix_1_Domain_Rules
// Also see http://www.nic.ir/Internationalized_Domain_Names
// Two <iran>.ir entries added at request of <tech-team@nic.ir>, 2010-04-16
// ir.md : https://nic.ir.md/
// Submitted by Ali Soizi <info@nic.ir.md>
```

## Result: Our app getting rejected 😭
 
[image]

### Reproduceable Code

```tsx
cd ios/Pods/MLKitTextRecognitionCommon/Frameworks/MLKitTextRecognitionCommon.framework

strings MLKitTextRecognitionCommon | grep -i 'nic.ir'
```

### Relevant log output

```shell
// ir : http://www.nic.ir/Terms_and_Conditions_ir,_Appendix_1_Domain_Rules
// Also see http://www.nic.ir/Internationalized_Domain_Names
// Two <iran>.ir entries added at request of <tech-team@nic.ir>, 2010-04-16
// ir.md : https://nic.ir.md/
// Submitted by Ali Soizi <info@nic.ir.md>
// ir : http://www.nic.ir/Terms_and_Conditions_ir,_Appendix_1_Domain_Rules
// Also see http://www.nic.ir/Internationalized_Domain_Names
// Two <iran>.ir entries added at request of <tech-team@nic.ir>, 2010-04-16
// ir.md : https://nic.ir.md/
// Submitted by Ali Soizi <info@nic.ir.md>
```

### Camera Device

```json
Not important for this case
```

### Device

iPhone Pro 16 Max

### VisionCamera Version

4.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [x] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

danieloprado, 2025-07-01:
Looks like it is resolved: https://github.com/googlesamples/mlkit/issues/939, just need to update MLKit

danieloprado, 2025-07-01:
> Are you using ExpoJS? If so, maybe it is worth it to try running a code modification plugin.
> 
> I use a custom plugin to modify Android generated code to inject some code within the `MainApplication.java` file (related to enabling a custom debug flag setting for the WebView). Maybe something similar can be done for iOS for removing those links.
> 
> For example, I quickly ran this in ChatGPT to get some sort of pseudo code to modify what seems to be a binary file in your case.
> 
> const { withDangerousMod } = require('@expo/config-plugins');
> const fs = require('fs');
> const path = require('path');
> 
> module.exports = function withMLKitPatch(config) {
>   return withDangerousMod(config, [
>     'ios',
>     async (config) => {
>       const projectRoot = config.modRequest.projectRoot;
>       const mlkitPath = path.join(
>         projectRoot,
>         'ios',
>         'Pods',
>         'MLKitTextRecognitionCommon',
>         'Frameworks',
>         'MLKitTextRecognitionCommon.framework',
>         'MLKitTextRecognitionCommon'
>       );
> 
>       if (fs.existsSync(mlkitPath)) {
>         const binary = fs.readFileSync(mlkitPath);
>         const searchString = '.ir';
>         const replacement = '\x00\x00\x00';
> 
>         const modifiedBinary = Buffer.from(binary.toString().replace(new RegExp(searchString, 'g'), replacement));
>         fs.writeFileSync(mlkitPath, modifiedBinary);
>         console.log('Patched MLKitTextRecognitionCommon to remove .ir references.');
>       } else {
>         console.warn('MLKitTextRecognitionCommon binary not found. Skipping patch.');
>       }
> 
>       return config;
>     },
>   ]);
> };
> Then in `app.config.js`
> 
> const plugins = [
>   './ios-mlkit-patch'.
>   // ... other plugins
> ];
> 
> export default ({ config }) => ({
>   ...config,
>   plugins,
>   // ... other config
> });
> This is untested code, so no warranty here. I will be looking if I can use `react-native-vision-camera` to do face detection for my react native app and when searching for iOS compatibility I stumbled on your issue and thought to share this thought with you.

Didn´t work because the binaries is loaded after plugins generation code.