margelo/react-native-vision-camera #3548: 🐛 Crash with android 15: FATAL EXCEPTION: create_react_context

namvu2101, 2025-06-11:
### What's happening?

Failed to load VisionCamera C++ library!

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

<Camera
        style={{flex: 1}}
        device={device}
        isActive={true}
        codeScanner={codeScanner}
      />
```

### Relevant log output

```shell
FATAL EXCEPTION: create_react_context
                                                                                                    Process: com.momitoku.salon, PID: 6115
                                                                                                    java.lang.UnsatisfiedLinkError: dlopen failed: empty/missing DT_HASH/DT_GNU_HASH in "/data/app/~~AJEJBATT-TCQ-opIn493AQ==-Gc0DFUG5yTpYGviTnqHIYQ==/base.apk!/lib/arm64-v8a/libVisionCamera.so" (new hash type from the future?)
                                                                                                    	at java.lang.Runtime.loadLibrary0(Runtime.java:1081)
                                                                                                    	at java.lang.Runtime.loadLibrary0(Runtime.java:1003)
                                                                                                    	at java.lang.System.loadLibrary(System.java:1765)
                                                                                                    	at com.mrousavy.camera.react.CameraViewModule.<clinit>(CameraViewModule.kt:48)
                                                                                                    	at com.mrousavy.camera.react.CameraPackage.createNativeModules(CameraPackage.kt:11)
                                                                                                    	at com.facebook.react.ReactPackageHelper.getNativeModuleIterator(ReactPackageHelper.kt:32)
                                                                                                    	at com.facebook.react.NativeModuleRegistryBuilder.processPackage(NativeModuleRegistryBuilder.kt:39)
                                                                                                    	at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:1567)
                                                                                                    	at com.facebook.react.ReactInstanceManager.processPackages(ReactInstanceManager.java:1538)
                                                                                                    	at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:1448)
                                                                                                    	at com.facebook.react.ReactInstanceManager.lambda$runCreateReactContextOnNewThread$2(ReactInstanceManager.java:1216)
                                                                                                    	at com.facebook.react.ReactInstanceManager.$r8$lambda$PrBhihCbbAFk4ZReAALGanVLCyc(Unknown Source:0)
                                                                                                    	at com.facebook.react.ReactInstanceManager$$ExternalSyntheticLambda1.run(D8$$SyntheticClass:0)
                                                                                                    	at java.lang.Thread.run(Thread.java:1012)
```

### Camera Device

```json
{}
```

### Device

android 15

### VisionCamera Version

v4

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

shadowsheep1, 2025-06-13:
> Am currently also experiencing this while trying to submit to the Play Store. Was picked up in their automated review process.
> 
> Seems to be related specifically to Android 15 and/or 16K page size devices.
> 
> ```
> Error Issue 1 of 1 Detected on 1 device during testing
> Google Medium Phone (16K page size), 6.4in/16cm (Arm) (virtual)
> ```
> 
> ```
> Exception java.lang.UnsatisfiedLinkError: dlopen failed: empty/missing DT_HASH/DT_GNU_HASH in "/data/app/~~<APPDETAILS>==/<app.name>-<APPDETAILS>==/split_config.arm64_v8a.apk!/lib/arm64-v8a/libVisionCamera.so" (new hash type from the future?)
>   at java.lang.Runtime.loadLibrary0 (Runtime.java:1081)
>   at java.lang.Runtime.loadLibrary0 (Runtime.java:1003)
>   at java.lang.System.loadLibrary (System.java:1765)
>   at com.mrousavy.camera.react.CameraViewModule.<clinit> (CameraViewModule.kt:48)
>   at com.mrousavy.camera.react.CameraPackage.createNativeModules (CameraPackage.kt:11)
>   at com.facebook.react.ReactPackageTurboModuleManagerDelegate.initialize (ReactPackageTurboModuleManagerDelegate.java:98)
>   at com.facebook.react.ReactPackageTurboModuleManagerDelegate.<init> (ReactPackageTurboModuleManagerDelegate.java:57)
>   at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate.<init> (DefaultTurboModuleManagerDelegate.kt:35)
>   at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate.<init> (Unknown Source)
>   at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate$Builder.build (DefaultTurboModuleManagerDelegate.kt:62)
>   at com.facebook.react.defaults.DefaultTurboModuleManagerDelegate$Builder.build (DefaultTurboModuleManagerDelegate.kt:42)
>   at com.facebook.react.ReactPackageTurboModuleManagerDelegate$Builder.build (ReactPackageTurboModuleManagerDelegate.java:266)
>   at com.facebook.react.runtime.ReactInstance.<init> (ReactInstance.java:185)
>   at com.facebook.react.runtime.ReactHostImpl.lambda$getOrCreateReactInstanceTask$22 (ReactHostImpl.java:1189)
>   at com.facebook.react.runtime.ReactHostImpl.$r8$lambda$sX0vNfk_MttxfVwE2w9ALnXHrPI (Unknown Source)
>   at com.facebook.react.runtime.ReactHostImpl$$ExternalSyntheticLambda41.then (D8$$SyntheticClass)
>   at com.facebook.react.runtime.internal.bolts.Task$7.run (Task.java:423)
>   at com.facebook.react.runtime.internal.bolts.Executors$ImmediateExecutor.execute (Executors.kt:41)
>   at com.facebook.react.runtime.internal.bolts.Task.completeImmediately (Task.java:418)
>   at com.facebook.react.runtime.internal.bolts.Task.continueWith (Task.java:288)
>   at com.facebook.react.runtime.internal.bolts.Task.continueWith (Task.java:299)
>   at com.facebook.react.runtime.internal.bolts.Task$5.then (Task.java:354)
>   at com.facebook.react.runtime.internal.bolts.Task$5.then (Task.java:346)
>   at com.facebook.react.runtime.internal.bolts.Task$8.run (Task.java:460)
>   at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1145)
>   at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:644)
>   at java.lang.Thread.run (Thread.java:1012)
> ```

Let's see if [this](https://github.com/mrousavy/react-native-vision-camera/tree/feat/16kb-page-size) helps.

walterholohan, 2025-06-16:
Im hoping https://github.com/mrousavy/react-native-vision-camera/pull/3543 will fix the crashing

damirjuretic, 2025-06-28:
Releasing the app in the Play Store throws an issue by Google.

```
  Exception java.lang.UnsatisfiedLinkError: dlopen failed: empty/missing DT_STRTAB in "/data/app/~~HfdEzwuB9IXXVa0Fvds_Kw==/~~<APPDETAILS>==/<app.name>-<APPDETAILS>==/split_config.arm64_v8a.apk!/lib/arm64-v8a/librnworklets.so"
  at java.lang.Runtime.loadLibrary0 (Runtime.java:1081)
  at java.lang.Runtime.loadLibrary0 (Runtime.java:1003)
  at java.lang.System.loadLibrary (System.java:1765)
  at com.mrousavy.camera.react.CameraViewModule.<clinit> (CameraViewModule.kt:48)
  at com.mrousavy.camera.react.CameraPackage.createNativeModules (CameraPackage.kt:11)
  at com.facebook.react.ReactPackageHelper.getNativeModuleIterator (ReactPackageHelper.kt:32)
  at com.facebook.react.NativeModuleRegistryBuilder.processPackage (NativeModuleRegistryBuilder.kt:39)
  at com.facebook.react.ReactInstanceManager.processPackage (ReactInstanceManager.java:1587)
  at com.facebook.react.ReactInstanceManager.processPackages (ReactInstanceManager.java:1558)
  at com.facebook.react.ReactInstanceManager.createReactContext (ReactInstanceManager.java:1468)
  at com.facebook.react.ReactInstanceManager.lambda$runCreateReactContextOnNewThread$2 (ReactInstanceManager.java:1236)
  at com.facebook.react.ReactInstanceManager.$r8$lambda$PrBhihCbbAFk4ZReAALGanVLCyc (Unknown Source)
  at com.facebook.react.ReactInstanceManager$$ExternalSyntheticLambda1.run (D8$$SyntheticClass)
  at java.lang.Thread.run (Thread.java:1012)
```

"react-native-vision-camera": "4.7.0"
"react-native-worklets-core": "1.5.0"
"react-native": "0.78.2"

Google was testing the app with this emulator: Medium Phone (16K page size), Android 15 (SDK 35), ABI arm64-v8a

patthipati, 2025-10-06:
**I’m facing the same issue**

I’m getting the exact same crash when using VisionCamera on Android:

```
java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "__cxa_init_primary_exception" 
referenced by "libVisionCamera.so"
```
Has anyone resolved this?

Jamal-Ud-Din-hub, 2025-11-08:
`ndkVersion = "27.1.12297006"` this worked for me

```
"react-native": "0.80.2",
"react-native-vision-camera": "^4.7.2",
"react-native-worklets-core": "^1.6.2",
```

Its working fine with NDK 27 my issue was using NDK 28

nitasn, 2025-12-04:
@phazei you're a god living among mortals

AvinashShine, 2025-12-31:
has anyone find solution with 

ndk 28 and rn 0.77.1