margelo/react-native-vision-camera #3551: 🔧 Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]

dev-hk22, 2025-06-13:
### How were you trying to build the app?

I tried cd android then ./gradlew clean then npm run android still im getting same issue even after ./gradlew clean some times its work then after give same issue here.

### Full build logs

```tsx
npm run android

> pythonSDK@0.0.1 android
> react-native run-android

info Installing the app...

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at /Users/hemant/react-native/python-native/pythonSDK/node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core found, Frame Processors are enabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/Users/hemant/react-native/python-native/pythonSDK/node_modules/react-native-vision-camera/android/.cxx/Debug/3g436y6y/arm64-v8a'
C/C++: : && /Users/hemant/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android26 --sysroot=/Users/hemant/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--no-undefined-version -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/3g436y6y/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/MutableJByteBuffer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraScheduler.cpp.o  /Users/hemant/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/26/liblog.so  -landroid  /Users/hemant/.gradle/caches/8.10/transforms/203706ffc735736c6975b5afb10dfeb4/transformed/react-android-0.76.1-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /Users/hemant/.gradle/caches/8.10/transforms/09f9f76132586f2a395eb243b3b6a097/transformed/fbjni-0.6.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  /Users/hemant/.gradle/caches/8.10/transforms/203706ffc735736c6975b5afb10dfeb4/transformed/react-android-0.76.1-debug/prefab/modules/reactnative/libs/android.arm64-v8a/libreactnative.so  -latomic -lm && :
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::initialize(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>> const&, facebook::jsi::Runtime*, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>)
C/C++: >>> referenced by JVisionCameraProxy.cpp:45 (/Users/hemant/react-native/python-native/pythonSDK/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp:45)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::JVisionCameraProxy(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraProxy, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jsi::Runtime*, std::__ndk1::shared_ptr<facebook::react::CallInvoker> const&, facebook::jni::basic_strong_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraScheduler, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*, facebook::jni::GlobalReferenceAllocator> const&))
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
C/C++: >>> referenced by JFrameProcessor.cpp:37 (/Users/hemant/react-native/python-native/pythonSDK/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp:37)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
C/C++: >>> referenced by JVisionCameraProxy.h:42 (/Users/hemant/react-native/python-native/pythonSDK/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.h:42)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp.o:(vision::JSharedArray::JSharedArray(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JSharedArray, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<facebook::jni::HybridClass<vision::JVisionCameraProxy, facebook::jni::detail::BaseHybridClass>::JavaPart, facebook::jni::JObject, void>::_javaobject*> const&, facebook::jni::alias_ref<facebook::jni::JByteBuffer>))
C/C++: ld.lld: error: undefined symbol: RNWorklet::RuntimeLifecycleMonitor::addListener(facebook::jsi::Runtime&, RNWorklet::RuntimeLifecycleListener*)
C/C++: >>> referenced by WKTRuntimeAwareCache.h:46 (/Users/hemant/react-native/python-native/pythonSDK/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTRuntimeAwareCache.h:46)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::RuntimeAwareCache<std::__ndk1::shared_ptr<facebook::jsi::Function>>::get(facebook::jsi::Runtime&))
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiWorkletContext::JsiWorkletContext(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>> const&)
C/C++: >>> referenced by shared_ptr.h:279 (/Users/hemant/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__memory/shared_ptr.h:279)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JVisionCameraProxy.cpp.o:(std::__ndk1::__shared_ptr_emplace<RNWorklet::JsiWorkletContext, std::__ndk1::allocator<RNWorklet::JsiWorkletContext>>::__shared_ptr_emplace[abi:v170000]<char const (&) [13]>(std::__ndk1::allocator<RNWorklet::JsiWorkletContext>, char const (&) [13]))
C/C++: ld.lld: error: undefined symbol: RNWorklet::RuntimeLifecycleMonitor::removeListener(facebook::jsi::Runtime&, RNWorklet::RuntimeLifecycleListener*)
C/C++: >>> referenced by WKTRuntimeAwareCache.h:37 (/Users/hemant/react-native/python-native/pythonSDK/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTRuntimeAwareCache.h:37)
C/C++: >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessors/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::RuntimeAwareCache<std::__ndk1::shared_ptr<facebook::jsi::Function>>::~RuntimeAwareCache())
C/C++: ld.lld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()
[... cut ...]
### Additional information

- [ ] I am using Expo
- [x] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jerictayson, 2025-06-15:
I'm also having the same issue, the fix for me is to prebuild the project with clean flag `npx expo prebuild --clean` (i'm using expo), then go to android folder, execute ./gradlew clean and retry to build the project again.

MaratowKyryl, 2025-06-16:
Does any solution for non-expo projects exist? VisionCamera_enableFrameProcessors=false in gradle.properties is the only way I can build android now.

dev-hk22, 2025-06-16:
@MaratowKyryl  Try this ./gradlew clean && rm -rf .gradle

Hemantkonda, 2025-06-16:
@MaratowKyryl Try this cd android ->  ./gradlew clean && rm -rf .gradle
It will help you.

Hemantkonda, 2025-06-16:
@MaratowKyryl did you add ??

module.exports = {
presets: ['module:@react-native/babel-preset'],
plugins: [
["react-native-worklets-core/plugin"],
// ["react-native-reanimated/plugin", { processNestedWorklets: true, globals: ['__labelImage'], }]
]
};
this plugin into babel.config.js

MaratowKyryl, 2025-06-16:
@Hemantkonda , yessir, I the first place

hemantvss, 2025-06-23:
> > > If I do `VisionCamera_enableFrameProcessors=false` it works but I need frame processor since we need ocr. Can anyone suggest any other alternative to make it work?
> > 
> > [@jerictayson](https://github.com/jerictayson) he has suggested try...
> 
> My project is a non expo project

cd android -> ./gradlew clean && rm -rf .gradle

Hemantkonda, 2025-06-24:
> > > If I do `VisionCamera_enableFrameProcessors=false` it works but I need frame processor since we need ocr. Can anyone suggest any other alternative to make it work?
> > 
> > 
> > [@jerictayson](https://github.com/jerictayson) he has suggested try...
> 
> My project is a non expo project
@faiyyazk25 cd android -> ./gradlew clean && rm -rf .gradle

Mohamed-Ziyad, 2025-07-20:
Same issue - RN 0.76.1     
 "react-native-vision-camera": "^4.6.1",
 "react-native-worklets-core": "^1.5.0",

any fixes?

MaratowKyryl, 2025-07-23:
I came across several solutions, two of them are working fine for me. The first one is mentioned in comment above(https://github.com/mrousavy/react-native-vision-camera/issues/3551#issuecomment-3007880900) and the second one more stable is to upgrade my min sdk version in my android/build.gradle to 26

myzkyy, 2025-08-14:
Try this:

```
./gradlew generateCodegenArtifactsFromSchema --rerun-tasks
```

In my case, this command solved the error of `Task :react-native-worklets-core:configureCMakeRelWithDebInfo[arm64-v8a]`.

cf. https://github.com/facebook/react-native/issues/47666#issuecomment-2717398025