margelo/react-native-vision-camera #3560: 🐛 Audio is not getting recoded while video recording in IOS

Zshubham, 2025-06-30:
### What's happening?

This code is working fine with Android, but I am getting issues in IOS that after recording the Video, the audio goes missing in the video.

### Reproduceable Code

```tsx
import React, {useState, useRef, useCallback, useEffect} from 'react';
import {View, Text, StyleSheet, Platform, Alert} from 'react-native';
import {
  Camera,
  useCameraDevices,
  CameraDevice,
} from 'react-native-vision-camera';
import {useNavigation, useIsFocused} from '@react-navigation/native';
import {HP} from '../../shared/themes/responsive';
import Header from '../../shared/components/header';
import {useAuth} from '../../../authenticationFlow/AuthContext';
import Loader from '../../shared/components/Loader';
import {UploadShortsNavigationProp} from './types';
import PreviewScreen from './PreviewScreen';
import {useToast} from '../../shared/contexts/useToast';
import {useAlert} from '../../shared/contexts/useAlert';
import CameraControls from './components/CameraControls';

// Import utilities
// import {
//   VideoUtils,
//   getVideoRecordingOptions,
//   getVideoPlaybackOptions,
// } from './utils/videoUtils';
// import {
//   getAudioRecordingOptions,
//   initializeAudioSession,
// } from './utils/audioUtils';

const UploadShorts: React.FC = () => {
  const navigation = useNavigation<UploadShortsNavigationProp>();
  const cameraRef = useRef<Camera>(null);
  const devices = useCameraDevices();
  const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>(
    'back',
  );
  const device = devices?.find(device => device.position === cameraPosition);
  const {userToken, userId} = useAuth();
  const isFocused = useIsFocused();
  const toast = useToast();
  const alert = useAlert();

  const [isRecording, setIsRecording] = useState(false);
  const [recordedVideo, setRecordedVideo] = useState<string | null>(null);
  const [isCompressing, setIsCompressing] = useState(false);
  const [errorMessage, setErrorMessage] = useState<string | null>(null);
  const [recordingTime, setRecordingTime] = useState(0);
  const [showPreview, setShowPreview] = useState(false);
  const [flashEnabled, setFlashEnabled] = useState(false);

  // Check permissions on mount
  useEffect(() => {
    const checkPermissions = async () => {
      try {
        const cameraPermission = await Camera.getCameraPermissionStatus();
        const microphonePermission =
          await Camera.getMicrophonePermissionStatus();

        console.log('Camera permission:', cameraPermission);
        console.log('Microphone permission:', microphonePermission);

        if (
          cameraPermission === 'denied' ||
[... 334 lines cut ...]
  "maxExposure": 8,
  "position": "back",
  "isMultiCam": true
}
```

### Device

iphone 11 pro max 18.5

### VisionCamera Version

4.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

saeidtkh, 2025-06-30:
@Zshubham For me there was a conflict with react-native-video. if you have react-native-video somewhere in your project, try to use this version of it:
` "react-native-video": "^6.8.2",
`
and also add this props:
                 
` disableAudioSessionManagement={true}
`

This worked for me and now I have the audio as well. :)

LOGANLEEE, 2025-07-01:
> I downgrade `"react-native-video": "6.10.0"` it works fine

what is your version of  react-native-vision-camera? and react-native also are you using new arch??

LOGANLEEE, 2025-07-02:
okay i tried this method and it works ios 18.5 iphone 16.

once component render initializing camera 3times then that error goes away but still why..?

anyway hope this appoarch can help others as i invested 2days only for this.

`useEffect(() => {
    const max = 3; // Number of times to initialize
    if (cameraKey < max) {
      reInitializing(); // Already mounted once, so do 2 more
    }
    // No cleanup needed since this is only on mount
  }, [cameraKey]);
`

`<Camera
        key={cameraKey}
        ref={camera}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={isActive}
        audio={true}
        video={true}
        onInitialized={() => {
          const timeout = setTimeout(() => {
            setIsCameraInitialized(true);
          }, 1000);

          return () => {
            clearTimeout(timeout);
          };
        }}
        onError={handleCameraError}
      />
`

nhutlyhcmus, 2025-07-07:
> what is your version of react-native-vision-camera? and react-native also are you using

"react-native-vision-camera": "^4.7.0",
"react-native-video": "6.10.0",

chqamarali, 2025-07-25:
i'm facing this error first time when recording the video through vision camera on ios:

[unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (-10868), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x13a486250 {Error Domain=NSOSStatusErrorDomain Code=-10868 "(null)"}}]

video is recorded but audio is missing.
when i restart the app than every thing is working fine.

i also see some people are saying about the react-native-video package downgrading the version but why we need to downgrade the version of react-native-video its different package form the vision camera.

is there any permanent solution available to fix this error?

vkanagarajan, 2025-09-10:
> Same goes for me too
> 
> The video recording works as expected on the first attempt, capturing both audio and video. However, after navigating away from the recording screen and returning, subsequent recordings only capture video while the audio track is missing. Interestingly, restarting the app restores the functionality for the first recording, but the same issue reoccurs on the second attempt.
> 
> I have both audio and camera permissions are granted.
> 
> it is only happening in IOS ,in android it is working fine.
> 
> <Camera key={this.state.cameraInstanceKey} ref={ref => {this.camera = ref}} style={styles.preview} device={cameraType ? backdevice : frontdevice} isActive={isCameraActive} video={true} audio={true} photo={false} onInitialized={this.onVisionCameraInitialized} onError={this.onVisionCameraError} />

i am also facing exactly  same issue. any solution found for this issue

Karina5Aryan, 2025-09-10:
@vkanagarajan 

Try this solution 

Worked for me

https://github.com/TheWidlarzGroup/react-native-video/commit/d2c92a1f3f6579aa6607389de8e51e4b75012f3b in library react-native-video

Abdullo-0901, 2025-10-20:
@LOGANLEEE @cloud-github 
```I have the same problem. On iOS, the sound disappears on the second or third recording. On Android, it happens during the recording — meaning, half of the video has sound, and the other half doesn’t.```

sleekLancelot, 2025-11-07:
I use: 
   ```
"react-native-video": "^6.17.0",
    "react-native-vision-camera": "^4.7.2",
```

But the video is still mute after adding `disableAudioSessionManagement={true}`